raspberrypi/linux #6559: rp1-firmware: probe of rp1_firmware failed with error -110

nbuchwitz, 2024-12-21:
### Describe the bug

Pi500 complains that it cannot load Rp1 firmware / deferred probe pending for PIO driver:

```
[    3.740264] rp1-firmware: probe of rp1_firmware failed with error -110
[    3.740342] rp1-firmware rp1_firmware: Failed to get mbox channel: -16
[    3.740346] rp1-firmware: probe of rp1_firmware failed with error -16
[...]
[   13.792437] platform 1f00178000.pio: deferred probe pending
[   40.642224] ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
```

Same kernel / sd card runs on Pi5 without any issues. Pi500 run fine with `Linux pi500 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux`. So probably a regression of the latest rp1-pio driver ?

### Steps to reproduce the behaviour

Boot Pi500 with latest updates.

### Device (s)

Raspberry Pi 5

### System

```
Linux pi500 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
```

pelwell, 2024-12-21:
What's the EEPROM version? It's almost certainly just too old.

nbuchwitz, 2024-12-21:
EEPROM: `Mon Sep 23 01:02:56 PM UTC 2024 (1727096576)`

Errors are gone after I ran `rpi-eeprom-update -a`. But this makes we wonder: Shouldn't the firmware updated automatically by `rpi-eeprom-update.service`? Otherwise pis will break due incompatibility issues like this. Afaik the eeprom was enabled by default?

peterharperuk, 2024-12-21:
If you upgrade the rpi-eeprom component via apt it should install the 12 Nov release I believe.

nbuchwitz, 2024-12-21:
Unfortunately it haven't. I had to manually trigger the update with rpi-eeprom-update

nbuchwitz, 2024-12-21:
Found something in the logs of `rpi-eeprom-update.service`:
```Dec 21 21:24:31 pi500 rpi-eeprom-update[550]: Skipping automatic bootloader upgrade. current 1727096576 >= min 1725975630```

I guess `BOOTLOADER_AUTO_UPDATE_MIN_VERSION` needs to be updated, @timg236?

pelwell, 2024-12-22:
PIO support is not a requirement. The error message is ugly, but few will see it. I don't see an urgent need to bump the minimum version, but it will happen eventually.

nbuchwitz, 2024-12-22:
No, it's not urgent, I agree. Just wanted to point out why the automatic update isnt happening (yet).

I close this issue as everything has been clarified. Thanks for the prompt feedbakc and a aappy 4th of advent.

closed 2024-12-22 by nbuchwitz