raspberrypi/linux #6565: Can't connect network.

strayedstar, 2024-12-24:
### Describe the bug

System:Rbpi OS
I build a kernel by the way:
[image]
but after flash and boot,I can't get any network connect.
dmesg -k
[image]
even set a static ip and route manually,it still can't work.

### Steps to reproduce the behaviour

build a kernel and boot it

### Device (s)

Raspberry Pi 5

### System

[image]
kernel version:6.6.67-rpi/6.13-rc3-rpi

kaysievers, 2024-12-30:
I see the same. The wired Ethernet does not transmit packages.

DrBest3787, 2025-01-05:
On Raspberry 4 the Ethernet AND the WiFi works. Dis you Check the cables and Adapters

strayedstar, 2025-01-14:
> On Raspberry 4 the Ethernet AND the WiFi works. Dis you Check the cables and Adapters

Yes,I recheck it,but unhelpful,the adapters don't accept any package.
Mainline kernel already support Raspberry Pi 4,so you can do it easy.

kaysievers, 2025-01-21:
> I see the same. The wired Ethernet does not transmit packages.

Updating the device tree files in /boot seems to fix it for me.