raspberrypi/linux #6573: RTL8188ftv/fu's firmware is missing

chemwolf6922, 2025-01-01:
### Describe the bug

I'm testing a rtl8188ftv USB wifi adapter on a CM4. Its driver is already included by option CONFIG_RTL8XXXU. And the kernel module can be correctly loaded. However, the chip's firmware is missing under /lib/firmware/rtlwifi. 

### Steps to reproduce the behaviour

1. Plug the USB wifi adapter. 
2. Read dmesg
```shell
[    7.954735] usb 1-1.1: rtl8xxxu: Loading firmware rtlwifi/rtl8188fufw.bin
[    7.955718] usb 1-1.1: Direct firmware load for rtlwifi/rtl8188fufw.bin failed with error -2
[    7.955759] usb 1-1.1: request_firmware(rtlwifi/rtl8188fufw.bin) failed
[    7.955771] usb 1-1.1: Fatal - failed to load firmware
```
```Shell
$ ls /lib/firmware/rtlwifi/rtl8188*
/lib/firmware/rtlwifi/rtl8188efw.bin  /lib/firmware/rtlwifi/rtl8188eufw.bin
```
The firmware is missing. 

### Device (s)

Raspberry Pi CM4 Lite

### System

Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

Aug 30 2024 19:17:39
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)

Linux pi4 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

chemwolf6922, 2025-01-01:
Moving this to the rpi os repo.

chemwolf6922, 2025-01-01:
Reopening this as I failed to find a place for this issue. The suggested "userland" repo is marked as ancient. If will be great if someone can point me to the right place (NOT the forum, I'm trying to report a configuration bug. Not starting a discussion with other users. )

cmitu, 2025-01-02:
The `/lib/firmware/rtlwifi/` folder comes from the `firmware-realtek` package, which I think it's packaged in RaspiOS from https://github.com/RPi-Distro/firmware-nonfree.

XECDesign, 2025-01-02:
We generally only update the firmware packages only to support our hardware.

The issue should be reported to Debian. If it's fixed there, it will end up in Raspberry Pi OS as well.

chemwolf6922, 2025-01-03:
> We generally only update the firmware packages only to support our hardware.
> 
> The issue should be reported to Debian. If it's fixed there, it will end up in Raspberry Pi OS as well.

Thanks for the info. The problem is Debian uses an older kernel without this chip’s driver. The driver was added in 6.2. So they would have no reason to add the firmware.

chemwolf6922, 2025-01-03:
> The `/lib/firmware/rtlwifi/` folder comes from the `firmware-realtek` package, which I think it's packaged in RaspiOS from https://github.com/RPi-Distro/firmware-nonfree.

Thanks. I’ll take a look.

cmitu, 2025-01-03:
> The driver was added in 6.2. So they would have no reason to add the firmware.

It's actually added, but in testing/unstable - https://packages.debian.org/trixie/all/firmware-realtek/filelist. I think @XECDesign is correct - the proper place to request this is the upstream Debian package, but it's already added there.

XECDesign, 2025-01-03:
The good news is that it's also in bookworm-backports and there's another change I need to make to the package. While I'm at it, I'll update to the newer version, which should resolve your problem.

chemwolf6922, 2025-01-03:
Thanks again. Closing this one since it's already being tracked. 
I copied the firmware from another distro as a work around for now.

closed 2025-01-03 by chemwolf6922