raspberrypi/linux #6577: [RPi5] [linux-rpi-16k-6.12.6-1-aarch64] Kernel panic

akuropka, 2025-01-01:
### Describe the bug

From time to time system stops with a kernel panic. This should be around since kernel 6.10.

### Steps to reproduce the behaviour

Power up the device and wait for sime time.
_Additional:_ one device is mounted with `fs=ntfs3`.

### Device (s)

Raspberry Pi 5

### System

root@raspberry ~ # cat /etc/os-release
NAME="Arch Linux ARM"
PRETTY_NAME="Arch Linux ARM"
ID=archarm
ID_LIKE=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinuxarm.org/"
DOCUMENTATION_URL="https://archlinuxarm.org/wiki"
SUPPORT_URL="https://archlinuxarm.org/forum"
BUG_REPORT_URL="https://github.com/archlinuxarm/PKGBUILDs/issues"
LOGO=archlinux-logo

root@raspberry ~ # vcgencmd version
2024/11/12 16:10:44
Copyright (c) 2012 Broadcom
version 4b019946 (release) (embedded)

root@raspberry ~ # uname -a
Linux raspberry 6.12.6-1-rpi-16k #1 SMP PREEMPT Fri Dec 20 07:29:33 MST 2024 aarch64 GNU/Linux

### Logs

[245585.901482] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: __schedule+0xb60/0xb68
[245585.911553] CPU: 2 UID: 1000 PID: 22112 Comm: smbd[192.168.18 Tainted: G         C         6.12.6-1-rpi-16k #1
[245585.921700] Tainted: [C]=CRAP
[245585.924763] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[245585.930711] Call trace:
[245585.933250]  dump_backtrace.part.0+0xe0/0x100
[245585.933716]  show_stack+0x20/0x40
[245585.941130]  dump_stack_lvl+0x60/0x80
[245585.944897]  dump_stack+0x18/0x28
[245585.948312]  panic+0x170/0x370
[245585.951466]  __ktime_get_real_seconds+0x0/0x20
[245585.956017]  __schedule+0xb60/0xb68
[245585.959608] SMP: stopping secondary CPUs
[245585.963635] Kernel Offset: 0x106f4e400000 from 0xffffc0080000000
[245585.969845] PHYS_OFFSET:0x0
[245585.972820] CPU features: 0x10,00000017,00280928,0200720b
[245585.978332] Memory Limit: none
[245585.981487] ---[ end Kernel panic - not syncing: stack protector: Kernel stack is corrupted in: __schedule++0xb60/0xb68 ]---

[image]
[image]

### Additional context

Kernel is based on commit 49679ed862ff4c05a3710882936182aa86354c13 (https://github.com/archlinuxarm/PKGBUILDs/commit/195c77ad555e60a5264fc5b0dfb761781eb22811).

pelwell, 2025-01-06:
I'm not an ntfs3 user, so this may be hard to diagnose. Can you try with `kernel=kernel8.img` in config.txt? This will force the Pi 5 to use a kernel with a more conventional page size of 4kB.

akuropka, 2025-01-08:
Thanks for your idea @pelwell, I switched the kernel to the lower page size but result was the same two days later.
I will be switching to ntfs-3g for testing to see if this has any effect.