raspberrypi/linux #6585: tc358743 not found on address 0x1e

hen-79, 2025-01-07:
[image]

To reproduce

1. Connect any HDMI-CSI bridge based on TC358743 to the Pi ( I am currently using Geekworm X1301 capture HAT)
2. Download and flash official Raspberry Pi OS full desktop version.
3. Add line dtoverlay=tc358743 to /boot/firmware/config.txt
4. Reboot and make sure that the device was unable to open video0 using v4l2-ctl --list-devices
5. Issue with reading register 0x0 from address 0xf. Additionally, address 0x1e does not seem to correspond to a TC358743 as expected using dmesg | grep tc35.

I am currently using raspberry pi 5.

[image]

hen-79, 2025-01-08:
@6by9 Understood thanks for the information.

closed 2025-01-08 by hen-79

msmuenchen, 2025-12-09:
@6by9 @hen-79  I don't know when it changed, but on an up-to-date Raspbian the overlay needs to be called `tc358743-pi5` for Pi 5 hosts.

See also https://github.com/raspberrypi/linux/issues/5764#issuecomment-3633096712