raspberrypi/linux #6593: [rpi5] crash on loading firmware rp1

spockfish, 2025-01-09:
### Describe the bug

With the latest build [62c3397] the kernel crashes on a Pi5 when it wants to load firmware for the rp1.

### Steps to reproduce the behaviour

boot

### Device (s)

Raspberry Pi 5

### System

Linux ropieee5 6.6.69-SPCKFSH-v8-16k #1 SMP PREEMPT Wed Jan  8 20:40:37 CET 2025 aarch64 GNU/Linux

### Logs

rp1-firmware: probe of rp1_firmware failed with error -110
[    2.719830] Unable to handle kernel paging request at virtual address ffffffc081c45f04
[    2.719841] Mem abort info:
[    2.719844]   ESR = 0x0000000096000047
[    2.719848]   EC = 0x25: DABT (current EL), IL = 32 bits
[    2.719854]   SET = 0, FnV = 0
[    2.719858]   EA = 0, S1PTW = 0
[    2.719861]   FSC = 0x07: level 3 translation fault
[    2.719866] Data abort info:
[    2.719869]   ISV = 0, ISS = 0x00000047, ISS2 = 0x00000000
[    2.719874]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[    2.719878]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    2.719883] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000001116000
[    2.719889] [ffffffc081c45f04] pgd=10000000fffff003, p4d=10000000fffff003, pud=10000000fffff003, pmd=10000000067a1003, pte=0000000000000000
[    2.719905] Internal error: Oops: 0000000096000047 [#1] PREEMPT SMP
[    2.719911] Modules linked in: vc4 snd_soc_hdmi_codec drm_display_helper cec snd_soc_core snd_compress aes_ce_blk aes_ce_cipher snd_pcm_dmaengine rpi_panel_attiny_regulator edt_ft5x06 tc358762 ghash_ce snd_pcm gf128mul libaes v3d drm_rp1_dsi regmap_i2c snd_timer spidev sha2_ce drm_dma_helper snd sha256_arm64 panel_simple gpu_sched sha1_ce drm_kms_helper drm_shmem_helper drm i2c_designware_platform spi_bcm2835 gpio_keys raspberrypi_hwmon i2c_designware_core rp1_pio(+) rp1_adc drm_panel_orientation_quirks raspberrypi_gpiomem rp1_mailbox rp1 backlight nvmem_rmem cfg80211 rfkill uio_pdrv_genirq uio sch_fq_codel pkcs8_key_parser fuse nfnetlink ipv6
[    2.720005] CPU: 1 PID: 179 Comm: (udev-worker) Not tainted 6.6.69-SPCKFSH-v8-16k #1
[    2.720014] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    2.720019] pstate: 00400009 (nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    2.720026] pc : __memcpy_toio+0x14/0x98
[    2.720036] lr : rp1_firmware_message+0x74/0x140 [rp1]

pelwell, 2025-01-09:
```
rp1-firmware: probe of rp1_firmware failed with error -110
```
This error shows that Pi 5 EEPROM needs to be updated to provide support for the RP1 firmware mailbox, but that shouldn't lead to a crash.

Unfortunately you've pruned the crash log so much that I can't work out what went wrong.
1. Did the Pi continue to boot?
2. Had you loaded any PIO-related overlays?

spockfish, 2025-01-09:
Hi @pelwell ,

In answer to your questions:

1. yes, all functional, the kernel just continues booting and everything is ok.
2. nope.

Here's a more complete dmesg:

```
[    2.719546] rp1-firmware: probe of rp1_firmware failed with error -110
[    2.719830] Unable to handle kernel paging request at virtual address ffffffc081c45f04
[    2.719841] Mem abort info:
[    2.719844]   ESR = 0x0000000096000047
[    2.719848]   EC = 0x25: DABT (current EL), IL = 32 bits
[    2.719854]   SET = 0, FnV = 0
[    2.719858]   EA = 0, S1PTW = 0
[    2.719861]   FSC = 0x07: level 3 translation fault
[    2.719866] Data abort info:
[    2.719869]   ISV = 0, ISS = 0x00000047, ISS2 = 0x00000000
[    2.719874]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[    2.719878]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    2.719883] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000001116000
[    2.719889] [ffffffc081c45f04] pgd=10000000fffff003, p4d=10000000fffff003, pud=10000000fffff003, pmd=10000000067a1003, pte=0000000000000000
[    2.719905] Internal error: Oops: 0000000096000047 [#1] PREEMPT SMP
[    2.719911] Modules linked in: vc4 snd_soc_hdmi_codec drm_display_helper cec snd_soc_core snd_compress aes_ce_blk aes_ce_cipher snd_pcm_dmaengine rpi_panel_attiny_regulator edt_ft5x06 tc358762 ghash_ce snd_pcm gf128mul libaes v3d drm_rp1_dsi regmap_i2c snd_timer spidev sha2_ce drm_dma_helper snd sha256_arm64 panel_simple gpu_sched sha1_ce drm_kms_helper drm_shmem_helper drm i2c_designware_platform spi_bcm2835 gpio_keys raspberrypi_hwmon i2c_designware_core rp1_pio(+) rp1_adc drm_panel_orientation_quirks raspberrypi_gpiomem rp1_mailbox rp1 backlight nvmem_rmem cfg80211 rfkill uio_pdrv_genirq uio sch_fq_codel pkcs8_key_parser fuse nfnetlink ipv6
[    2.720005] CPU: 1 PID: 179 Comm: (udev-worker) Not tainted 6.6.69-SPCKFSH-v8-16k #1
[    2.720014] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    2.720019] pstate: 00400009 (nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    2.720026] pc : __memcpy_toio+0x14/0x98
[    2.720036] lr : rp1_firmware_message+0x74/0x140 [rp1]
[    2.720051] sp : ffffffc081be3810
[    2.720055] x29: ffffffc081be3810 x28: ffffffc0797d81c0 x27: ffffffc0814d6780
[    2.720064] x26: ffffffc081be3c30 x25: 0000000000020004 x24: ffffffc081be3870
[    2.720073] x23: 0000000000000008 x22: ffffffc081be386c x21: ffffffc0797b1050
[    2.720081] x20: 0000000000000004 x19: ffffff800a14c600 x18: ffffffc081beb850
[    2.720089] x17: 0000000000000000 x16: 0000000000000000 x15: ffffff800b18e2ca
[    2.720098] x14: 0000000000000001 x13: 006567646972625f x12: 000000000000023c
[    2.720106] x11: 000000000000003e x10: 0000000000001a50 x9 : ffffffc0797af07c
[    2.720115] x8 : ffffff800637f6f0 x7 : 000000000000010e x6 : dead000000000122
[    2.720123] x5 : ffffffc0797b1060 x4 : ffffffc0797b1060 x3 : 0000000000000020
[    2.720131] x2 : 0000000000000004 x1 : ffffffc081be386d x0 : ffffffc081c45f04
[    2.720140] Call trace:
[    2.720144]  __memcpy_toio+0x14/0x98
[    2.720150]  rp1_firmware_get_feature+0x54/0xb0 [rp1]
[    2.720162]  rp1_pio_probe+0xf0/0x338 [rp1_pio]
[    2.720182]  platform_probe+0x70/0xd8
[    2.720195]  really_probe+0x150/0x2c8
[    2.720203]  __driver_probe_device+0x80/0x140
[    2.720210]  driver_probe_device+0x44/0x170
[    2.720217]  __driver_attach+0x98/0x1b0
[    2.720224]  bus_for_each_dev+0x84/0xf0
[    2.720231]  driver_attach+0x2c/0x40
[    2.720237]  bus_add_driver+0xec/0x218
[    2.720243]  driver_register+0x64/0x138
[    2.720251]  __platform_driver_register+0x30/0x48
[    2.720259]  rp1_pio_driver_init+0x28/0xff8 [rp1_pio]
[    2.720275]  do_one_initcall+0x60/0x2c0
[    2.720281]  do_init_module+0x5c/0x210
[    2.720288]  load_module+0x1ee8/0x2000
[    2.720295]  __do_sys_init_module+0x198/0x1d8
[    2.720301]  __arm64_sys_init_module+0x24/0x38
[    2.720308]  invoke_syscall+0x50/0x128
[    2.720316]  el0_svc_common.constprop.0+0x48/0xf0
[    2.720323]  do_el0_svc+0x24/0x38
[    2.720330]  el0_svc+0x38/0xc8
[    2.720339]  el0t_64_sync_handler+0x120/0x130
[    2.720347]  el0t_64_sync+0x190/0x198
[    2.720354] Code: d503201f b50000e2 14000020 38401423 (39000003) 
[    2.720360] ---[ end trace 0000000000000000 ]---
[    2.721011] edt_ft5x06 11-0038: supply iovcc not found, using dummy regulator
[    2.721131] drm-rp1-dsi 1f00130000.dsi: rp1dsi_host_attach: Attach DSI device name=tc358762 channel=0 lanes=1 format=0 flags=0x815 hs_rate=0 lp_rate=0
[    2.722612] [drm] Initialized drm-rp1-dsi 1.0.0 0 for 1f00130000.dsi on minor 0
[    2.723934] fbcon_init: detected unhandled fb_set_par error, error code -16
[    2.723940] Console: switching to colour frame buffer device 100x30
[    2.724699] drm-rp1-dsi 1f00130000.dsi: [drm] fb0: drm-rp1-dsidrmf frame buffer device
[    2.727537] drm-rp1-dsi 1f00130000.dsi: rp1dsi_bind succeeded

```

spockfish, 2025-01-10:
Thanks!

I'll test this on my unit and report back.

spockfish, 2025-01-11:
This seems resolved with the latest build:

```
Jan 10 23:38:39 ropieee5 kernel: [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 2
Jan 10 23:38:39 ropieee5 kernel: vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
Jan 10 23:38:39 ropieee5 kernel: rp1-firmware: probe of rp1_firmware failed with error -110
Jan 10 23:38:39 ropieee5 kernel: rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
Jan 10 23:38:39 ropieee5 kernel: rp1-pio: probe of 1f00178000.pio failed with error -2
Jan 10 23:38:39 ropieee5 kernel: Console: switching to colour frame buffer device 100x30
Jan 10 23:38:39 ropieee5 kernel: drm-rp1-dsi 1f00130000.dsi: [drm] fb0: drm-rp1-dsidrmf frame buffer device

```
Thanks @pelwell !

closed 2025-01-11 by spockfish