raspberrypi/linux #6604: Latest 6.6 does not compile native for Pi Zero 2 W, or boot up when cross-compiled

scottea12, 2025-01-11:
### Describe the bug

When trying to compile the 6.6 default kernel nativity on pi zero 2 w, the ssh session locks up, so not sure what if it is just the ssh connection that disconnects or that the build actually locks up.  When cross-compiling and installed on the drive, the pi does not boot up.  Installing the 6.6.51 (download from the rpi imager, it works just fine.  I am not an expert, so don't have many details.

### Steps to reproduce the behaviour

Cross-compile on ubuntu, copy to ssd... doesn't boot.

Steps I followed:
        sudo apt install git bc bison flex libssl-dev make libc6-dev libncurses5-dev crossbuild-essential-arm64
        git clone --depth=1 --branch=rpi-6.6.y https://github.com/raspberrypi/linux
		    6.6.51+rpt-rpi-v8
        cd linux
        KERNEL=kernel8
	CONFIG_LOCALVERSION="-v8-my-kernel"
        make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
        make -j6 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
        install kernel (from vm machine):
	    sudo cp /media/pi/bootfs/$KERNEL.img /media/pi/bootfs/$KERNEL-backup.img
            sudo cp arch/arm64/boot/Image /media/pi/bootfs/$KERNEL.img
            sudo cp arch/arm64/boot/dts/broadcom/*.dtb /media/pi/bootfs/
            sudo cp arch/arm64/boot/dts/overlays/*.dtb* /media/pi/bootfs/overlays/
            sudo cp arch/arm64/boot/dts/overlays/README /media/pi/bootfs/overlays/

### Device (s)

Raspberry Pi Zero 2 W

### System

This is not helpful, this if from the default install:
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2
$ vcgencmd version
Aug 30 2024 19:19:11 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)
$ uname -a
Linux inVPend 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

Dr-Drone, 2025-01-11:
Add or modify the following line, if you miss this you’d have to connect sd card and edit file in root folder, your pi will keep restarting otherwise: kernel= kernel8.img add in /boot/config.txt

lategoodbye, 2025-01-11:
bcm2711_defconfig is for Rpi 4 and CM4. Please try bcmrpi3_defconfig

scottea12, 2025-01-11:
dr-drone:

Sorry, I don't understand what line you are talking about adding or modifying and what the modification should be.

lategoodbye:

Trying to build arch64 for the rpi zero 2 w, according to: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#natively-build-a-kernel
bcm2711_defconfig is correct.  bcmrpi3_defconfig is for 32-bit rpi zero w and  bcm2709_defconfig is for 32-bit rpi zero 2 w.  Is that page incorrect?

pelwell, 2025-01-11:
@lategoodbye bcm2711_defconfig is suitable for use on any 64-bit-capable Pi. bcmrpi3_defconfig is an old, b community-contributed defconfig which should probably be deleted.

lategoodbye, 2025-01-11:
@scottea12 Is it possible you missed the * for the cp commands for dtb or is it just a copy & paste error?

popcornmix, 2025-01-11:
> make -j6 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs

Remove the `-j6` on a 512M Pi0 2W. It will likely run out of RAM when running 6 compilers in parallel.

scottea12, 2025-01-11:
@popcornmix - Originally I was trying to build natively and tried both 1 and 6 compilers with the same lockup result (by lockup, the ssh connection disconnects, so I cannot tell if it is a lock up or just a disconnect).  Because of that I switched to cross-compiling on an i9 machine with 64g ram.  It completes the compile fine that way.

@lategoodbye - always a good thing to check.  I am mainly copy and pasting from the build document.  The commands are correct and are copying all dtb files.  However, this made me look closer and figured out that I missed the root copy update:
sudo env PATH=$PATH make -j6 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- INSTALL_MOD_PATH=/media/pi/rootfs/ modules_install
so boot got updated and root did not.

Thank you all for your efforts to help me figure out my own stupid mistake.

closed 2025-01-11 by scottea12