raspberrypi/linux #6605: Are there any side effects to kernel page size configuration?

tuzbtla, 2025-01-11:
### Describe the bug

After I configured the kernel to 4K page table, there was an exception when opening chromium-browser.But my system still works.

### Steps to reproduce the behaviour

menuconfig changes the page size to 4K
[image]

### Device (s)

Raspberry Pi CM5

### System

```
cat /etc/rpi-issue 
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

vcgencmd version 
2024/09/23 14:02:56 
Copyright (c) 2012 Broadcom
version 26826259 (release) (embedded)

uname -a
Linux raspberrypi 6.6.69-v8-16k+ #3 SMP PREEMPT Sat Jan 11 16:48:06 CST 2025 aarch64 GNU/Linux
```

### Logs

[   47.264394] v3d 1002000000.v3d: MMU error from client PTB (1) at 0x1761000, pte invalid

pelwell, 2025-01-11:
If all you want to do is change the Pi5 page size to 4kB, use bcm2711_defconfig - it's the only difference between the two defconfigs. And if you don't need to build your own kernel for other reasons, use the prebuilt kernel8.img, which works on all 64-bit-capable Pis:
```
kernel=kernel8.img
```

tuzbtla, 2025-01-11:
> If all you want to do is change the Pi5 page size to 4kB, use bcm2711_defconfig - it's the only difference between the two defconfigs. And if you don't need to build your own kernel for other reasons, use the prebuilt kernel8.img, which works on all 64-bit-capable Pis:
> 
> ```
> kernel=kernel8.img
> ```

uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

Thanks, I used defconfig because I wanted to eliminate the effects of my own configuration. I tried kernel8.img you said and it doesn't have this log “ v3d 1002000000.v3d: MMU error from client PTB (1) at 0x1761000, pte invalid” Does that mean there's something wrong with the latest kernel?

tuzbtla, 2025-01-11:
> 如果你只想将 Pi5 页面大小更改为 4kB，请使用 bcm2711_defconfig - 这是两个 defconfig 之间的唯一区别。如果您出于其他原因不需要构建自己的内核，请使用预构建的 kernel8.img，它适用于所有支持 64 位的 Pis：
> 
> ```
> kernel=kernel8.img
> ```

uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 https://github.com/raspberrypi/linux/issues/1 SMP PREEMPT Debian 1：6.6.51-1+rpt3 （2024-10-08） aarch64 GNU/Linux

Thanks, I used defconfig because I wanted to eliminate the effects of my own configuration. I tried kernel8.img you said and it doesn't have this log v3d 1002000000.v3d: MMU error from client PTB (1) at 0x1761000, pte invalid. Does that mean there's something wrong with the latest kernel?

tuzbtla, 2025-01-11:
> If all you want to do is change the Pi5 page size to 4kB, use bcm2711_defconfig - it's the only difference between the two defconfigs. And if you don't need to build your own kernel for other reasons, use the prebuilt kernel8.img, which works on all 64-bit-capable Pis:
> 
> ```
> kernel=kernel8.img
> ```

uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 https://github.com/raspberrypi/linux/issues/1 SMP PREEMPT Debian 1：6.6.51-1+rpt3 （2024-10-08） aarch64 GNU/Linux

Thanks, I used defconfig because I wanted to eliminate the effects of my own configuration. I tried kernel8.img you said and it doesn't have this log ”v3d 1002000000.v3d: MMU error from client PTB (1) at 0x1761000, pte invalid.“ Does that mean there's something wrong with the latest kernel?

pelwell, 2025-01-13:
> I tried kernel8.img you said and it doesn't have this log ”v3d 1002000000.v3d: MMU error from client PTB (1) at 0x1761000, pte invalid.“ Does that mean there's something wrong with the latest kernel?

It's not clear which of those two kernels is "the latest" in your view, but those error message have dogged the V3D driver over the years. Whenever I/we have investigated we've put them down to prefetch/overreading by the 3D hardware when objects end right at the end of a memory page _and_ the following page is not mapped. I submitted a Mesa patch that added some padding after each object. It's possible that for newer V3D blocks more padding is required, but AFAIK these errors are still harmless.

pelwell, 2025-01-13:
And to answer the question in the issue title, have you seen this feedback issue?: https://github.com/raspberrypi/bookworm-feedback/issues/107

tuzbtla, 2025-01-13:
> It's not clear which of those two kernels is "the latest" in your view, but those error message have dogged the V3D driver over the years. Whenever I/we have investigated we've put them down to [...]

Thank you, I tried the 4K kernel of 6.6.47, sometimes it shows up, sometimes it doesn't. In the 6.6.69 kernel, he also appeared occasionally. My program was working, so I decided not to worry about it.