raspberrypi/linux #6606: Kernel 6.6 unable to trim uSD card

rsaxvc, 2025-01-11:
### Describe the bug

Cannot trim root filesystem on 6.6(https://github.com/raspberrypi/linux/pull/6602), but 6.5 works.

### Steps to reproduce the behaviour

With an SD card that supports erase operations, run `sudo fstrim /`

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
rsaxvc@raspberrypi:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f2d385517c9631f2ded876deb1115725d0c75995, stage2
rsaxvc@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.70-v7l+ #1 SMP Fri Jan 10 16:29:30 UTC 2025 armv7l GNU/Linux
```

### Logs

dmesg doesn't seem to show any differences.

```rsaxvc@raspberrypi:~ $ uname -a
Linux raspberrypi 6.5.0-rc7-v7l-rsaxvc-spifast+ #1 SMP Fri Aug 25 01:22:50 BST 2023 armv7l GNU/Linux
rsaxvc@raspberrypi:~ $ sudo fstrim /
rsaxvc@raspberrypi:~ $ lsblk -D
NAME        DISC-ALN DISC-GRAN DISC-MAX DISC-ZERO
mmcblk0            0        4M   112.6G         0
├─mmcblk0p1        0        4M   112.6G         0
└─mmcblk0p2        0        4M   112.6G         0
rsaxvc@raspberrypi:~ $ cat /sys/block/mmcblk0/queue/discard_max_hw_bytes 
120921784320
rsaxvc@raspberrypi:~ $ sudo hwinfo --disk --only /dev/mmcblk0
23: None 00.0: 10600 Disk                                       
  [Created at block.245]
  Unique ID: HAKk.Fxp0d3BezAE
  Parent ID: f4bI._NN+DZ1HDn4
  SysFS ID: /class/block/mmcblk0
  SysFS BusID: mmc0:5048
  SysFS Device Link: /devices/platform/emmc2bus/fe340000.mmc/mmc_host/mmc0/mmc0:5048
  Hardware Class: disk
  Model: "Disk"
  Driver: "sdhci-iproc", "mmcblk"
  Device File: /dev/mmcblk0
  Device Files: /dev/mmcblk0, /dev/disk/by-path/platform-fe340000.mmc, /dev/disk/by-id/mmc-SD64G_0x0001ce0f
  Device Number: block 179:0-179:31
  Geometry (Logical): CHS 0/4/16
  Size: 121634816 sectors a 512 bytes
  Capacity: 58 GB (62277025792 bytes)
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #5 (MMC Controller)
```

```
rsaxvc@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.70-v7l+ #1 SMP Fri Jan 10 16:29:30 UTC 2025 armv7l GNU/Linux
rsaxvc@raspberrypi:~ $ sudo fstrim /
fstrim: /: the discard operation is not supported
rsaxvc@raspberrypi:~ $ lsblk --discard
NAME        DISC-ALN DISC-GRAN DISC-MAX DISC-ZERO
mmcblk0            0        0B       0B         0
├─mmcblk0p1        0        0B       0B         0
└─mmcblk0p2        0        0B       0B         0
rsaxvc@raspberrypi:~ $ cat /sys/block/mmcblk0/queue/discard_max_hw_bytes 
0
rsaxvc@raspberrypi:~ $ sudo hwinfo --disk --only /dev/mmcblk0
23: None 00.0: 10600 Disk                                       
  [Created at block.245]
  Unique ID: HAKk.Fxp0d3BezAE
  Parent ID: f4bI._NN+DZ1HDn4
  SysFS ID: /class/block/mmcblk0
  SysFS BusID: mmc0:5048
  SysFS Device Link: /devices/platform/emmc2bus/fe340000.mmc/mmc_host/mmc0/mmc0:5048
  Hardware Class: disk
  Model: "Disk"
  Driver: "sdhci-iproc", "mmcblk"
  Device File: /dev/mmcblk0
  Device Files: /dev/mmcblk0, /dev/disk/by-path/platform-fe340000.mmc, /dev/disk/by-id/mmc-SD64G_0x0001ce0f
  Device Number: block 179:0-179:31
  Geometry (Logical): CHS 0/4/16
  Size: 121634816 sectors a 512 bytes
  Capacity: 58 GB (62277025792 bytes)
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #5 (MMC Controller)
```

P33M, 2025-01-11:
Which SD card?

rsaxvc, 2025-01-11:
MicroCenter 64GB U1.

Perhaps it was the pull-request kernel that was off? 6.6.51+rpt-rpi-v8 can fstrim rootfs with a different uSD card.

rsaxvc, 2025-01-11:
Seems to be specific to this model of SD card.

6.6.51+rpt-rpi-v8, 6.6.70-v7l+, 6.5.0-rc7-v7l-rsaxvc-spifast+ can all fstrim a Microcenter 16GB Class 10 uSD

P33M, 2025-01-13:
This is a side-effect of https://github.com/raspberrypi/linux/pull/6419/commits/d0abbb2931048f0512fae335b55ba81773b5519a - as the commit message states, it will cover cards that both have bugs in erase implementations and other cards that don't.

Please post the output of `dmesg | grep mmc0` in both 6.6.51 and 6.6.70 with the 64GB card.

rsaxvc, 2025-01-14:
```
$ uname -a
Linux raspberrypi 6.6.51-v8+ #1796 SMP PREEMPT Fri Sep 13 14:38:11 BST 2024 aarch64 GNU/Linux
$ dmesg | grep mmc0
[    1.775471] mmc0: SDHCI controller on fe340000.mmc [fe340000.mmc] using ADMA
[    1.870250] mmc0: new ultra high speed DDR50 SDXC card at address 5048
[    1.871093] mmcblk0: mmc0:5048 SD64G 58.0 GiB
[    1.874200] mmcblk0: mmc0:5048 SD64G 58.0 GiB
```

```
$ uname -a
Linux raspberrypi 6.6.70-v8+ #1 SMP PREEMPT Fri Jan 10 16:35:38 UTC 2025 aarch64 GNU/Linux
$ dmesg | grep mmc0
[    1.644152] mmc0: SDHCI controller on fe340000.mmc [fe340000.mmc] using ADMA
[    1.741926] mmc0: new ultra high speed DDR50 SDXC card at address 5048
[    1.742341] mmcblk0: mmc0:5048 SD64G 58.0 GiB
[    1.744193] mmcblk0: mmc0:5048 SD64G 58.0 GiB (quirks 0x80000000)
```

closed 2025-01-14 by P33M

rsaxvc, 2025-01-15:
Thank you for tracking that down and explaining it to me.