raspberrypi/linux #6607: [Request] Enable kernel config option CONFIG_LEDS_TRIGGER_GPIO

Malvineous, 2025-01-12:
### Describe the bug

I'd like to request the kernel option `CONFIG_LEDS_TRIGGER_GPIO` be enabled in the Pi kernel config.  It already appears to be enabled in the `bcm2835_defconfig` but none of the others.

The module allows LED devices to reflect the state of GPIO inputs without any additional userspace code running.  I would like to use it to have an LCD backlight switch on and off in response to a cabinet being opened or closed, by linking a reed switch on the cabinet door to a GPIO, and in turn coupling that to the LED device controlling the LCD's backlight.

### Steps to reproduce the behaviour

N/A

### Device (s)

Raspberry Pi 4 Mod. B

### System

N/A

pelwell, 2025-01-15:
Have you tested this feature in a custom kernel build? I'd like some assurance that it works are required. Note that bcm2835_defconfig is an upstream file, not one of ours.

Malvineous, 2025-01-15:
I haven't on the Pi as most Linux distros seem to include it as a module already, so I figured it must be pretty stable if it's so widely available.

But I can give it a go to confirm, if I can work out how to cross compile a kernel.

popcornmix, 2025-01-15:
https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

pelwell, 2025-01-15:
I can create a PR for the change, which means you'll be able to install a test build after about 40 minutes. Watch this space.

pelwell, 2025-01-15:
No I can't:
```
config LEDS_TRIGGER_GPIO
	tristate "LED GPIO Trigger"
	depends on GPIOLIB || COMPILE_TEST
	depends on BROKEN
	help
	  This allows LEDs to be controlled by gpio events. It's good
...
```
The config symbol `BROKEN` is deliberately impossible to enable. It was created as a way of forcible deprecating things without deleting them, as a way of gauging whether a feature is still required by anyone.

`LEDS_TRIGGER_GPIO` was deprecated by this commit: 8f0adae1cb1a3cf83e38dd435831baa38dd84b4c The author blames the volatility and deprecation of the global GPIO number space, and offers to write an alternative that uses Device Tree to configure the GPIO. And sure enough, he does: 4a11dbf04f31c71eb458c062129e95b7aa308464 However, you have to wait until 6.12 to get it, and the change completely removes the sysfs interface, forcing you to use an overlay (or custom DTB) to use it.

Malvineous, 2025-01-16:
Oh wow that's a good find.  I grabbed the latest tag on the releases page ([rpi-6.12.y_20241206_2](https://github.com/raspberrypi/linux/releases/tag/rpi-6.12.y_20241206_2)) and used a Docker container to cross-compile it successfully with the module enabled, so I'll see if I can work out how to do the DTB stuff to test it.

Malvineous, 2025-01-18:
Just an update on this.  I successfully compiled the 6.12 kernel, although the fbtft overlay seems broken in this version as I don't get any framebuffer coming up on an ili9341 SPI LCD.  However the GPIO triggers do work, but they are a bit of a pain to activate.

It looks like you can't just write an overlay for them, because they have to modify the LED definition in the device tree, but it doesn't seem that you can pass in any LED you want as an overlay parameter.  You could limit it to GPIO LEDs, or have a separate overlay for each of power, activity, GPIO-LEDs, etc. but that doesn't seem like a great solution.

I'm currently investigating options.  I might close this issue for now because as the module stands currently, it is probably not so practical to use.

closed 2025-01-18 by Malvineous

pelwell, 2025-01-18:
Come back if you find a solution that works for you. Help is available with the specifics of writing an overlay - visit our Device Tree forum: https://forums.raspberrypi.com/viewforum.php?f=107