raspberrypi/linux #6608: ili210x_i2c spams log with read failure

apirek, 2025-01-12:
### Describe the bug

Commit e442e5c1ab6bff5b5460b4fc949beb72aaf77970 adds an entry for a touchscreen controller on Waveshare displays that does not exist on [my display](https://www.waveshare.com/7.9inch-dsi-lcd.htm) (and probably all the others except the new one).

The ili210x_i2c driver spams the log with read failures (from dmesg, truncated):
```
[    4.524126] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.524146] ili210x_i2c 10-0041: Invalid resolution reported by controller.
[    4.524152] ili210x_i2c 10-0041: Unable to cache firmware information, err: -5
[    4.524326] input: ILI210x Touchscreen as /devices/platform/soc/fe205000.i2c/i2c-22/i2c-10/10-0041/input/input6
[    4.543775] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.543793] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.566944] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.566997] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.584434] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.584451] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.603166] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.603185] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.623179] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.623194] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.643166] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.643180] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.663279] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.663295] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.683885] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.683902] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.703150] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.703175] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.723510] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.723538] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.743432] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.743450] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.763150] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.763166] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.787114] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.787131] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.807198] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.807215] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.827805] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.827818] ili210x_i2c 10-0041: Unable to get touch data: -5
[    4.847120] ili210x_i2c 10-0041: ili251x_read_reg_common failed: -5
[    4.847136] ili210x_i2c 10-0041: Unable to get touch data: -5
```

This even causes elevated CPU usage by systemd-journald.

### Steps to reproduce the behaviour

Get a kernel with commit e442e5c1ab6bff5b5460b4fc949beb72aaf77970, I used `rpi-update`.

Add the Waveshare display overlay to `/boot/firmware/config.txt`:
```
dtoverlay=vc4-kms-v3d
max_framebuffers=2
dtoverlay=vc4-kms-dsi-waveshare-panel,7_9_inch
```

Reboot

### Device (s)

Raspberry Pi CM4

### System

OS and version: 
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

Firmware version: 
Dec  5 2024 11:45:37 
Copyright (c) 2012 Broadcom
version 03554ca336a03ace164f36755144e0d8c060062d (clean) (release) (start)

Kernel version: 
Linux raspberrypi 6.6.70-v8+ #1837 SMP PREEMPT Fri Jan 10 12:02:51 GMT 2025 aarch64 GNU/Linux

### Additional context

Raspberry Pi OS does not package that kernel version yet but Arch Linux ARM does.

pelwell, 2025-01-12:
Any thoughts, @waveshare?

6by9, 2025-01-13:
Quickest fix would be to only enable it in the overlay for that one panel

```
index 46feb1434d85..9973ac0ce389 100644
--- a/arch/arm/boot/dts/overlays/vc4-kms-dsi-waveshare-panel-overlay.dts
+++ b/arch/arm/boot/dts/overlays/vc4-kms-dsi-waveshare-panel-overlay.dts
@@ -55,6 +55,7 @@ touch: goodix@14 {
                        touch2: ilitek@41 {
                                compatible = "ilitek,ili251x";
                                reg = <0x41>;
+                               status = "disabled";
                        };
                };
        };
@@ -125,8 +126,10 @@ __overrides__ {
                                   <&touch>, "touchscreen-inverted-x?",
                                   <&touch>, "touchscreen-inverted-y?";
                8_8_inch = <&panel>, "compatible=waveshare,8.8inch-panel";
-               13_3_inch_4lane = <&panel>, "compatible=waveshare,13.3inch-4lane-panel";
-               13_3_inch_2lane = <&panel>, "compatible=waveshare,13.3inch-2lane-panel";
+               13_3_inch_4lane = <&panel>, "compatible=waveshare,13.3inch-4lane-panel",
+                                 <&touch2>, "status=okay";
+               13_3_inch_2lane = <&panel>, "compatible=waveshare,13.3inch-2lane-panel",
+                                 <&touch2>, "status=okay";
                i2c1 = <&i2c_frag>, "target:0=",<&i2c1>,
                       <0>, "-3-4+5";
                disable_touch = <&touch>, "status=disabled";
```
The Goodix driver already does an I2C read in the probe function (`goodix_i2c_test`), so there's no need to disable that touch controller for the new display.

DrBest3787, 2025-01-18:
Just Write a Script, which deletes all Duplikates

6by9, 2025-01-20:
Fixed by https://github.com/raspberrypi/linux/pull/6619

6by9, 2025-01-20:
#6619 is now merged.
Please retest with the latest kernel and reopen if issues still remain (I've tested with my Waveshare 10inch panel and get no log spamming).

closed 2025-01-20 by 6by9

apirek, 2025-01-21:
I can confirm it works properly with my display, thank you.