raspberrypi/linux #6616: fs/namespace.c:1245 cleanup_mnt+0x108/0x138 error with 6.6.72-v8+

satmandu, 2025-01-19:
### Describe the bug

I'm seeing this error spamming my dmesg:
```
[  721.834475] ------------[ cut here ]------------
[  721.834480] WARNING: CPU: 1 PID: 3350 at fs/namespace.c:1245 cleanup_mnt+0x108/0x138
[  721.834489] Modules linked in: rfcomm nf_conntrack_netlink veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bridge stp llc xfrm_user xfrm_algo xt_addrtype nft_compat nf_tables nfnetlink wireguard libchacha20poly1305 chacha_neon poly1305_neon ip6_udp_tunnel udp_tunnel libcurve25519_generic libchacha overlay algif_hash algif_skcipher af_alg bnep binfmt_misc vc4 aes_ce_blk aes_ce_cipher ghash_ce gf128mul sha2_ce spidev zfs(PO) brcmfmac_wcc hci_uart btbcm sha256_arm64 bluetooth sha1_ce snd_soc_hdmi_codec drm_display_helper ecdh_generic cec brcmfmac rpivid_hevc(C) brcmutil drm_dma_helper pisp_be v4l2_mem2mem drm_kms_helper ecc spl(O) cfg80211 libaes videobuf2_dma_contig raspberrypi_hwmon snd_soc_core videobuf2_memops videobuf2_v4l2 snd_compress videodev snd_pcm_dmaengine spi_bcm2835 i2c_brcmstb snd_pcm v3d snd_timer snd rfkill videobuf2_common gpu_sched gpio_keys pwm_fan drm_shmem_helper mc rp1_pio rp1_mailbox raspberrypi_gpiomem rp1 rp1_adc nvmem_rmem sg uio_pdrv_genirq uio
[  721.834554]  tcp_bbr fuse drm drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[  721.834563] CPU: 1 PID: 3350 Comm: buildkitd Tainted: P        WC O       6.6.72-v8+ #1839
[  721.834566] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[  721.834567] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  721.834569] pc : cleanup_mnt+0x108/0x138
[  721.834572] lr : cleanup_mnt+0x20/0x138
[  721.834574] sp : ffffffc084d1bcc0
[  721.834575] x29: ffffffc084d1bcc0 x28: ffffff8023823d80 x27: 0000000000000000
[  721.834578] x26: 0000000000000000 x25: 0000000045585401 x24: ffffff8023825860
[  721.834580] x23: 0000000020001000 x22: ffffffd085870f78 x21: ffffff80238245fc
[  721.834583] x20: ffffff8099090280 x19: 0000000000000000 x18: 000000072dc83730
[  721.834585] x17: 000000000000001b x16: ffffffd0842dbe20 x15: 0000cce6bdbb50e2
[  721.834588] x14: ffffffd084d5ca90 x13: ffffffd085025158 x12: ffffffd08564d998
[  721.834590] x11: 0000000000000000 x10: ffffffffffffffff x9 : ffffffd084385fd0
[  721.834593] x8 : ffffffc084d1bc80 x7 : 0000000000000001 x6 : ffffffd0856df5b0
[  721.834595] x5 : 0000000000000064 x4 : ffffffaffad28000 x3 : fffffffdeffeb728
[  721.834598] x2 : 0000000000000005 x1 : 0000000000000004 x0 : 00000000ffffe41b
[  721.834601] Call trace:
[  721.834602]  cleanup_mnt+0x108/0x138
[  721.834605]  __cleanup_mnt+0x1c/0x30
[  721.834607]  task_work_run+0x78/0xe0
[  721.834610]  do_notify_resume+0x218/0xf88
[  721.834614]  el0_svc+0xc8/0xd8
[  721.834617]  el0t_64_sync_handler+0x13c/0x158
[  721.834619]  el0t_64_sync+0x190/0x198
[  721.834621] ---[ end trace 0000000000000000 ]---
```

### Steps to reproduce the behaviour

I do have two disks connected via USB that are a zfs 2.3.0 mirror.

(Zfs was built via some variant of doing `dget -u --build http://deb.debian.org/debian/pool/contrib/z/zfs-linux/zfs-linux_2.3.0-1~exp1.dsc` after installing the relevant zfs dependencies, after which I installed the resultant non-dbg `.deb` packages.)

This happens at every boot, and is new behavior with Kernel 6.6.72. (I'm using the v8+ kernel on here since I have a self-hosted armv7l github  runner on this system as well.)

### Device (s)

Raspberry Pi 5

### System

```bash
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2
$ vcgencmd version
2025/01/14 00:16:48
Copyright (c) 2012 Broadcom
version 0451f142 (release) (embedded)
$ uname -a
Linux rpi51 6.6.72-v8+ #1839 SMP PREEMPT Fri Jan 17 16:49:11 GMT 2025 aarch64 GNU/Linux
```

### Logs

[6.6.72-dmesg.txt](https://github.com/user-attachments/files/18470373/6.6.72.txt)

satmandu, 2025-01-19:
This isn't related to ZFS, as I'm seeing this on a second Raspberry Pi 5 without ZFS installed. Docker is also installed on this host.

dmesg: 

[rpi52-6.6.72.dmesg.txt](https://github.com/user-attachments/files/18470858/rpi52-6.6.72.dmesg.txt)

pelwell, 2025-01-20:
Have you tried rpi-6.12.y yet? We'll be switching over to it as our master/stable kernel imminently, and you can install a trial build with `sudo rpi-update next`.

satmandu, 2025-01-20:
I will try that next ...

6by9, 2025-01-20:
The comment before that WARN is
```
	/*
	 * The warning here probably indicates that somebody messed
	 * up a mnt_want/drop_write() pair.  If this happens, the
	 * filesystem was probably unable to make r/w->r/o transitions.
	 * The locking used to deal with mnt_count decrement provides barriers,
	 * so mnt_get_writers() below is safe.
	 */
	WARN_ON(mnt_get_writers(mnt));
```
Those `mnt_want/drop_write()` calls are made by the specific file system implementation in use, so any diagnostics will need to know which fs is likely to have the issue.
If 6.12 doesn't fix things, please provide the output from `mount`.

satmandu, 2025-01-20:
6.12 does appear to make this warning go away!

closed 2025-01-20 by satmandu