raspberrypi/linux #6622: Enable BPF Type Format (BTF)`CONFIG_DEBUG_INFO_BTF`

javierhonduco, 2025-01-21:
### Describe the bug

BTF is necessary for running modern [BPF](https://ebpf.io/) applications, enabling it would allow for greater observability and debuggability of software running on Raspberry Pis. Without it, many BPF applications won't run.

### Steps to reproduce the behaviour

Run any modern BPF application, such as [lightswitch](https://github.com/javierhonduco/lightswitch) on any Raspberry Pi running Raspberry Pi OS. I only tested in on a Pi 4 though.

### Device (s)

Raspberry Pi 4 Mod. B

### System

(does not apply)

### Logs

(does not apply)

### Additional context

libbpf fails to load programs and [logs](https://github.com/libbpf/libbpf/blob/324f3c3846d99c8a1e1384a55591f893f0ae5de4/src/btf.c#L5216):

> kernel BTF is missing at /sys/kernel/btf/vmlinux, was CONFIG_DEBUG_INFO_BTF enabled?

pelwell, 2025-01-22:
If you'd tried to build the kernel yourself you would have discovered that you can't just enable DEBUG_INFO_BTF - it depends on DEBUG_INFO being enabled for the whole kernel build, which it isn't. The kconfig file says:
```
	prompt "Debug information"
	depends on DEBUG_KERNEL
	help
	  Selecting something other than "None" results in a kernel image
	  that will include debugging info resulting in a larger kernel image.
	  This adds debug symbols to the kernel and modules (gcc -g), and
	  is needed if you intend to use kernel crashdump or binary object
	  tools like crash, kgdb, LKCD, gdb, etc on the kernel.

	  Choose which version of DWARF debug info to emit. If unsure,
	  select "Toolchain default".

config DEBUG_INFO_NONE
	bool "Disable debug information"
	help
	  Do not build the kernel with debugging information, which will
	  result in a faster and smaller build.
```
1. Just selecting DEBUG_INFO_DWARF_TOOLCHAIN_DEFAULT=y increased the 64-bit kernel size from 32,449,104 bytes to 279,913,152 bytes (yes, that's an order of magnitude larger).
2. Selecting DEBUG_INFO_REDUCED=y dropped that to 102,210,280 bytes, so only 3 times larger, but that's still enormous. Unfortunately, DEBUG_INFO_BTF is explicitly incompatible with DEBUG_INFO_REDUCED.
3. I tried building with DEBUG_INFO_BTF enabled, and it failed because pahole is not available, which seems to be a tool for generating the BTF type info.

I'll leave this issue open for comments, but I'm pretty sure the answer is going to be a no.

javierhonduco, 2025-01-22:
Hi, yes, this was a rough proposal without going into the exact config changes to see if it would be something you would consider and to understand any historical context. 

The specific flags I have experimented with:

```
CONFIG_DEBUG_INFO=y
CONFIG_DEBUG_INFO_DWARF_TOOLCHAIN_DEFAULT=y
CONFIG_DEBUG_INFO_COMPRESSED_ZLIB=y
CONFIG_DEBUG_INFO_BTF=y
```

Which in my box resulted in a ~5MB increase. This seems very low, even with compression, so perhaps I did something wrong. Another possibility would be to [extract the debug info](https://sourceware.org/gdb/current/onlinedocs/gdb.html/Separate-Debug-Files.html) from the final `Image`. I think it would be very positive to do this and upload it to a [debuginfod](https://developers.redhat.com/blog/2019/10/14/introducing-debuginfod-the-elfutils-debuginfo-server) server to make Rasberry Pi OS debuggable and more amenable to profiling using DWARF info (rather than plain kallsyms which don't have line information). I guess that would be another issue. A quick Google search surfaced this question on the forums: https://forums.raspberrypi.com/viewtopic.php?t=359436.

To make these changes I've added the new flags to `./arch/arm64/configs/bcm2711_defconfig`, and then ran:

```shell
$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs -j$(nproc --all)
```

Hoernchen, 2025-04-03:
I ended up here because all the network tools like pwru and retis need this, but I do unterstand that a 300mb+ kernel is not a great default after rebuilding it. But then again, given the abundance of bpf based tools, would not be worth it to offer a separate btf kernel package?

lightsabit, 2026-09-23:
Another data point, from the Kubernetes side: I run Cilium on a small mixed cluster that includes a Pi 5, and 1.20 flat out doesn't come up on it. It loads the socket-LB programs even with socket LB turned off, those need BTF, so the agent just sits there retrying "no BTF found for kernel version" every ten seconds. 1.19 was fine on the exact same kernel because it skipped the socket-LB programs when they were disabled. Rolled the whole cluster back earlier this week, and this is now the one thing holding up our Cilium and Kubernetes upgrades.

The external BTF file approach is great for tools you launch yourself, but a daemon like cilium-agent is just going to read /sys/kernel/btf/vmlinux, and as far as I can tell there's no way to point it somewhere else — so I don't see that path helping here.

javierhonduco, 2026-09-24:
@lightsabit I won't be able to make the code changes since I don't use any of these packages, but I would suggest to propose this change in https://github.com/cilium/ebpf. See [around these lines](https://github.com/cilium/ebpf/blob/f788cf14bd26765c2023b273de26008c869a732f/btf/kernel.go#L62) for where the custom path could be added. Then it would be a matter of wiring things in Cilium ot make it work.

As a quick and dirty hack to not have to wait until this is all in motion, you could maybe write the BTF file generated off-band to [one of the inspected paths](https://github.com/cilium/ebpf/blob/f788cf14bd26765c2023b273de26008c869a732f/btf/kernel.go#L119-L127). Let me know if you end up using https://github.com/javierhonduco/raspberrypi-kernel-btf for this, haven't tried on a Raspberry Pi 5 yet!

lightsabit, 2026-10-03:
Worked, thanks. Pi 5 here (6.12.75+rpt-rpi-2712, Raspberry Pi OS),
Cilium 1.19.1 → 1.20.2 clean. That was the version looping "no BTF
found" every 10 seconds before.

I patched your builder in two spots to get an exact build: a
KERNEL_COMMIT input (git clone --branch won't take a SHA, and I needed
the exact rpi-6.12.y snapshot my shipped kernel was built from, not
the branch tip) and KERNEL_CONFIG_FILE to build with the Pi's actual
/boot config instead of bcm2712_defconfig. raspios ships a tuned
config, and struct layouts follow it.

The part that took longest: the agent runs in a container, so the
blob in host /boot is invisible until you add a read-only /boot
hostPath to the cilium-agent pod. jangroth's writeup covers this
(https://github.com/jangroth/homekube/issues/51).

Also a correction from my September comment: I said the agent only
reads /sys/kernel/btf/vmlinux. Wrong: findVMLinux's fallback list is
the whole trick. Should have read the loader before posting that.

Blob came out at 4.8 MB, build takes ~25 min on free Actions runners.
FWIW innovara's Fedora Pi kernel measured in-kernel BTF at ~6 MB of
image size (27→33 MB), so a separate package wouldn't be crazy.

smiggiddy, 2026-10-29:
Ubuntu made the same change for its Pi kernel (linux-raspi, Noble) in July 2024: "[Config] raspi: Set DEBUG_INFO_BTF=y":

https://lists.snapcraft.io/archives/kernel-team/2024-July/152459.html

ilmax, 2026-11-12:
Any chance of getting an additional OS version with BTF enabled? 
That would make allow avoiding the image size increase and unblock people who depends on BTF being present