raspberrypi/linux #6624: Kernel 6.13.0  AARCH64 Freezes Up

Dark-Sky, 2025-01-22:
### Describe the bug

The 6.13.0 aarch64 kernel freezes up at different stages from the login screen onwards. It is most prevelent after opening up firefox or chromium trying to play a video after a few seconds. The screen will lock up and the audio goes into a short loop. 

### Steps to reproduce the behaviour

Follow the official RPi guide to compile and install the kernel. Kernel compiled at d53a52016796784104ce07c2fdaad0d020a01977.

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 5

### System

Manjaro ARM
XFCE
Linux jellyfin 6.13.0-v8-16k #1 SMP PREEMPT Wed Jan 22 01:33:02 CST 2025 aarch64 GNU/Linux

[ray@jellyfin ~]$ sudo vcgencmd version
2025/01/14 00:16:48 
Copyright (c) 2012 Broadcom
version 0451f142 (release) (embedded)

### Logs

```
dmesg log
---------

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.13.0-v8-16k (ray@jellyfin) (aarch64-unknown-linux-gnu-gcc (GCC) 14.1.1 20240507, GNU ld (GNU Binutils) 2.42.0) #1 SMP PREEMPT Wed Jan 22 01:33:02 CST 2025
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000e000000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000e000000..0x000000002dffffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd164e0..0x000000003fd16517 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000003fffffff] (1024MB)
[    0.000000] Faking node 1 at [mem 0x0000000040000000-0x000000007fffffff] (1024MB)
[    0.000000] Faking node 2 at [mem 0x0000000080000000-0x00000000bfffffff] (1024MB)
[    0.000000] Faking node 3 at [mem 0x00000000c0000000-0x00000000ffffffff] (1024MB)
[    0.000000] Faking node 4 at [mem 0x0000000100000000-0x000000013fffffff] (1024MB)
[    0.000000] Faking node 5 at [mem 0x0000000140000000-0x000000017fffffff] (1024MB)
[    0.000000] Faking node 6 at [mem 0x0000000180000000-0x00000001bfffffff] (1024MB)
[    0.000000] Faking node 7 at [mem 0x00000001c0000000-0x00000001ffffffff] (1024MB)
[    0.000000] NUMA: Initialized distance table, cnt=8
[    0.000000] NODE_DATA(0) allocated [mem 0x3f7fd200-0x3f7fffff]
[    0.000000] NODE_DATA(1) allocated [mem 0x7fffd200-0x7fffffff]
[    0.000000] NODE_DATA(2) allocated [mem 0xbfffd200-0xbfffffff]
[    0.000000] NODE_DATA(3) allocated [mem 0xffffd200-0xffffffff]
[    0.000000] NODE_DATA(4) allocated [mem 0x13fffd200-0x13fffffff]
[    0.000000] NODE_DATA(5) allocated [mem 0x17fffd200-0x17fffffff]
[    0.000000] NODE_DATA(6) allocated [mem 0x1bfffd200-0x1bfffffff]
[    0.000000] NODE_DATA(7) allocated [mem 0x1ffb91200-0x1ffb93fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   1: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   node   2: [mem 0x0000000080000000-0x00000000bfffffff]
[    0.000000]   node   3: [mem 0x00000000c0000000-0x00000000ffffffff]
[    0.000000]   node   4: [mem 0x0000000100000000-0x000000013fffffff]
[... 625 lines cut ...]
    LATEST: Tue Jan 14 12:16:48 AM UTC 2025 (1736813808)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)

```

### Additional context

It is almost impossible to get any errors reported when it freezes. journalctl -f and dmesg -w does nor produce anything.

I did get his one time when opening firefox in a terminal but most of the time I see no errors:

```
Crash Annotation GraphicsCriticalError: |[C0][GFX1-]: Couldn't sanitize GL_RENDERER 
"V3D 7.1.7.0" (t=34.0883) [GFX1-]: Couldn't sanitize GL_RENDERER "V3D 7.1.7.0"
```

Another guy in our forums testing reported getting this:

```
Jan 21 11:02:19 bloodmoon-pc kernel: vc4-drm axi:gpu: [drm] *ERROR* Failed to get ddc i2c adapter by node
Jan 21 11:02:19 bloodmoon-pc kernel: vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
Jan 21 11:02:19 bloodmoon-pc kernel: rpi-gpiomem 107d510700.gpiomem: window base 0x107d510700 size 0x00000020
Jan 21 11:02:19 bloodmoon-pc kernel: rpi-gpiomem 107d504100.gpiomem: initialised 1 regions as /dev/gpiomem3
Jan 21 11:02:19 bloodmoon-pc kernel: rpi-gpiomem 107d504100.gpiomem: window base 0x107d504100 size 0x00000020
```

pelwell, 2025-01-22:
A build of https://github.com/raspberrypi/linux/commit/d53a52016796784104ce07c2fdaad0d020a01977 on a fresh install of RPi OS is running happily for me with either the regular v8.

No - spoke too soon. Playing a music video in YouTube while updating to the 16k page build led to:
```
 [  309.731943] Unable to handle kernel NULL pointer dereference at virtual address 00000000000000c0
[  309.740781] Mem abort info:
[  309.743578]   ESR = 0x0000000096000005
[  309.747334]   EC = 0x25: DABT (current EL), IL = 32 bits
[  309.752662]   SET = 0, FnV = 0
[  309.755719]   EA = 0, S1PTW = 0
[  309.758864]   FSC = 0x05: level 1 translation fault
[  309.763755] Data abort info:
[  309.766637]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[  309.772138]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[  309.777203]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[  309.782529] user pgtable: 4k pages, 39-bit VAs, pgdp=000000004a405000
[  309.788992] [00000000000000c0] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  309.797727] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
[  309.804015] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device algif_hash algif_skcipher af_alg bnep binfmt_misc vc4 snd_soc_hdmi_codec drm_display_helper cec drm_client_lib brcmfmac_wcc spidev drm_dma_helper brcmfmac rpivid_hevc(C) hci_uart pisp_be aes_ce_blk snd_soc_core btbcm aes_ce_cipher brcmutil snd_compress v4l2_mem2mem ghash_ce bluetooth snd_pcm_dmaengine gf128mul sha2_ce videobuf2_dma_contig snd_pcm ecdh_generic cfg80211 videobuf2_memops ecc sha256_arm64 videobuf2_v4l2 snd_timer sha1_ce rfkill v3d videodev libaes snd sha1_generic gpu_sched videobuf2_common drm_shmem_helper mc raspberrypi_hwmon spi_bcm2835 i2c_brcmstb gpio_keys drm_kms_helper rp1_pio rp1 raspberrypi_gpiomem rp1_adc rp1_mailbox nvmem_rmem uio_pdrv_genirq uio drm i2c_dev drm_panel_orientation_quirks ledtrig_pattern backlight dm_mod fuse ip_tables x_tables ipv6
[  309.880285] CPU: 0 UID: 1000 PID: 2334 Comm: chromium Tainted: G         C         6.13.0-v8+ #18
[  309.889193] Tainted: [C]=CRAP
[  309.892163] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[  309.898013] pstate: 404000c9 (nZcv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  309.904999] pc : v3d_irq+0xe8/0x2e0 [v3d]
[  309.909029] lr : __handle_irq_event_percpu+0x60/0x228
[  309.914098] sp : ffffffc080003ea0
[  309.917417] x29: ffffffc080003ea0 x28: ffffff81c0caa100 x27: 0000000000000000
[  309.924579] x26: ffffffd08dd81d7b x25: ffffff81c0caa100 x24: ffffff8002b0ca00
[  309.931740] x23: 00000000000000a7 x22: 00000000000000a7 x21: 0000000000000000
[  309.938902] x20: ffffff80417a0000 x19: 0000000000000002 x18: 0000000000000000
[  309.946064] x17: ffffffafadfb0000 x16: ffffffc080000000 x15: 0000000000000000
[  309.953225] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[  309.960386] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffffd08c527eb0
[  309.967547] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[  309.974709] x5 : ffffffd08c4100d0 x4 : ffffffafadfb0000 x3 : ffffffc080003f70
[  309.981870] x2 : ffffffc0829b8058 x1 : 0000000000000000 x0 : 0000000000000000
[  309.989031] Call trace:
[  309.991478]  v3d_irq+0xe8/0x2e0 [v3d] (P)
[  309.995503]  __handle_irq_event_percpu+0x60/0x228
[  310.000223]  handle_irq_event+0x54/0xb8
[  310.004071]  handle_fasteoi_irq+0xac/0x240
[  310.008180]  generic_handle_domain_irq+0x34/0x58
[  310.012811]  gic_handle_irq+0x48/0xd8
[  310.016483]  call_on_irq_stack+0x24/0x58
[  310.020414]  do_interrupt_handler+0x88/0x98
[  310.024608]  el0_interrupt+0x44/0xc0
[  310.028191]  __el0_irq_handler_common+0x18/0x28
[  310.032733]  el0t_64_irq_handler+0x10/0x20
[  310.036839]  el0t_64_irq+0x198/0x1a0
[  310.040423] Code: 94002934 360ffc33 f9442a80 52800001 (f9406015)
[  310.046536] ---[ end trace 0000000000000000 ]---
[  310.051166] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[  310.058064] SMP: stopping secondary CPUs
[  310.061997] Kernel Offset: 0x100c400000 from 0xffffffc080000000
[  310.067934] PHYS_OFFSET: 0x0
[  310.070817] CPU features: 0x100,00000170,00901250,0200720b
[  310.076319] Memory Limit: none
[  310.079380] ---[ end Kernel panic - not syncing: Oops: Fatal exception in interrupt ]---
```

pelwell, 2025-01-22:
And another, this time trying to exit full screen:
```
[  466.310099] Unable to handle kernel NULL pointer dereference at virtual address 00000000000000c0
[  466.318928] Mem abort info:
[  466.321723]   ESR = 0x0000000096000005
[  466.325479]   EC = 0x25: DABT (current EL), IL = 32 bits
[  466.330807]   SET = 0, FnV = 0
[  466.333864]   EA = 0, S1PTW = 0
[  466.337010]   FSC = 0x05: level 1 translation fault
[  466.341900] Data abort info:
[  466.344783]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[  466.350285]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[  466.355350]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[  466.360677] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000089772000
[  466.367140] [00000000000000c0] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  466.375875] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
[  466.382163] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device algif_hash algif_skcipher af_alg bnep binfmt_misc vc4 snd_soc_hdmi_codec drm_display_helper cec brcmfmac_wcc spidev rpivid_hevc(C) drm_client_lib brcmfmac hci_uart drm_dma_helper pisp_be btbcm brcmutil snd_soc_core aes_ce_blk v4l2_mem2mem bluetooth aes_ce_cipher snd_compress videobuf2_dma_contig ghash_ce cfg80211 gf128mul snd_pcm_dmaengine videobuf2_memops ecdh_generic sha2_ce ecc videobuf2_v4l2 snd_pcm v3d sha256_arm64 rfkill videodev snd_timer sha1_ce libaes gpu_sched snd videobuf2_common sha1_generic drm_shmem_helper mc rp1_pio drm_kms_helper raspberrypi_hwmon spi_bcm2835 gpio_keys i2c_brcmstb rp1 raspberrypi_gpiomem rp1_mailbox rp1_adc nvmem_rmem uio_pdrv_genirq uio i2c_dev drm ledtrig_pattern drm_panel_orientation_quirks backlight fuse dm_mod ip_tables x_tables ipv6
[  466.458429] CPU: 0 UID: 1000 PID: 2008 Comm: chromium Tainted: G         C         6.13.0-v8+ #18
[  466.467336] Tainted: [C]=CRAP
[  466.470306] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[  466.476157] pstate: 404000c9 (nZcv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  466.483143] pc : v3d_irq+0x118/0x2e0 [v3d]
[  466.487258] lr : __handle_irq_event_percpu+0x60/0x228
[  466.492327] sp : ffffffc080003ea0
[  466.495646] x29: ffffffc080003ea0 x28: ffffff80c0c94200 x27: 0000000000000000
[  466.502807] x26: ffffffd08dd81d7b x25: ffffff80c0c94200 x24: ffffff8003bdc200
[  466.509969] x23: 0000000000000001 x22: 00000000000000a7 x21: 0000000000000000
[  466.517130] x20: ffffff8041bb0000 x19: 0000000000000001 x18: 0000000000000000
[  466.524291] x17: ffffffafadfb0000 x16: ffffffc080000000 x15: 0000000000000000
[  466.531452] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[  466.538613] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffffd08c527eb0
[  466.545777] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[  466.552941] x5 : ffffffd08c4100d0 x4 : ffffffafadfb0000 x3 : ffffffc080003f70
[  466.560102] x2 : ffffffc0829e8058 x1 : 0000000000000001 x0 : 0000000000000000
[  466.567263] Call trace:
[  466.569711]  v3d_irq+0x118/0x2e0 [v3d] (P)
[  466.573826]  __handle_irq_event_percpu+0x60/0x228
[  466.578546]  handle_irq_event+0x54/0xb8
[  466.582391]  handle_fasteoi_irq+0xac/0x240
[  466.586498]  generic_handle_domain_irq+0x34/0x58
[  466.591128]  gic_handle_irq+0x48/0xd8
[  466.594798]  call_on_irq_stack+0x24/0x58
[  466.598730]  do_interrupt_handler+0x88/0x98
[  466.602923]  el0_interrupt+0x44/0xc0
[  466.606508]  __el0_irq_handler_common+0x18/0x28
[  466.611050]  el0t_64_irq_handler+0x10/0x20
[  466.615156]  el0t_64_irq+0x198/0x1a0
[  466.618740] Code: 52800035 3607faf3 f9442e80 52800021 (f9406018)
[  466.624853] ---[ end trace 0000000000000000 ]---
[  466.629483] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[  466.636384] SMP: stopping secondary CPUs
[  466.640320] Kernel Offset: 0x100c400000 from 0xffffffc080000000
[  466.646259] PHYS_OFFSET: 0x0
[  466.649141] CPU features: 0x100,00000170,00901250,0200720b
[  466.654644] Memory Limit: none
[  466.657706] ---[ end Kernel panic - not syncing: Oops: Fatal exception in interrupt ]---
```

Dark-Sky, 2025-01-22:
Yep. It seems to happen at random times.

pelwell, 2025-01-22:
Any ideas, @mairacanal?

Dark-Sky, 2025-01-22:
I did revert https://github.com/raspberrypi/linux/commit/b9acfd8823ce64007caa0667c05c3fd209ab8651 yesterday but it did not fix it.

pelwell, 2025-01-22:
The immediate cause of the crash is a NULL `render_job` here: https://github.com/raspberrypi/linux/blob/rpi-6.13.y/drivers/gpu/drm/v3d/v3d_irq.c#L119

https://github.com/raspberrypi/linux/pull/6625 adds a guard against that, logging a WARN when it happens. It also guards bin_job and csd_job, just in case.

After the warning, the GUI freezes, but serial and ssh sessions survive, which might make it easier to check the logs, or at least just sync the disc.

mairacanal, 2025-01-22:
@pelwell I'll start debugging this issue now, hopefully I'll bring a solution in a couple of hours

pelwell, 2025-01-22:
That's great, but this is the bleeding edge kernel, so the stakes are bit lower.

By the way, I just saw a NULL bin_job.

mairacanal, 2025-01-22:
The crash is caused by e4b5ccd392b9 ("drm/v3d: Ensure job pointer is set to NULL after job completion"). Therefore, be careful as this patch is also in the stable branches (such as 6.6). I mainly tested unload/load for that patch and basic rendering applications.

The fix is already available [in this branch](https://github.com/mairacanal/linux-rpi/tree/v3d/downstream/null-fix). Could you test it and confirm that it fixes the issue for you? 

I'm playing several YouTube videos for testing and it looks okay here.

If it solves the issue for you, let me know and I'll proceed to upstream the patch (your R-b will be welcomed) and send a PR to this repo.

pelwell, 2025-01-22:
Thanks, Maira - it's looking good so far. I've cherry-picked the fix as https://github.com/raspberrypi/linux/pull/6626.

pelwell, 2025-01-22:
Do you happen to know why one of the KUnit tests is failing? https://github.com/raspberrypi/linux/actions/runs/12909243870/job/35996480799#step:4:21

pelwell, 2025-01-22:
Maira's patch is now merged to rpi-6.13.y. Although the offending commit has been merged to the upstream stable branch, it doesn't yet appear to have been back-ported to any of the LTS branches, so I think we're safe for now.

mairacanal, 2025-01-22:
> Do you happen to know why one of the KUnit tests is failing? https://github.com/raspberrypi/linux/actions/runs/12909243870/job/35996480799#step:4:21

Phil, I didn't manage to reproduce this error locally. Could you reproduce it locally? Usually, "Error: 51] [ERROR] Test: <missing>: Could not find any KTAP output. Did any KUnit tests run?" indicates some output error, not a test failure specifically.

Dark-Sky, 2025-01-22:
Seems to be working good. Thanks everyone!!

pelwell, 2025-01-22:
> Could you reproduce it locally?

You're right - it works when I run the tests here.

Something else is different between the arm and arm64 tests. The arm log begins:
```
Run ./tools/testing/kunit/kunit.py run \
  ./tools/testing/kunit/kunit.py run \
      --kunitconfig=drivers/gpu/drm/vc4/tests \
      --cross_compile=arm-linux-gnueabihf- --arch=arm
  shell: /usr/bin/bash -e {0}
[13:[2](https://github.com/raspberrypi/linux/actions/runs/12909243870/job/35996480472#step:4:2)9:24] Configuring KUnit Kernel ...
```
Compare that to the arm64 log:
```
Run ./tools/testing/kunit/kunit.py run \
  ./tools/testing/kunit/kunit.py run \
      --kunitconfig=drivers/gpu/drm/vc4/tests \
      --cross_compile=aarch64-linux-gnu- --arch=arm64
  shell: /usr/bin/bash -e {0}
stty: 'standard input': Inappropriate ioctl for device
[13:[2](https://github.com/raspberrypi/linux/actions/runs/12909243870/job/35996480799#step:4:2)9:15] Configuring KUnit Kernel ...
```
I don't know why ARCH=arm64 would be different in this regard.

mairacanal, 2025-01-23:
I sent the patch to the dri-devel mailing list for review [1].

[1] https://lore.kernel.org/dri-devel/20250123012403.20447-1-mcanal@igalia.com/T/

pelwell, 2025-01-28:
As Maira's patch is in all the affected branches and linux-next, I'm closing this.

closed 2025-01-28 by pelwell

mairacanal, 2025-02-10:
The fix landed on all stable branches from 5.1 to 6.13 [1].

[1] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-6.13.y&id=3059e7aaa280daea57bb069fbc65225e1bb95014

pelwell, 2025-02-10:
And we have/will drop the downstream versions as necessary. Thanks, Maíra.