raspberrypi/linux #6630: KBUILD_MODNAME=rp1 system_heap

ShorTie8, 2025-01-24:
### Describe the bug

Trying to compile the kernel and it's erroring out with
`  if nm drivers/mfd/rp1.o 2>/dev/null | grep -q ' __export_symbol_'; then /usr/bin/gcc -E -D__GENKSYMS__ -Wp,-MMD,drivers/mfd/.rp1.o.d  -nostdinc -I./arch/arm64/include -I./arch/arm64/include/generated  -I./include -I./arch/arm64/include/uapi -I./arch/arm64/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -DKASAN_SHADOW_SCALE_SHIFT= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mgeneral-regs-only -Wno-psabi -mabi=lp64 -fno-asynchronous-unwind-tables -fno-unwind-tables -Wa,-march=armv8.3-a -DARM64_ASM_ARCH='"armv8.3-a"' -DKASAN_SHADOW_SCALE_SHIFT= -fno-delete-null-pointer-checks -O2 --param=allow-store-data-races=0 -fstack-protector-strong -fno-omit-frame-pointer -fno-optimize-sibling-calls -pg -falign-functions=4 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wvla -Wno-pointer-sign -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare    -DKBUILD_MODFILE='"drivers/mfd/rp1"' -DKBUILD_BASENAME='"rp1"' -DKBUILD_MODNAME='"rp1"' -D__KBUILD_MODNAME=kmod_rp1 drivers/mfd/rp1.c | scripts/genksyms/genksyms   -r /dev/null >> drivers/mfd/.rp1.o.cmd; fi
drivers/dma-buf/heaps/system_heap.c:57:40: error: initializer element is not constant
 static unsigned int module_max_order = orders[0];
                                        ^~~~~~
make[6]: *** [scripts/Makefile.build:243: drivers/dma-buf/heaps/system_heap.o] Error 1
make[5]: *** [scripts/Makefile.build:480: drivers/dma-buf/heaps] Error 2
make[4]: *** [scripts/Makefile.build:480: drivers/dma-buf] Error 2
make[4]: *** Waiting for unfinished jobs....
`

### Steps to reproduce the behaviour

Just doing a normal kernel build

### Device (s)

Raspberry Pi 5

### System

 cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4
pi5a build # vcgencmd version
2024/11/12 16:10:44
Copyright (c) 2012 Broadcom
version 4b019946 (release) (embedded)
pi5a build # uname -a
Linux pi5a 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

pelwell, 2025-01-24:
For build problems you need to say which HEAD commit, which ARCH and defconfig, and which toolchain you are using.

However, I can see that the line in question looks correct - the value is contant, but a less smart compiler might not realise that.

ShorTie8, 2025-01-24:
> but a less smart compiler might not realise that.

That might be it because this is gcc -6.5.0
Doing a standard "make -C linux bcm2712_defconfig" in a older chroot with gcc-6.5.0

closed 2025-01-25 by ShorTie8