raspberrypi/linux #6631: Critical Bug in Unfreeze Propagation

umarsync, 2025-01-26:
### Describe the bug

In the `cgroup_propagate_frozen` function, during unfreezing (when `frozen` is false), the code subtracts `desc` (which increments for each unfrozen parent) from each ancestor's `nr_frozen_descendants`. This results in over-subtraction, leading to negative counts and incorrect frozen states.

`kernel/cgroup/freezer.c`

**Example Scenario**:
- Hierarchy: A -> B -> C
- All cgroups (A, B, C) are frozen.
- Unfreeze C:
- **Current Code**: Subtracts `desc=1` from B's count (correctly setting it to 0). Then subtracts `desc=2` from A's count, reducing it from 2 to 0. If A has another frozen child D, this incorrectly sets A's count to 0 instead of 1 (since D is still frozen).

### Additional Considerations
- **Negative Count Prevention**: Added `WARN_ON_ONCE` to catch underflows, though proper decrement by 1 should prevent negatives.
- **Propagation Continuation**: The loop continues upwards until all affected ancestors are updated.

The identified bug leads to incorrect frozen state propagation, particularly in hierarchical cgroups. The proposed fix corrects the decrement logic during unfreezing, ensuring accurate `nr_frozen_descendants` counts and proper state transitions. This issue warrants a kernel patch to prevent system instability from incorrect cgroup freezing.

### Steps to reproduce the behaviour

*Test Case:* Incorrect unfreezing of parent cgroup when one of multiple frozen descendants is unfrozen.

1. **Setup Hierarchy**:
```bash
# Create cgroups
mkdir /sys/fs/cgroup/A
mkdir /sys/fs/cgroup/A/B
mkdir /sys/fs/cgroup/A/D
mkdir /sys/fs/cgroup/A/B/C
```

2. **Freeze Leaf Cgroups**:
```bash
# Freeze C and D (A and B will auto-freeze via propagation)
echo 1 > /sys/fs/cgroup/A/B/C/cgroup.freeze
echo 1 > /sys/fs/cgroup/A/D/cgroup.freeze
```

3. **Verify All Cgroups Are Frozen**:
```bash
cat /sys/fs/cgroup/A/cgroup.freeze # Should show 1 (frozen)
cat /sys/fs/cgroup/A/B/cgroup.freeze # Should show 1 (frozen)
cat /sys/fs/cgroup/A/B/C/cgroup.freeze # Should show 1 (frozen)
cat /sys/fs/cgroup/A/D/cgroup.freeze # Should show 1 (frozen)
```

4. **Unfreeze Cgroup C**:
```bash
echo 0 > /sys/fs/cgroup/A/B/C/cgroup.freeze
```

5. **Observe Incorrect Behavior**:
```bash
cat /sys/fs/cgroup/A/cgroup.freeze # ❌ Shows 0 (unfrozen) **BUG**
cat /sys/fs/cgroup/A/D/cgroup.freeze # Still shows 1 (frozen)
```

**Expected Behavior**:
- `A` should **remain frozen** because its descendant `D` is still frozen.
- `B` should unfreeze (correctly) because its only descendant `C` was unfrozen.

**Actual Behavior (with Bug)**:
- `A` incorrectly unfreezes due to over-decrement in `nr_frozen_descendants`.

**Debugging Tips**:
1. Check kernel logs for warnings like:
```bash
WARN_ON_ONCE(cgrp->freezer.nr_frozen_descendants < 0)
```
2. Use `ftrace` to monitor `cgroup_propagate_frozen` calls:
```bash
echo 1 > /sys/kernel/debug/tracing/events/cgroup/enable
cat /sys/kernel/debug/tracing/trace_pipe
```

### Device (s)

Other

### System

`rpi-6.6.y`

popcornmix, 2025-01-30:
`kernel/cgroup/freezer.c` is unchanged from upstream.
I think you'll have better luck reporting this issue upstream - it is unlikely to be a Pi specific issue,
and there may be developers of this actual code who can comment.

Your detailed report should be helpful to the maintainers.

See: https://docs.kernel.org/admin-guide/reporting-issues.html

pelwell, 2025-01-30:
Yes - see dialog at https://github.com/raspberrypi/linux/pull/6632.