raspberrypi/linux #6642: Pi5 failed to contact RP1 firmware

llllllllqq, 2025-02-01:
### Describe the bug

dmesg says:

...
`rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
[    3.957885] rp1-pio: probe of 1f00178000.pio failed with error -2`
...

Although nothing strange occurred. 
The OS still works fine without reboot.

### Steps to reproduce the behaviour

It randomly occurs after `sudo reboot --halt` and manually press the on-board power button to poweron. (with default eeprom config)

It also randomly occurs when use raspi-config to change power off behavior to "full power off" and then reboot/poweroff.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-11-19                     Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

2025/01/22 00:16:51
Copyright (c) 2012 Broadcom                           version a7753063 (release) (embedded)

Linux rpi5 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

### Logs

[1.txt](https://github.com/user-attachments/files/18627781/1.txt)

### Additional context

Can it be safely ignored?

pelwell, 2025-02-03:
That's interesting - it looks like a dependency/ordering issue, and not one I've seen before:
```
[    3.922665] rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
[    3.922671] rp1-pio: probe of 1f00178000.pio failed with error -2
...
[    3.932284] rp1-firmware rp1_firmware: RP1 Firmware version eb39cfd516f8c90628aa9d91f52370aade5d0a55
```
I get:
```
[   13.445656] rp1-firmware rp1_firmware: RP1 Firmware version eb39cfd516f8c90628aa9d91f52370aade5d0a55
[   13.550428] rp1-pio 1f00178000.pio: Created instance as pio0
```
The code for communicating with the firmware exists in the firmware module, so I thought that would enforce an ordering. Some investigation is required.

llllllllqq, 2025-02-03:
> That's interesting - it looks like a dependency/ordering issue, and not one I've seen before:
>
> ```
> [    3.922665] rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
> [    3.922671] [...]

Thanks for your reply.
Here is my dmesg when the rp1&pio works fine.
I hope it can help to investigate this issue.

[normal.txt](https://github.com/user-attachments/files/18640719/normal.txt)

pelwell, 2025-02-03:
Yes - in the working case the order is reversed:
```
[    3.955929] rp1-firmware rp1_firmware: RP1 Firmware version eb39cfd516f8c90628aa9d91f52370aade5d0a55
[    3.961971] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    3.967397] rp1-pio 1f00178000.pio: Created instance as pio0
```

aBUGSworstnightmare-rpi, 2025-02-03:
seems quite random. Seen it frequently on reboots (running 'sudo reboot' on CLI)

mayo228, 2025-02-03:
is there a way to fix this? I keep getting this error and I can't boot the OS, should I be worried that I will loose data on my raspberry?

pelwell, 2025-02-03:
I don't understand why this affecting some people while I can't reproduce it without hacking the driver, but regardless, https://github.com/raspberrypi/linux/pull/6645 should fix the issue. After about 40 minutes you should be able to install a trial kernel with `sudo rpi-update pulls/6645`, after first backing up any important data.

pelwell, 2025-02-03:
And no, you shouldn't worry - it would only affect you if you wanted to use piolib to drive the GPIOs.

aBUGSworstnightmare-rpi, 2025-02-03:
I have no idea if and how below is related but I just wanted to state that I'm not making use of polio for accessing GPIO's.

```
[   13.678330] ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
[   15.374753] Bluetooth: RFCOMM TTY layer initialized
[   15.374765] Bluetooth: RFCOMM socket layer initialized
[   15.374772] Bluetooth: RFCOMM ver 1.11
[   27.660651] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[   27.768681] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[   27.872656] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[   27.976661] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[   29.636663] brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
[   29.636875] brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
[   29.637196] brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
[   29.637400] brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
[   29.637594] brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
[   29.637791] brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
```

llllllllqq, 2025-02-04:
It seems after applying `sudo rpi-update pulls/6645`, this issue disappeared. (I tested about 10 times reboot/halt/suddenly poweroff). Thanks for your work! If anything strange found, I'll post here.

BTW, can I safely stay at this firmware and wait for the next stable release offered by apt? Or do I need a `sudo apt install --reinstall raspi-firmware` to restore first before the next stable release?

pelwell, 2025-02-04:
You should be fine with this kernel until the next proper release - I'm not aware of any new issues.

randomMesh, 2025-02-07:
I can confirm the error happens on my PI 5 too:

```
[    0.692347] rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
[    0.692370] rp1-pio: probe of 1f00178000.pio failed with error -2
```

I am running the latest firmware:
```
BOOTLOADER: up to date
   CURRENT: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
    LATEST: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

helpful-emu57, 2025-02-09:
A week ago I also started receiving these errors during bootup:  

rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware

At the same time I also started getting protocol errors for the wireless mouse and wireless keyboard
logitech-hidpp-device    -    hidpp_root_get_  protocol_version: received protocol error 0x08
(Although - both devices continued to work appropriately.)

Changing to wired devices there are no mouse or keyboard errors but still have the rp1-pio error

I have performed the "sudo rpi-update pulls/6645" and no change was observed.

pelwell, 2025-02-09:
What does `rpi-eeprom-update` show?

nickongit, 2025-02-09:
How do you do "sudo rpi-update pulls/6645" if you can't boot into it?

My new pi-500 seems to be bricked with this error. I just tried to install moonlight-qt and performed the "sudo reboot" and can't get it to boot up.

pelwell, 2025-02-10:
> My new pi-500 seems to be bricked with this error. I just tried to install moonlight-qt and performed the "sudo reboot" and can't get it to boot up.

From the detailed description of your problem I'm going to guess that:
1. You had a working Pi 500.
2. You installed some new software.
3. It no longer boots.
4. The only error message you can see on screen is the one about the RP1 firmware.

As I've explained above, the only effect of that error is to make PIO unavailable - it will not cause the boot to fail. I suggest you edit `cmdline.txt` in another computer, replacing `quiet` with `ignore_loglevel`, then try again. You should see much more output, and that may make it clearer what is going wrong.

helpful-emu57, 2025-02-12:
RE: follow-up 3 days ago - Boot RP1 Firmware Error
	
      Excerpt From Boot Log
      [    0.858841] rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
      [    0.858866] rp1-pio: probe of 1f00178000.pio failed with error -2
      
      Is location 1f00178000 reserved for particular hardware? 
      What is an "rp1-pio error -2" error ?

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

RE: Wireless Device Errors
      The protocol error is gone when using original wireless mouse, keyboard and dongle 
      (would similar make/model with different dongle cause error?)

pelwell, 2025-02-12:
I think it might be best if the error was demoted to a warning.

helpful-emu57, 2025-02-20:
I re-installed rpi-eeprom [Wed 12 Feb 2025 10:51:52 AM UTC (1739357512)]
and then [rpi-update pulls/6645] again and this time it worked.

hillphantom, 2025-02-21:
Hello just want to report I have the same error and unfortunately even after re-install of rpi-eeprom it is still kicking the error.

Erwin40, 2025-02-21:
Hello.

How did you do that?

popcornmix, 2025-02-21:
> Hello just want to report I have the same error and unfortunately even after re-install of rpi-eeprom it is still kicking the error.
 
just to be sure it installed okay, report the output of `vcgencmd bootloader_version`.

helpful-emu57, 2025-02-21:
In the Terminal Run "rpi-eeprom-update" and report output

`rpi-eeprom-update`

hillphantom, 2025-02-21:
Hi All, thanks for all the help, the SD port is just bad and failed on my
brand new Pi5 16GB and I am hearing lots of folks are experiencing the SD
port failing on the 16GB version. I can get it to boot and run just fine
via USB or SSD. Sadly I just got one of the bad ones I think. One bad one
out of 18 pie's in my life isn't bad. I just wanted to report it as I think
this is actually an issue the foundation should be aware of and fix if they
can before they churn out more 16GB models.

pelwell, 2025-02-23:
1. What makes you say that the SD "port" is bad - what are the symptoms?
2. Can you show us where you've encountered these other folks experiencing a broken SD card slot on a 16GB Pi 5 (or any Pi 5, for that matter)? It's not something I've heard about.

hillphantom, 2025-02-23:
Hi! So 1. It gives the same error and its not fixable by Eprom updates, but
it will boot from USB or PCIe via Nvme. Once booted it can not see any SD
cards or anything in that port ( tried multiple cards on boot and as
secondary drives
           2. If you google this issue you will see the folks that are
reporting this error mostly in the last two weeks right around the time
most of us were getting our pi 16gbs. Also, it was working, then one day I
watched it all of sudden shut down
               and since that day, no go on the SD slot. I have had one go
bad before and had to soldier on another but, 2 out of what 100 or so isn't
bad.

pelwell, 2025-02-23:
I did Google, and I didn't find anything.

hillphantom, 2025-02-23:
It literally is littered all of these current threads mate. But yea, just
reporting it, not trying to make it a crusade.....Report, Adjust, move on,
I am not going to google things for you.

JamesH65, 2025-02-24:
I just Googled "Pi 5 broken SD card slot" and got one relevant hit, where a user had broken the slot whilst putting the PI 5 into a case. So I'm not finding these reports either. Is there a better search term (or search engine) to use?

hillphantom, 2025-02-24:
Sure... as I said not widespread but certainly being reported in the
threads if you are wanting to dig as I did...."pi busybox"  "failed to
connect rp 1 firmware"  - I just popped on a hat and booted from nvme, no
biggie, don't really need the SD personally others might. Cheers.

pelwell, 2025-02-24:
>  I am hearing lots of folks are experiencing the SD
port failing on the 16GB version.

> It literally is littered all of these current threads mate.

> as I said not widespread

Spot the contradiction.

hillphantom, 2025-02-24:
Let me ask you a question bucko, when you see litter, outside of a rubbish
bin or a dump, is it "all over"? If you care to actually look, you might
find it, in spots, as you do with litter in most cases.  This is my first
and last time reporting something so well done boys. Have a wonderful day!

hillphantom, 2025-02-24:
And just to be clear, I also "hear" about lots of litter in spots
too.....kick rocks, was just trying to report some trees  through the
forest ole boy

pelwell, 2025-02-24:
Off-topic discussion deleted.

ozapft, 2025-03-02:
I also had the error message “failed to contact rp1 firmware“.
So I installed a new sd card as a test. This error message didn't appear with this new sd card, even though it uses the same kernel. When comparing the /boot/firmware directory, I noticed that the initramfs files on the older card were  ~17 MB in size and the two new ones were 12.7 MB. After copying the two files from the new sd card to the old one, the error disappeared.

moodeaudio, 2025-03-08:
I'm also seeing this error sometimes, but only on Pi5 with NVMe (non-boot) drive attached.
```
RPiOS: 12.9 Bookworm 64-bit | Linux: 6.6.74 64-bit (built with Pi-Gen)
ii  raspi-firmware   1:1.20250305-1  all  Raspberry Pi family GPU firmware and bootloaders
```
I have 2 such Pi5's and one symptom is that sometimes reboot or power off does not complete and the Pi never reaches the halt state (red LED). Two other Pi5's w/o NVMe drive are fine.

___Pi5 with NVMe drive___
```
pi@moode9:~ $ dmesg | grep rp1
[    0.805353] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    0.823933] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    2.284970] rp1 0001:01:00.0: bar0 len 0x4000, start 0x1f00410000, end 0x1f00413fff, flags, 0x40200
[    2.294054] rp1 0001:01:00.0: bar1 len 0x400000, start 0x1f00000000, end 0x1f003fffff, flags, 0x40200
[    2.303317] rp1 0001:01:00.0: enabling device (0000 -> 0002)
[    2.309638] rp1 0001:01:00.0: chip_id 0x20001927
[    2.318624] genirq: irq_chip rp1_irq_chip did not update eff. affinity mask of irq 106
[    5.742889] rp1-firmware: probe of rp1_firmware failed with error -110
[    5.749497] rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
[    5.749505] rp1-pio: probe of 1f00178000.pio failed with error -2
pi@moode9:~ $
```
___Pi5 w/o NVMe drive___
```
pi@moode:~ $ dmesg | grep rp1
[    1.015138] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    1.033523] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    2.137633] rp1 0000:01:00.0: bar0 len 0x4000, start 0x1f00410000, end 0x1f00413fff, flags, 0x40200
[    2.146719] rp1 0000:01:00.0: bar1 len 0x400000, start 0x1f00000000, end 0x1f003fffff, flags, 0x40200
[    2.155981] rp1 0000:01:00.0: enabling device (0000 -> 0002)
[    2.162270] rp1 0000:01:00.0: chip_id 0x20001927
[    2.170959] genirq: irq_chip rp1_irq_chip did not update eff. affinity mask of irq 100
[    5.009561] rp1-firmware rp1_firmware: RP1 Firmware version eb39cfd516f8c90628aa9d91f52370aade5d0a55
[    5.237119] rp1-pio 1f00178000.pio: Created instance as pio0
pi@moode:~ $
```

PCIe controller: Pimoroni NVMe Base for Raspberry Pi 5
https://www.adafruit.com/product/5845?gad_source=1&gbraid=0AAAAADx9JvT2Uc1scGbd8pfLnNHlqrIyp&gclid=EAIaIQobChMIz8PE47P6iwMVo15_AB2ZfhpaEAQYASABEgKY__D_BwE

NVMe stick: Crucial P3 1TB PCIe Gen3 3D NAND NVMe M.2 SSD
https://www.bhphotovideo.com/c/product/1710645-REG/crucial_ct1000p3ssd8_1tb_p3_nvme_pcie.html/?ap=y&ap=y&smp=y&smp=y&store=420&lsft=BI%3A5451&gad_source=1&gbraid=0AAAAAD7yMh0CnuDG9lQosL1qx1hNsryh6&gclid=EAIaIQobChMIiaSZwLT6iwMVHDStBh17LgH0EAQYASABEgLYhvD_BwE

hillphantom, 2025-03-08:
> [    0.805353] ***@***.***/rp1: Fixed dependency cycle(s) with ***@***.***/rp1
> [    0.823933] ***@***.***/rp1: Fixed dependency cycle(s) with ***@***.***/rp1
> [    1.015138] ***@***.***/rp1: Fixed dependency cycle(s) with ***@***.***/rp1
> [    1.033523] ***@***.***/rp1: Fixed dependency cycle(s) with ***@***.***/rp1

Yup the owner of this group won't believe you and will belittle you so be
careful

Mike Vanderpool
ValleyApps <http://www.valleyapps.com>

pelwell, 2025-03-08:
It's possible that the presence of the NVME drive changes the timing of the boot procedure. Between 6.6.74 and 6.6.75 the RP1 PIO and firmware drivers were improved to cope with arbitrary startup order. I can't tell whether or not the kernel you have includes that patch, but you can ensure that it does by running `sudo rpi-update 1705add`.

moodeaudio, 2025-03-08:
Makes sense. The kernel we use is the official release package so not patched.

theepicflyer, 2025-03-25:
I'm getting the same error too but maybe a little a different. One installation that I had piolib working on stopped working last week or so. A fresh OS install (with the updated eeprom) on the same Pi does work consistently though.

### Not working install:
```
[    1.032795] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    1.051186] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    2.163309] rp1 0000:01:00.0: bar0 len 0x4000, start 0x1f00410000, end 0x1f00413fff, flags, 0x40200
[    2.172393] rp1 0000:01:00.0: bar1 len 0x400000, start 0x1f00000000, end 0x1f003fffff, flags, 0x40200
[    2.181659] rp1 0000:01:00.0: enabling device (0000 -> 0002)
[    2.187947] rp1 0000:01:00.0: chip_id 0x20001927
[    2.196677] genirq: irq_chip rp1_irq_chip did not update eff. affinity mask of irq 100
[    2.957783] rp1-pio 1f00178000.pio: error -ENOENT: failed to contact RP1 firmware
[    2.965577] rp1-pio: probe of 1f00178000.pio failed with error -2
[    5.861008] rp1-firmware rp1_firmware: RP1 Firmware version eb39cfd516f8c90628aa9d91f52370aade5d0a55
```

[notworking.txt](https://github.com/user-attachments/files/19448915/notworking.txt)

### Working install:
```
[    1.032788] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    1.051175] /axi/pcie@120000/rp1: Fixed dependency cycle(s) with /axi/pcie@120000/rp1
[    2.167264] rp1 0000:01:00.0: bar0 len 0x4000, start 0x1f00410000, end 0x1f00413fff, flags, 0x40200
[    2.176349] rp1 0000:01:00.0: bar1 len 0x400000, start 0x1f00000000, end 0x1f003fffff, flags, 0x40200
[    2.185614] rp1 0000:01:00.0: enabling device (0000 -> 0002)
[    2.191893] rp1 0000:01:00.0: chip_id 0x20001927
[    2.200581] genirq: irq_chip rp1_irq_chip did not update eff. affinity mask of irq 100
[    4.932575] rp1-firmware rp1_firmware: RP1 Firmware version eb39cfd516f8c90628aa9d91f52370aade5d0a55
[    4.960712] rp1-pio 1f00178000.pio: Created instance as pio0
```

[working.txt](https://github.com/user-attachments/files/19448925/working.txt)

Let me know if there is any other info I can provide that will help.

pelwell, 2025-03-25:
The important changes to resolve this failure went into https://github.com/raspberrypi/linux/pull/6645. Although it was merged to 6.6.74, that happened a week after your kernel was built (2025-02-04 vs 2025-01-27).

hillphantom, 2025-03-25:
Careful of the arrogant pompous moderator when reporting issues on this group he's not really here to help you but would rather tell you your lying or try to belittle you for reporting an issue. He's a full jack a lope