raspberrypi/linux #6648: Update kernel: break functionality: NFS exec: permission denied

danergo, 2025-02-04:
On kernel v5.10.17+ I had an NFS server.

Config is simple. A small embedded system boots up from its own storage, then mounts an NFS share, then pivot_roots into it.

IMPORTANT! This worked on Kernel v5.10.17+ (Debian 10).

I'm trying to build a new NFS server, now with much fresher Debian 12 and Kernel v6.12.1-v8+.

So the ONLY change was made to NFS server. Client is kept the SAME.

Mounting NFS is working from both old and new NFS server perfectly.

However, after running pivot_root, I get nothing but "Permission denied" messages onto every single command you can imagine.

(Only "cd" and "pwd" work).

Must mention that same method, same mount options, same pivot_root with old server works as I wish: no permission denied messages, and my rootfs is served from NFS (by the old server).

How can I detect what's causing this? And moreover how can I fix this? I took hours+++ to investigate this but nothing helped (no_root_squash, etc, everything is setup correctly, at least to let the old NFS server function properly).

pelwell, 2025-02-04:
This is probably a configuration issue rather than a kernel issue, but it's definitely nothing to do with the firmware.

Why don't you break the problem down and try looking at the mounted FS _before_ running `pivot_root`?  You should be able to run commands off it using explicit paths.

danergo, 2025-02-04:
All configs are the same (copied to new from old).

Source rootfs on both new and old server are the same (copied from old to new).

After mounting both new and old on the same client, mount options are also the same.

Nfs configuration (although it seems a different version, but) is also the same (same version, same protocols).

Only nfs_acl seems to be different: not presented in old, working config.

But it seems a kernel thing, hence my question here.

pelwell, 2025-02-04:
It's not necessarily true that "same" means "correct" - things change, especially when the kernel you are switching from is 4 years old.

6by9, 2025-02-04:
"Linux 6.10 Is Disabling NFS v2 Client Support By Default" - https://www.phoronix.com/news/Linux-6.10-NFS-Client

Seeing as 6.12.1 isn't the latest kernel on the rpi-6.12.y branch (6.12.12 is current), are you building your own kernel with a defconfig other than the ones we provide? Our defconfig does appear to still have NFS_V2 set to y.
Is it even our kernel tree that your building, or the mainline one?

You've also switched from a 32bit kernel (-v7+) to a 64bit kernel (-v8+).

What are your mount options?

danergo, 2025-02-04:
I didn't customised the kernels (neither on old nor on the new NFS server).

Export options: rw, sync, no_subtree_check, no_root_squash.

pelwell, 2025-02-04:
1. So the obvious(?) next question is... where did you get your kernel? Which distribution are you running, and which version?

2. Which versions of NFS does it support?
    ```
    sudo modprobe configs
    zgrep NFS_V /proc/config.gz
    ```
3. Which version of NFS are you using?

Try to be a bit more forthcoming with information, or people are likely to lose interest.

danergo, 2025-02-04:
> 1. So the obvious(?) next question is... where did you get your kernel? Which distribution are you running, and which version?

I'm using "DietPi" which is based on Raspbian and is not altering Raspbian's kernel in any aspect.

This is true for both new and old system, but old is created long time ago.

Old:
Debian GNU/Linux 10
Kernel 5.10.17+ ##1421

New: 
Debian GNU/Linux 12
Kernel 6.12.1-v8+ ##1822

> 2. Which versions of NFS does it support?>    ```

Old:

```
# zgrep NFS_V /proc/config.gz
CONFIG_NFS_V2=y
CONFIG_NFS_V3=y
CONFIG_NFS_V3_ACL=y
CONFIG_NFS_V4=y
CONFIG_NFS_V4_1=y
CONFIG_NFS_V4_2=y
CONFIG_NFS_V4_1_IMPLEMENTATION_ID_DOMAIN="kernel.org"
# CONFIG_NFS_V4_1_MIGRATION is not set
CONFIG_NFS_V4_SECURITY_LABEL=y
# CONFIG_NFS_V4_2_READ_PLUS is not set
```

New:

```
# zgrep NFS_V /proc/config.gz
CONFIG_NFS_V2=y
CONFIG_NFS_V3=y
CONFIG_NFS_V3_ACL=y
CONFIG_NFS_V4=y
CONFIG_NFS_V4_1=y
CONFIG_NFS_V4_2=y
CONFIG_NFS_V4_1_IMPLEMENTATION_ID_DOMAIN="kernel.org"
# CONFIG_NFS_V4_1_MIGRATION is not set
CONFIG_NFS_V4_SECURITY_LABEL=y
CONFIG_NFS_V4_2_READ_PLUS=y
```

> 3. Which version of NFS are you trying to use on the client?

This particular client supports only NFS v3.

I really appreciate your thoughts and help 🙏

pelwell, 2025-02-04:
How about my first suggestion?
> Why don't you break the problem down and try looking at the mounted FS before running pivot_root? You should be able to run commands off it using explicit paths.

danergo, 2025-02-04:
@pelwell:

Sorry wanted to answer to that too:

In case I don't do "pivot_root", I can perfectly run anything from NFS paths. Both from old and new server.

Only difference is that new server denies every exec permission once I call pivot_root on that directory.

Examining mounts: both old and new NFS servers' folders are mounted with exact same options. I can post it later.

danergo, 2025-02-04:
So these are the mount details:

OLD NFS Server (execution works: before AND after pivot_root)
```
OLDIP:/nfs/common on /media/old type nfs (rw,sync,relatime,vers=3,rsize=65536,wsize=65536,namlen=255,hard,nolock,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=OLDIP,mountvers=3,mountproto=tcp,local_lock=all,addr=OLDIP)
```

NEW NFS Server (execution only works before pivot_root)
```
NEWIP:/nfs/common2 on /media/new type nfs (rw,sync,relatime,vers=3,rsize=131072,wsize=131072,namlen=255,hard,nolock,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=NEWIP,mountvers=3,mountproto=tcp,local_lock=all,addr=NEWIP)
```

As I remembered, mounts are pretty much the same, just some buffer differences are there.

Moreover, here are details of the servers too:

Old:
```
nfsstat: 1.3.3
# rpcinfo -p | grep nfs
    100003    3   tcp   2049  nfs
    100003    4   tcp   2049  nfs
    100003    3   udp   2049  nfs
```

New:
```
nfsstat: 2.6.2
#rpcinfo -p | grep nfs
    100003    3   tcp   2049  nfs
    100003    4   tcp   2049  nfs
    100227    3   tcp   2049  nfs_acl
    100003    3   udp   2049  nfs
    100227    3   udp   2049  nfs_acl
```

6by9, 2025-02-04:
```
    100227    3   tcp   2049  nfs_acl
...
    100227    3   udp   2049  nfs_acl
```
look likely, ACL being Access Control List.

Whilst CONFIG_NFS_V3_ACL was set on the old kernel, something has obviously changed to mean that it has a separate network connection active. I know no details of how ACLs will function with pivot_root.

danergo, 2025-02-04:
Yes. I tried to turn ACL off, but can't find any documentation. I guess it's compiled into kernel?

6by9, 2025-02-04:
You can disable NFS_V3_ACL in the kernel config. 

Help text for that kernel option does include
```
	  Most NFS servers don't support the Solaris NFSv3 ACL protocol
	  extension.  You can choose N here or specify the "noacl" mount
	  option to prevent your NFS client from trying to use the NFSv3
	  ACL protocol.
```

pelwell, 2025-02-04:
I'd say the `noacl` mount option sounds like the better way to go (if you can).

danergo, 2025-02-04:
> I'd say the `noacl` mount option sounds like the better way to go (if you can).

You mean this on client side?

danergo, 2025-02-04:
Tried adding `noacl` to mount options:

`mount -t nfs -o vers=3,rw,sync,nolock,noacl NEWIP:/nfs/common2`

Reported 0 errors.

Execution fails. Permission denied after pivot_root.

One more addition: everything is owned by "root". On both client and server side. If that matters.

I really wanted to avoid deepdive into this *** but it seems I'll have to disassemble the packets and browse thorougly the NFS code which I really really wanted to skip.

danergo, 2025-02-04:
Here are some more details "var/lib/nfs/etab":

OLD:
`/nfs/common OLDIP(rw,sync,wdelay,hide,nocrossmnt,secure,no_root_squash,no_all_squash,no_subtree_check,secure_locks,acl,no_pnfs,anonuid=65534,anongid=65534,sec=sys,rw,secure,no_root_squash,no_all_squash)`

NEW
`/nfs/common2 NEWIP(rw,sync,wdelay,hide,nocrossmnt,secure,no_root_squash,no_all_squash,no_subtree_check,secure_locks,acl,no_pnfs,anonuid=65534,anongid=65534,sec=sys,rw,secure,no_root_squash,no_all_squash)`

I know this is losing interest now, but just wanted to emphasize how sad am I. Even these options are exactly the same.

6by9, 2025-02-04:
Both of those still appear to have `acl` in the status line.
`mount` does have a `-v` option for verbose logging.

You've jumped nearly exactly 4 years in kernel development (5.10.17 was 17th Feb 2021), and 2 versions of Debian. Development doesn't just stand still.
If it were between two LTS kernel versions (eg 6.6 to 6.12), or 1 version of distribution, then we might have a chance. But with the window being that large, there are too many things that could have changed.

You could take a copy of your old system SD card, and then run `sudo rpi-update rpi-6.12.y` to get the 6.12 kernel. If that will still pivot_root as you want, then you know you need to look for a distribution change. If not then try `sudo rpi-update rpi-6.6.y` to get the latest 6.6 kernel and test that. Our CI builds don't go back before 6.6, but you could build them yourself.

danergo, 2025-02-04:
> `mount` does have a `-v` option for verbose logging.

Not on this device unfortunately. It's just one line: success/fail even if I try -vvvvvvvv

> You could take a copy of your old system SD card, and then run...

It's not having SD card, but I can dd the partitions.

So how can I save my current, working kernel (5.10.17+) and more important: switch back to it? 

I'm happy to test different kernel versions to see if those helps.

So, shall I update 5.10.17+ to 6.12.y? Will it work with Debian 10?

danergo, 2025-02-05:
I have checked the NFS packets with Wireshark. They are identical during mount for new and old server.

New vs old:

[image] 
vs

[image]

danergo, 2025-02-05:
Update: tried from a different machine with a bit newer kernel: **6.8.0-51-generic**: it works.

danergo, 2025-02-05:
I have tar-ed everything from working server and copied to non-working one, and now it started working the new one too.

Have no idea what was the problem, but it might be that scp-ing the rootfs was a bad idea. Now I scp-d the rootfs.tgz, and extracted on the target directly. Sorry folks for burning your time, but I really appreciated your quick responsed, and I learned valuable things (for example `modprobe configs`, I didn't know it before).

Thanks!

closed 2025-02-05 by danergo