raspberrypi/linux #6652: Enable `FONT_TER16x32` for better high-dpi support in VTs?

xofe, 2025-02-05:
### Describe the bug

Could kernel config option `FONT_TER16x32` please be enabled by default? 

It adds in a larger font (double the size of the default) and, if compiled in, [will be automatically used on high resolution displays](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=dfd19a5004eff03755967086aa04254c3d91b8ec), improving readability when needing to use a VT.

### Steps to reproduce the behaviour

N/A

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS:
```
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2
```
Firmware:
```
Nov 26 2024 12:52:28 
Copyright (c) 2012 Broadcom
version 2ae30f53898ae2f1ba77ff570a92991bedfb0398 (clean) (release) (start)
```

Kernel:
```
Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
```

popcornmix, 2025-02-05:
Our usual question when is feature is requested that may be only wanted by a small number, is what is the cost?

Simplest is memory - after the config option is added what is the change in size of kernel8.img?
Also reporting output of `free` after a boot (before and after).

Perhaps something like:
```
time for i in $(seq 1 1000); do dmesg ; done
```
run from console in a mode that uses the new font. (before and after).

Are you able to build the kernel and test this?

xofe, 2025-02-09:
Alright, I built the kernel with defaults using the `apt source linux` tarball (version 6.6.74-1+rpt1), to confirm it reproducibly built (and it did). I then rebuilt it with these additions to the config: 
```
CONFIG_FONTS=y
CONFIG_FONT_TER16x32=y
```

The only thing of note that I can see is the doubling of time for your suggested `dmesg` test - my guess here is that VT output isn’t that fast, and with the larger font (16×32 instead of 8×16), the output is effectively double the number of lines, pixel-wise.

<details>
<summary>File size results (0.4kB smaller compressed, 65kB bigger uncompressed):</summary>

```
-rwxr-xr-x 1 root root  9362120 Feb  8 01:48 kernel8.img
-rwxr-xr-x 1 root root 25592320 Feb  8 01:48 kernel8.img.uncompressed
-rwxr-xr-x 1 root root  9361761 Feb  8 05:09 kernel8withlargefont.img
-rwxr-xr-x 1 root root 25657856 Feb  8 05:09 kernel8withlargefont.img.uncompressed
```
</details>

<details>
<summary>Memory:</summary>

Default kernel, after fresh boot:
```
               total        used        free      shared  buff/cache   available
Mem:         1891116      160480     1662356        1356      124704     1730636
Swap:         524284           0      524284
```
Built with `FONT_TER16x32`, after fresh boot:
```
               total        used        free      shared  buff/cache   available
Mem:         1891052      161020     1661984        1356      124516     1730032
Swap:         524284           0      524284
```
Default kernel, after `dmesg` 1k runs:
```
               total        used        free      shared  buff/cache   available
Mem:         1891116      162272     1659428        1356      126108     1728844
Swap:         524284           0      524284
```
Built with `FONT_TER16x32`, after `dmesg` 1k runs:
```
               total        used        free      shared  buff/cache   available
Mem:         1891052      162544     1593840        1356      191440     1728508
Swap:         524284           0      524284
```
</details>

<details>
<summary><code>dmesg</code> 1k runs time:</summary>

Default kernel, using default font:
```
real    22m53.654s
user    0m8.882s
sys     22m44.921s
```
Built with `FONT_TER16x32`, using larger font:
```
real    44m12.136s
user    0m8.519s
sys     44m3.543s
```
</details>

<details>
<summary>Generated config differences:</summary>

```diff
--- a/config-6.6.74+rpt-rpi-v8
+++ b/config-6.6.74+rpt-rpi-v8_withlargefont
@@ -8867,9 +8867,20 @@ CONFIG_HAVE_GENERIC_VDSO=y
 CONFIG_GENERIC_GETTIMEOFDAY=y
 CONFIG_GENERIC_VDSO_TIME_NS=y
 CONFIG_FONT_SUPPORT=y
-# CONFIG_FONTS is not set
-CONFIG_FONT_8x8=y
+CONFIG_FONTS=y
+# CONFIG_FONT_8x8 is not set
 CONFIG_FONT_8x16=y
+# CONFIG_FONT_6x11 is not set
+# CONFIG_FONT_7x14 is not set
+# CONFIG_FONT_PEARL_8x8 is not set
+# CONFIG_FONT_ACORN_8x8 is not set
+# CONFIG_FONT_MINI_4x6 is not set
+# CONFIG_FONT_6x10 is not set
+# CONFIG_FONT_10x18 is not set
+# CONFIG_FONT_SUN8x16 is not set
+# CONFIG_FONT_SUN12x22 is not set
+CONFIG_FONT_TER16x32=y
+# CONFIG_FONT_6x8 is not set
 CONFIG_SG_POOL=y
 CONFIG_ARCH_STACKWALK=y
 CONFIG_STACKDEPOT=y
```
</details>

P33M, 2025-02-10:
The dmesg time doubling is strange - it'll be printing the same number of lines, but twice as many vertical pixels. Is console output waiting for vsync for each line?