raspberrypi/linux #6656: Kernel 6.14-rc1 No AudioI have no HDMI audio

Dark-Sky, 2025-02-05:
### Describe the bug

I have no HDMI audio with kernel 6.14-rc1 16k. I do not see any errors/failed in the logs and everything looks like it should have audio. Playing a movie with mpv the video is jerky and no audio. Vlc seems to play smooth with no sound. Youtube video seems to be smooth with no sound.

### Steps to reproduce the behaviour

Compile and install the kernel following the official RPi Guide. I have not tested the other pi devices other than the pi5 so do not know their status.

### Device (s)

Raspberry Pi 5

### System

[ray@jellyfin ~]$ uname -a
Linux jellyfin 6.14.0-rc1-v8-16k #1 SMP PREEMPT Wed Feb  5 15:37:55 CST 2025 aarch64 GNU/Linux

```
[ray@jellyfin ~]$ sudo vcgencmd version
2025/01/22 00:16:51 
Copyright (c) 2012 Broadcom
version a7753063 (release) (embedded)
```

aplay -l:

```
[ray@jellyfin ~]$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

### Logs

dmesg:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.14.0-rc1-v8-16k (ray@jellyfin) (aarch64-unknown-linux-gnu-gcc (GCC) 14.1.1 20240507, GNU ld (GNU Binutils) 2.42.0) #1 SMP PREEMPT Wed Feb  5 15:37:55 CST 2025
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000e000000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000e000000..0x000000002dffffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd164e0..0x000000003fd16517 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000003fffffff] (1024MB)
[    0.000000] Faking node 1 at [mem 0x0000000040000000-0x000000007fffffff] (1024MB)
[    0.000000] Faking node 2 at [mem 0x0000000080000000-0x00000000bfffffff] (1024MB)
[    0.000000] Faking node 3 at [mem 0x00000000c0000000-0x00000000ffffffff] (1024MB)
[    0.000000] Faking node 4 at [mem 0x0000000100000000-0x000000013fffffff] (1024MB)
[    0.000000] Faking node 5 at [mem 0x0000000140000000-0x000000017fffffff] (1024MB)
[    0.000000] Faking node 6 at [mem 0x0000000180000000-0x00000001bfffffff] (1024MB)
[    0.000000] Faking node 7 at [mem 0x00000001c0000000-0x00000001ffffffff] (1024MB)
[    0.000000] NUMA: Initialized distance table, cnt=8
[    0.000000] NODE_DATA(0) allocated [mem 0x3f7fd1c0-0x3f7fffff]
[    0.000000] NODE_DATA(1) allocated [mem 0x7fffd1c0-0x7fffffff]
[    0.000000] NODE_DATA(2) allocated [mem 0xbfffd1c0-0xbfffffff]
[    0.000000] NODE_DATA(3) allocated [mem 0xffffd1c0-0xffffffff]
[    0.000000] NODE_DATA(4) allocated [mem 0x13fffd1c0-0x13fffffff]
[    0.000000] NODE_DATA(5) allocated [mem 0x17fffd1c0-0x17fffffff]
[    0.000000] NODE_DATA(6) allocated [mem 0x1bfffd1c0-0x1bfffffff]
[... 591 lines cut ...]
[    8.741767] Bluetooth: hci0: BCM: features 0x2f
[    8.747911] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    8.754370] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    8.760950] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[   10.245868] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   10.251447] Bluetooth: BNEP filters: protocol multicast
[   10.256906] Bluetooth: BNEP socket layer initialized
[   10.264758] Bluetooth: MGMT ver 1.23
[   10.272360] NET: Registered PF_ALG protocol family
[   10.404774] loop27: detected capacity change from 0 to 8
[   10.981850] zram: Added device: zram0
[   11.001798] zram0: detected capacity change from 0 to 24759168
[   11.027460] Adding 12379568k swap on /dev/zram0.  Priority:100 extents:1 across:12379568k SS
[   11.208816] macb 1f00100000.ethernet end0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[   11.237770] macb 1f00100000.ethernet end0: configuring for phy/rgmii-id link mode
[   11.251826] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[   11.288773] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   11.837541] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.083331] NET: Registered PF_VSOCK protocol family
[   15.262222] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.270532] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   29.502417] Bluetooth: RFCOMM TTY layer initialized
[   29.507353] Bluetooth: RFCOMM socket layer initialized
[   29.512628] Bluetooth: RFCOMM ver 1.11
```

popcornmix, 2025-02-06:
Can you confirm the kernel you tested contained https://github.com/raspberrypi/linux/commit/5745a677e82b84d9e646e19d2642fdc431309694 ?
Can you confirm 6.13 kernel works?

pelwell, 2025-02-06:
I can confirm that 6.13 HDMI audio works, and that it's broken in 6.14. That code has been significantly reworked in 6.14 - see 9640f1437a88d8c617ff5523f1f9dc8c3ff29121. Unfortunately, reverting that commit isn't enough to make it work again.

Dark-Sky, 2025-02-06:
> Can you confirm the kernel you tested contained [5745a67](https://github.com/raspberrypi/linux/commit/5745a677e82b84d9e646e19d2642fdc431309694) ? Can you confirm 6.13 kernel works?

Yes that is the commit I used. Sorry for not stating.
Latest 6.13 just fine at  f33efa774d3501313aa875bd09f17c6e2d2d6836.

Dark-Sky, 2025-02-06:
> I can confirm that 6.13 HDMI audio works, and that it's broken in 6.14. That code has been significantly reworked in 6.14 - see [9640f14](https://github.com/raspberrypi/linux/commit/9640f1437a88d8c617ff5523f1f9dc8c3ff29121). Unfortunately, reverting that commit isn't enough to make it work again.

Thanks for confirming. I'm in no hurry; do what yo have to do on your end.

6by9, 2025-02-06:
Give me a moment. I had looked at an issue around those helpers and infoframes. I now just need to remember what and where!

pelwell, 2025-02-06:
I'm deep in this already. Alsa seems to be repeatedly detecting an over/underrun and restarting the stream.

pelwell, 2025-02-06:
It's looking as though the audio DMA is unpaced, which causes ALSA to report an underrun.

Ah - spot the difference - 6.13:
```
cyclic(262144, 8192) - dreq 41fa000c, dst 107c72001c
```
vs 6.14:
```
cyclic(262144, 8192) - dreq 41fa000c, dst 7c72001c
```
It's a slave DMA address translation issue.

pelwell, 2025-02-06:
Answer - they're still doing it wrong in the DMA driver:
```
		dst = c->cfg.dst_addr;
```
vs:
```
		dst = phys_to_dma(chan->device->dev, c->cfg.dst_addr);
```

6by9, 2025-02-06:
That's the one that I and others have tried to upstream a couple of times.

[Robin said that dma_map_resource](https://lkml.org/lkml/2024/2/5/1161) is the correct call, but needs https://lore.kernel.org/linux-iommu/20220610080802.11147-1-Sergey.Semin@baikalelectronics.ru/ to deal with dma-ranges.
Every time that has been sent upstream it's had an unsure response, and then nothing when answered.
https://lore.kernel.org/linux-arm-kernel/20240524182702.1317935-2-dave.stevenson@raspberrypi.com/

Upstream doesn't have a DMA channel assigned to the HDMI blocks, so no audio support.

pelwell, 2025-02-06:
I think it's more complicated/less well thought out than that.

`dma_map_resource` is rarely used. Most of the occurrences appear to be in bus-mastering devices, but there are a few (3) in DMA controllers such as the pl330. Here it is called (via a helper function) from the `.device_prep_slave_sg` and `.device_prep_dma_cyclic` methods. However, there are no `.device_unprep_*` methods, so in order to get matching calls to `dma_unmap_resource` each driver has to take care to call it before either changing the mapping (address, direction, etc.) or from the `.device_free_chan_resources` method. This mapping wasn't created in `.device_alloc_chan_resources`, and the asymmetry makes it look like a kludge.

Behind the scenes, `dma_map_resource` looks like it should call `dma_direct_map_resource` for addresses which it believes are directly addressable by the DMA controller, and thence end up (with the patch mentioned above) calling `phys_to_dma_direct`. The equivalent path in `dma_direct_unmap_resource` is a no-op. `dma_map_resource` does allow for the possibility that the slave could be behind an IOMMU, but does that ever happen?

Anyway, with the one line patch to `dma_direct_map_resource` applied and replacing the offending direct calls to `phys_to_dma` with `dma_map_resource`, it currently doesn't work.

pelwell, 2025-02-07:
> it currently doesn't work.

Correction - it does work, once you get the parameters right. I can push these patches to at least rpi-6.13.y and rpi-6.14.y.

Dark-Sky, 2025-02-10:
Just to get on record:

v6.14-rc2 (16k) at 2784deac5dad0d04bbbc350336317f3d924bd0d3: Still does not have HDMI audio

pelwell, 2025-02-11:
Sorry, Ray - that's because I merged the uncorrected version. It should be fixed in 08f6bbe2d993f.

Dark-Sky, 2025-02-11:
LOL. Thanks Phil. I will give it a go.

Dark-Sky, 2025-02-11:
Works like a champ. Thanks a lot!!

closed 2025-02-14 by popcornmix

Seegras, 2025-04-09:
I do have the same problem, on a desktop. Has this been fixed upstream?

popcornmix, 2025-04-09:
We still have the fix in downstream rpi-6.14.y and rpi-6.15.y, so it is not in stable upstream trees.
@6by9 may know if it's been suibmitted.

6by9, 2025-04-10:
No, I haven't looked at upstreaming this again. The responses dried up so rapidly on both previous occasions that it seemed like no one wanted to be involved.

Upstreams DT files have no DMA channel assigned, therefore there is no HDMI audio.