raspberrypi/linux #6659: overlayroot does not work

probonopd, 2025-02-08:
### Describe the bug

overlayroot does not work, the microSD card filesystem is still writable and no union mount is to be seen.

### Steps to reproduce the behaviour

On `2024-11-19-raspios-bookworm-armhf-lite.img.xz` using `raspi-config`, set up overlayroot and reboot.
Make some modifications in the filesystem and reboot.
The modifications are still there.

### Device (s)

Raspberry Pi 5

### System

`2024-11-19-raspios-bookworm-armhf-lite.img.xz`

pelwell, 2025-02-08:
> set up overlayroot

That's not a step, that's a bunch of steps, and they may differ between users in s significant way. 

What did _you_ do.

probonopd, 2025-02-08:
Cannot reproduce either way. Sorry for the noise.
(I could swear I had done the exact same steps the first time around, but welll...)

closed 2025-02-08 by probonopd

pelwell, 2025-02-08:
I wish they were all that easy.