raspberrypi/linux #6661: nvme0: controller is down (additional log, >1 year worth) now runs hot

graemev, 2025-02-10:
### Describe the bug

I raised this , via Debian's reportbug(1) , it got returned as "Closing as this is not a Debian system but running on a derivative."

So, I'll try to shoehorn the original report into these boxes ...I'm guessing I'll hit size limits 

This is simply an attempt to provide more logs of what is likely a 3 year old problem.

2 points of interest, 1: ran without issues for over a year (same hardware, limited use, apt-gt update on most uses)
                                2: without a power cycle (but with a reboot) produces different errors

I'd anticipate this gets merged with an existing bug (just to add the logs)

AFYI. Some reading I did around this suggests that M$ in windows 10 only used the deepest power saving mode of NVMe while the system was suspended. It appears the Rpi allows this mode while running normally (e.g. allows a latency larger than it can actually accept during normal running)

OK, it block each of these tables at  65536 characters , so I can't even fir the output of raspinfo(1) into one

### Steps to reproduce the behaviour

Subject: nvme0: controller is down (additional log, >1 year worth) now runs hot
Package: src:linux
Version: 1:6.6.62-1+rpt1
Severity: normal

Dear Maintainer,

This a Raspberry Pie 5 running bookworm for over 1 year. I have the full syslog (journal) , but points of interest:

1: The system ran from Jan 2024 to Jan 2025 without errors
2: The system started reporting errors on Jan 28 2025
3: I applied the "workaround" Jan 31 2025 ( nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_port_pm=off  )
4: Workaround "fixed" the problem, but still showed "nvme nvme0: I/O 329 (I/O Cmd) QID 3 timeout, aborting"
5: A subsequent POWER CYCLE (not just reboot) removed the timeouts
6: The system now runs at 50-60C , vs 20C beforehand

I appreciate it is likely the root cause is "non-compliant" hardware, I'm adding this log to the evidence because:

i:  The long period it ran without errors (A kernel change over that time?)
ii: The behavior WRT POWER CYCLE vs just boot, points at some NVMe activity while not running kernel but still powered.

HARDWARE:
NVMe => "fanxiang S501Q" (2TB Gen3) , case+NVMe Hat ==> "Argon NEO 5"

MESSAGE (ERRORS):
Example error (shows workaround applied):

Jan 31 17:45:57 argon kernel: nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0x10       
Jan 31 17:45:57 argon kernel: nvme nvme0: Does your device have a faulty power saving mode enabled?              
Jan 31 17:45:57 argon kernel: nvme nvme0: Try "nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_port_pm=$
Jan 31 17:45:57 argon kernel: nvme 0000:01:00.0: enabling device (0000 -> 0002)

I will likely now try a "chop" (non-binary) of these options to see which works around the problem and which makes
the system run hot ...and hope they are not the same one. I will proably not update this report with that information but can do
if it helps.

**UPDATE(done the chop nvme_core.default_ps_max_latency_us=0 , or  nvme_core.default_ps_max_latency_us=15500 avoids the problem)

(syslog @ the end)

-- Package-specific info:
** Version:
Linux version 6.6.62+rpt-rpi-2712 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25)

** Command line:
reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  numa=fake=8 system_heap.max_order=0 smsc95xx.macaddr=2C:CF:67:30:F4:D9 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=b07e1308-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles usbhid.mousepoll=0 nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_port_pm=off

** Tainted: C (1024)
 * staging driver was loaded

** Kernel log:
[... 404 lines cut ...]
### System

graeme@argon:~$ raspinfo 

(I had to dlet all the output of raspinfo(1) in order for the webpage to accept the input (all too big)

System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4

### Logs

Logs are embedded in the  reportbug(1) mail. ....I've had to remove ALL the useful logs in order to allow the web interface to accept this bug.... please provide an email address or similar where I can send the original bugreport.

### Additional context

(all removed to fir webpage)

graemev, 2025-02-10:
Subject: nvme0: controller is down (additional log, >1 year worth) now runs hot
Package: src:linux
Version: 1:6.6.62-1+rpt1
Severity: normal

Dear Maintainer,

This a Raspberry Pie 5 running bookworm for over 1 year. I have the full syslog (journal) , but points of interest:

1: The system ran from Jan 2024 to Jan 2025 without errors
2: The system started reporting errors on Jan 28 2025
3: I applied the "workaround" Jan 31 2025 ( nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_port_pm=off  )
4: Workaround "fixed" the problem, but still showed "nvme nvme0: I/O 329 (I/O Cmd) QID 3 timeout, aborting"
5: A subsequent POWER CYCLE (not just reboot) removed the timeouts
6: The system now runs at 50-60C , vs 20C beforehand

I appreciate it is likely the root cause is "non-compliant" hardware, I'm adding this log to the evidence because:

i:  The long period it ran without errors (A kernel change over that time?)
ii: The behavior WRT POWER CYCLE vs just boot, points at some NVMe activity while not running kernel but still powered.

HARDWARE:
NVMe => "fanxiang S501Q" (2TB Gen3) , case+NVMe Hat ==> "Argon NEO 5"

MESSAGE (ERRORS):
Example error (shows workaround applied):

Jan 31 17:45:57 argon kernel: nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0x10       
Jan 31 17:45:57 argon kernel: nvme nvme0: Does your device have a faulty power saving mode enabled?              
Jan 31 17:45:57 argon kernel: nvme nvme0: Try "nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_port_pm=$
Jan 31 17:45:57 argon kernel: nvme 0000:01:00.0: enabling device (0000 -> 0002)

I will likely now try a "chop" (non-binary) of these options to see which works around the problem and which makes
the system run hot ...and hope they are not the same one. I will proably not update this report with that information but can do
if it helps.

(syslog @ the end)

-- Package-specific info:
** Version:
Linux version 6.6.62+rpt-rpi-2712 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25)

** Command line:
reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  numa=fake=8 system_heap.max_order=0 smsc95xx.macaddr=2C:CF:67:30:F4:D9 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=b07e1308-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles usbhid.mousepoll=0 nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_port_pm=off

** Tainted: C (1024)
 * staging driver was loaded

** Kernel log:
[    2.752497] fuse: init (API version 7.39)
[    2.757446] systemd[1]: Finished fake-hwclock.service - Restore / save the current clock.
[    2.758137] systemd[1]: Finished kmod-static-nodes.service - Create List of Static Device Nodes.
[    2.758542] systemd[1]: modprobe@configfs.service: Deactivated successfully.
[    2.758681] systemd[1]: Finished modprobe@configfs.service - Load Kernel Module configfs.
[    2.758920] systemd[1]: modprobe@dm_mod.service: Deactivated successfully.
[    2.759027] systemd[1]: Finished modprobe@dm_mod.service - Load Kernel Module dm_mod.
[    2.759245] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    2.759350] systemd[1]: Finished modprobe@drm.service - Load Kernel Module drm.
[    2.759592] systemd[1]: modprobe@efi_pstore.service: Deactivated successfully.
[    2.759709] systemd[1]: Finished modprobe@efi_pstore.service - Load Kernel Module efi_pstore.
[    2.759962] systemd[1]: modprobe@fuse.service: Deactivated successfully.
[    2.760082] systemd[1]: Finished modprobe@fuse.service - Load Kernel Module fuse.
[    2.760358] systemd[1]: modprobe@loop.service: Deactivated successfully.
[    2.762854] systemd[1]: Finished modprobe@loop.service - Load Kernel Module loop.
[    2.764167] systemd[1]: Mounting sys-fs-fuse-connections.mount - FUSE Control File System...
[    2.765159] systemd[1]: Mounting sys-kernel-config.mount - Kernel Configuration File System...
[    2.765287] systemd[1]: systemd-repart.service - Repartition Root Disk was skipped because no trigger condition checks were met.
[    2.766661] systemd[1]: Mounted sys-fs-fuse-connections.mount - FUSE Control File System.
[    2.767022] i2c_dev: i2c /dev entries driver
[    2.774734] systemd[1]: Mounted sys-kernel-config.mount - Kernel Configuration File System.
[... 1418 lines cut ...]
Feb 05 18:51:49 argon kernel: EXT4-fs (nvme0n1p2): re-mounted fc7a1f9e-4967-4f41-a1f5-1b5927e6c5f9 r/w. Quota mode: none.
Feb 05 18:51:49 argon systemd-fsck[468]: /dev/nvme0n1p1: 390 files, 21164/130812 clusters
Feb 05 18:51:50 argon kernel: EXT4-fs (nvme0n1p3): mounted filesystem 08870205-494a-4ccd-a77b-5d0acea2400f r/w with ordered data mode. Quota mode: none.
Feb 05 18:51:50 argon systemd[1]: nvmefc-boot-connections.service - Auto-connect to subsystems on FC-NVME devices found during boot was skipped because of an unmet condition check (ConditionPathExists=/sys/class/fc/fc_udev_device/nvme_discovery).
Feb 05 18:51:57 argon systemd[1]: Starting nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot...
Feb 05 18:51:57 argon modprobe[992]: modprobe: FATAL: Module nvme-fabrics not found in directory /lib/modules/6.6.62+rpt-rpi-2712
Feb 05 18:51:57 argon systemd[1]: Failed to start nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot.
[... cut ...]
Feb 07 18:29:13 argon systemd[1]: nvmefc-boot-connections.service - Auto-connect to subsystems on FC-NVME devices found during boot was skipped because of an unmet condition check (ConditionPathExists=/sys/class/fc/fc_udev_device/nvme_discovery).
Feb 07 18:29:20 argon systemd[1]: Starting nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot...
Feb 07 18:29:20 argon modprobe[984]: modprobe: FATAL: Module nvme-fabrics not found in directory /lib/modules/6.6.62+rpt-rpi-2712
Feb 07 18:29:20 argon systemd[1]: Failed to start nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot.

graemev, 2025-02-10:
graeme@argon:~$ raspinfo 
System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4

Linux argon 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
Revision	: d04170
Serial		: 37bb44e1b6eb260d
Model		: Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

Videocore information
---------------------

2025/01/22 00:16:51 
Copyright (c) 2012 Broadcom
version a7753063 (release) (embedded)

vc_gencmd_read_response returned -1
error=1 error_msg="Command not registered"

Filesystem information
----------------------
Filesystem                1K-blocks     Used  Available Use% Mounted on
udev                        4083152        0    4083152   0% /dev
tmpfs                        825776     6112     819664   1% /run
/dev/disk/by-label/root3  131001348 10472640  113801440   9% /
tmpfs                       4128832      336    4128496   1% /dev/shm
tmpfs                          5120       48       5072   1% /run/lock
/dev/nvme0n1p1               523248    84668     438580  17% /boot/firmware
/dev/nvme0n1p3           1835260644       28 1741961624   1% /QNAP/mounts/rest1
tmpfs                        825760      192     825568   1% /run/user/1501
tmpfs                        825760      192     825568   1% /run/user/106

Filename				Type		Size		Used		Priority
/var/swap                               file		204736		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20241211
raspberrypi-sys-mods:
  Installed: 20241202
openbox:
  Installed: 3.6.1-10+rpt1
lxpanel:
  Installed: 0.10.1-2+rpt21
pcmanfm:
  Installed: 1.3.2-1+rpt29
rpd-plym-splash:
  Installed: 0.42

Networking Information
----------------------

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet x.x.x.x  netmask x.x.x.x  broadcast x.x.x.x
        inet6 y.y.y.y.y.y.y.y  prefixlen 64  scopeid 0x0<global>
        inet6 y::y.y.y.y  prefixlen 64  scopeid 0x20<link>
        ether m.m.m.m  txqueuelen 1000  (Ethernet)
        RX packets 13572  bytes 1978244 (1.8 MiB)
[... 1769 lines cut ...]
[   79.340991] nvme nvme0: Does your device have a faulty power saving mode enabled?
[   79.340994] nvme nvme0: Try "nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_port_pm=off" and report a bug
[   79.396996] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[   79.404526] nvme nvme0: 4/0/0 default/read/poll queues
[   79.410364] nvme nvme0: Ignoring bogus Namespace Identifiers

EEPROM
------

BOOTLOADER: up to date
   CURRENT: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
    LATEST: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.

graeme@argon:~$ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4
graeme@argon:~$ vcgencmd version
2025/01/22 00:16:51 
Copyright (c) 2012 Broadcom
version a7753063 (release) (embedded)
graeme@argon:~$ uname -a
Linux argon 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
graeme@argon:~$

pelwell, 2025-02-10:
When the template says "Describe the bug", you are meant to describe the bug. Much TL, so DR. Have you heard of pastebin et al?

graemev, 2025-02-10:
I was pointed here as a location to file a reportbug(1) report. The support line in the raspberry pi variant of Debian say: 

```
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

This is clearly not correct as "Debian" reject these as:

"Closing as this is not a Debian system but running on a derivative."

So the support URL and probably the mail address ins reportbug(1)  should probably be updated to point at the correct support channel .

FYI I am not raising this because I expect help , I'm trying to get the log added to the appropriate bug . My 1st guess would be the rpi maintainer will say it's not pi specific and pass it upstream ....but I could be wrong, there may well be a unique Pi defect he's aware of.

(and yes I agree, it looks a mess ... not a very good interface to submit a bug report, more like some kind of "online help & support page" ) ...even "attach a file" would be more usable

pelwell, 2025-02-10:
>  I'm trying to get the log added to the appropriate bug

You've reached the horses mouth.

> the rpi maintainer

That would be me, or one of a very small set of colleagues who will already have seen your report.

> even "attach a file" would be more usable

You must have missed the part directly below the initial "Describe the bug" section where there's a paperclip icon and the words "Paste, drop or click to add files".

graemev, 2025-02-10:
Ahh, thanks ...and yes I did "miss the paper-clip"  ..... feel free to delete the above , I'll submit a more readable version (reportbug(1) output) via the paperclip.

AFYI:  The box labelled " System*" says:

Copy and paste the results of the raspinfo command

The output of raspinfo(1) was too big to cut&paste into that box.

In my defence, I'd spent a while collecting data for this to submit to Debian , when they punted it back, I had trouble finding a better location for the report ... then, when I found somewhere (by asking directions) I found a GUI with the "damn fixed text boxes"   one finds on marketing type sites (and they usually lack "paperclips")  ...after about 20+ failed submits I was just hacking large sections out of the report to try to fit it in.

gordoste, 2025-03-18:
Just thought I'd chime in as this error caused me a bit of confusion. In my case, it was caused by a lack of shielding on the cable between my Pi5 and my M.2 NVMe hat. Wrapping a piece of aluminium foil around the cable resolved it (although I've now bought a proper shielded cable).

P33M, 2025-04-03:
What drive? What interface hardware? What power supply?

`Jan 28 14:50:33 argon kernel: nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0x10` = you got here because your PCIe link stopped working and either the nvme driver got all 1s back from a read, or some critical command timed out. There's many reasons why that could happen.

graemev, 2025-04-07:
Raspberry Pi 5, with the **Argon Neo 5 NVMe case**  (so the NVMe controller they include)  [ https://argon40.com/products/argon-neo-5-m-2-nvme-for-raspberry-pi-5 ] the NVMe stick was 

`fanxiang M.2 SSD 2TB, Up to 3500MB/s, 2TB NVMe SSD PCIe Gen3 x4 2280,QLC 3D NAND, Internal Solid State Drives with Graphene Cooling Sticker for Desktop, Laptop -S501Q`

I eventually gave up on this combo and replaced the NVMe with 

`WD Green SN350 NVMe SSD 2TB M.2`

From then on the problem did not reoccur.

The PSU is the "genuine" 27W PSU from PiHut

As, I say, I'm not expecting a "fix" , just that with a year of logs and various kernel update over that period I think there may be some useful data in the logs.

fredrikaverpil, 2025-08-24:
On the Raspberry Pi 5, I never ran the Argon One v5 install script at `https://download.argon40.com/argon1v5.sh` and have been running NixOS just fine with the with the Argon One v5 dual NVMe case. Up until today, that is, when I ran the install script as part of installing the Argon v5 OLED accessory. That messed something up so badly I started getting the error described in this issue and eventually corrupted data on the first NVMe partition which prompted a total re-partition and re-installation of NixOS.

Unfortunately, I don't know exactly what was in my `/boot/firmware/config.txt` prior to running that shell script, but apparently it adds stuff in there. I _think_ that caused interference of some kind, as it all went away with this config under the `[all]` section:

```ini
[all]
dtparam=nvme
dtparam=pciex1
dtoverlay=dwc2,dr_mode=host
```

And the OLED accessory is _not_ installed now.