raspberrypi/linux #6669: Pi 4 mod. B / BCM4345/6 / not set regulatory country code

rseleven, 2025-02-13:
### Describe the bug

```
$ raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.5
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

Linux raspberrypi 6.6.74+rpt-rpi-v8 raspberrypi/bookworm-feedback#1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Revision	: d03115
Serial		:
Model		: Raspberry Pi 4 Model B Rev 1.5
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

$ dmesg
...
[    0.076283] raspberrypi-firmware soc:firmware: Attached to firmware from 2024-11-26T12:52:28, variant start
[    0.080287] raspberrypi-firmware soc:firmware: Firmware hash is 2ae30f53898ae2f1ba77ff570a92991bedfb0398
...
[   18.738682] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[   18.748360] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
...
[   46.086090] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
...
[ 1898.255613] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
[ 1902.355723] brcmfmac: brcmf_set_channel: set chanspec 0x100c fail, reason -52
[ 1902.356167] brcmfmac: brcmf_set_channel: set chanspec 0x100d fail, reason -52
[ 1902.356576] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[ 1902.467742] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[ 1902.575619] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
...
[ 1954.246037] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
...
```

### Details to fix this yourself

1. From the repository [https://github.com/Infineon/ifx-linux-firmware](https://github.com/Infineon/ifx-linux-firmware) downloaded the new firmware:
- https://github.com/Infineon/ifx-linux-firmware/raw/refs/heads/master/firmware/cyfmac43455-sdio.bin
- https://github.com/Infineon/ifx-linux-firmware/raw/refs/heads/master/firmware/cyfmac43455-sdio.clm_blob
2. Replaced the files in the directory /usr/lib/firmware/cypress/:
- cyfmac43455-sdio.clm_blob
- cyfmac43455-sdio-standard.bin

```
$ dmesg
...
[   12.948059] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Oct 28 2024 23:27:00 version 7.45.286 (be70ab3 CY) FWID 01-95efe7fa
...
```

rseleven, 2025-02-13:
1. For example: BY, KZ, RU, UA.

As I understand it, the clm blob file is part of Raspberry Pi OS from 2021-11-15.

pelwell, 2025-02-14:
> Raspberry Pi OS from 2021-11-15

Yes, and it needs updating.