raspberrypi/linux #6672: Feature request: Include iwlwifi kernel module as part of standard kernel build

timwb, 2025-02-17:
### Describe the bug

This kernel module was already included in kernel 6.5 with feature request #6497 - However in the -next firmware branch with kernel 6.12 it's missing again. Can iwlwifi be added to that one (and all subsequent versions) as well? Thanks in advance!

### Steps to reproduce the behaviour

* Get a Pi 5, M.2 E-key HAT and Intel BE200 card. Works with Raspbian out of the box.
* Run BRANCH=next rpi-update & reboot
* Card no longer works because there's no iwlwifi module.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

2025/02/12 10:51:52
Copyright (c) 2012 Broadcom
version f788aab6 (release) (embedded)

Linux raspi-ap-2 6.12.13-v8-16k+ #1848 SMP PREEMPT Mon Feb 10 13:18:35 GMT 2025 aarch64 GNU/Linux

pelwell, 2025-02-18:
Sorry about that - it's the result of a process failure on my part. When we add a new kernel branch to our repo we check for config settings that may have got lost in the version bump - this catches most things, although there is still some human effort involved in confirming that some losses are correct. From that point on we check that commits that go into the main branch also go into the active newer branches. Until those newer branches become the new stable branch, or go EOL, there is a lot of churn because all the changes to the Device Tree files and defconfig files get squashed together.

The IWLWIFI addition happened fairly late in the life of rpi-6.6.y, and at a time when we were starting to prepare rpi-6.12.y to go mainstream. My guess is that the arrival of the defconfig change in rpi-6.6.y coincided with the squashing of other defconfig changes in rpi-6.12.y, so that when I scanned the list of changes I ignored it thinking that it was missing because it had already been squashed, rather than because it hadn't yet been cherry-picked.

Anyway, fd2db7eaeef1b is there now, and will remain until the next squash. Judging by the time this issue was created, there should be a trial build available when you read this - just run `sudo rpi-update rpi-6.12.y` to install it. At some point after that it will get moved to `next`, and on to the new default branch in the near future.

timwb, 2025-02-23:
Hi @pelwell  , thank you very much for your comprehensive explanation. I suspected it was something like that :)

After running rpi-update:
```
$ uname -a
Linux raspi-ap-2 6.12.15-v8-16k+ #1 SMP PREEMPT Fri Feb 21 11:40:47 UTC 2025 aarch64 GNU/Linux

$ lspci -vvv
[...]
0000:01:00.0 Network controller: Intel Corporation Wi-Fi 7(802.11be) AX1775*/AX1790*/BE20*/BE401/BE1750* 2x2 (rev 1a)
        Subsystem: Intel Corporation Wi-Fi 7(802.11be) AX1775*/AX1790*/BE20*/BE401/BE1750* 2x2 (BE200 320MHz [Gale Peak])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 38
        Region 0: Memory at 1b80000000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: <access denied>
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi
```