raspberrypi/linux #6677: "display_auto_detect" is not enabled by default on Pi 5

spockfish, 2025-02-19:
### Describe the bug

The documentation states that `display_auto_detect` is on by default. However, on the Pi 5 it is not.

### Steps to reproduce the behaviour

Without *any* graphics overlay configured, with `display_auto_detect` on, the touchscreen driver is loaded. So for an official Raspberry Pi touchscreen this is the `raspberrypi_ts` module. For the the official Display 2 this is the `goodix_ts` module.

However, on a Pi 5 this is not happening, unless `display_auto_detect=1` is explicitly set `config.txt`, suggesting that it is `0` by default. 

### Device (s)

Raspberry Pi 5

### System

6.6.78

popcornmix, 2025-02-19:
display_auto_detect is enabled by default in the config.txt file supplied with bookworm.
```
$ grep display_auto_detect /boot/firmware/config.txt 
display_auto_detect=1
```
It is not set by default in the bootloader/firmware.

pelwell, 2025-02-19:
This behaviour does not come from the kernel and it's by design.

closed 2025-02-19 by pelwell

spockfish, 2025-02-19:
Ok.... but then I think the documentation is not entirely clear: you state that it is by default available in the config.txt provided by bookworm, but the documention does not mention a specific distribution anywhere. The only thing that is stated is this:

> Auto detect
> 
> Auto detect allows your Raspberry Pi to connect with a display without a manually configured device tree overlay. Auto detection is enabled by default. You can enable display auto detect by adding the following line to config.txt:
>