raspberrypi/linux #6704: Bluetooth usb dongle not seen at startup

NeverBeenHere999, 2025-03-06:
### Describe the bug

Hello, Ihave a raspi5 and using an external bluetooth usb dongle ( tplink ub500 plus).

I disabled the bt on the board, dtoverlay=disable-bt
Installed blueman to manage the controller.
Everything works ok BUT after almost every boot i have to unplug & replug the dongle in order to be used properly.

The system always see it, but does not use it as bt controller.

Considering that my keyboard is bt, this is very annoying.
Any help?

### Steps to reproduce the behaviour

Disable onboard bt (dtoverlay=disable-bt in in /boot/firmware/config.txt
Install blueman
Plug usb dongle
Restart

Problem appears

### Device (s)

Raspberry Pi 5

### System

Raspberry pi os at the latest rev at 6 march 2025 (but the behavior is present at least since a year ago)

6by9, 2025-03-06:
Provide logs - `dmesg`

Does the Bluetooth adapter get enumerated as a USB device by the kernel? `lsusb -t` and `sudo lsusb -vv`

6by9, 2025-03-06:
Duplicated by https://forums.raspberrypi.com/viewtopic.php?p=2300623 and https://forums.raspberrypi.com/viewtopic.php?t=384144

Please don't post the same issue in multiple places without at least providing a link to the other enquiries. You've had a very similar response from jdb on the forum as you got from me here - https://forums.raspberrypi.com/viewtopic.php?p=2300625#p2300625

closed 2025-03-06 by 6by9