raspberrypi/linux #6718: Latest kernel trees: hevc: Failed to find a V4L2 device for H265

Dark-Sky, 2025-03-13:
### Describe the bug

With the very latest kernel trees I am unable to do V4L2 H265 HW decoding using ffmpeg built from jc-kynesim's fork.

### Steps to reproduce the behaviour

Example v6.12.18 does not work but v6.12.17 does.

### Device (s)

Raspberry Pi 5

### System

[ray@jellyfin ~]$ sudo vcgencmd version
2025/02/12 10:51:52 
Copyright (c) 2012 Broadcom
version f788aab6 (release) (embedded)

[ray@jellyfin ~]$ uname -a
Linux jellyfin 6.12.18-1-MANJARO-RPI5 #1 SMP PREEMPT Mon Mar 10 15:04:45 UTC 2025 aarch64 GNU/Linux

Mesa: 24.3.4-1 and 25.0.1

### Logs

[ray@jellyfin ~]$ mpv --profile=fast --hwdec=auto-safe ./127w_nasa_uhd_02042022.ts 
[ffmpeg/video] hevc: PPS id out of range: 0
[ffmpeg/video] hevc: Skipping invalid undecodable NALU: 39
[ffmpeg/video] hevc: PPS id out of range: 0
[ffmpeg/video] hevc: Skipping invalid undecodable NALU: 9
[ffmpeg/video] hevc: PPS id out of range: 0
repeats several times.....

● Video  --vid=1               (hevc 3840x2160 59.9401 fps)
● Audio  --aid=1  --alang=eng  (mp2 2ch 48000 Hz 128 kbps)
○ Audio  --aid=2  --alang=eng  (aac 2ch 48000 Hz 129 kbps)
[ffmpeg/video] hevc: Failed to find a V4L2 device for H265
[ffmpeg/video] hevc: Failed setup for format drm_prime: hwaccel initialisation returned error.
[ffmpeg/video] hevc: Could not find ref with POC -15067
[ffmpeg/video] hevc: Could not find ref with POC -15059
[ffmpeg/video] hevc: Device does not support the VK_KHR_video_decode_queue extension!
[ffmpeg/video] hevc: Failed setup for format vulkan: hwaccel initialisation returned error.
[ffmpeg/video] hevc: Could not find ref with POC -15067
[ffmpeg/video] hevc: Could not find ref with POC -15059
AO: [pipewire] 48000Hz stereo 2ch s16p
[ffmpeg/video] hevc: Failed to find a V4L2 device for H265
[ffmpeg/video] hevc: Failed setup for format drm_prime: hwaccel initialisation returned error.
● Video  --vid=1               (hevc [Main 10] 3840x2160 59.9401 fps)
● Audio  --aid=1  --alang=eng  (mp2 2ch 48000 Hz 128 kbps)
○ Audio  --aid=2  --alang=eng  (aac 2ch 48000 Hz 129 kbps)
○ Subs   --sid=1               (eia_608) [default]
VO: [gpu] 3840x2160 yuv420p10

Audio/Video desynchronisation detected! Possible reasons include too slow
hardware, temporary CPU spikes, broken drivers, and broken files. Audio
position will not match to the video (see A-V status field).
Consider trying `--profile=fast` and/or `--hwdec=auto-safe` as they may help.

AV: 00:00:01 / 00:02:03 (1%) A-V:  1.309 Dropped: 22

### Additional context

If I had to guess there is also an issue with the pi4 also as someone posted this on the arch-arm forums although I have no issue with v6.12.17:

https://archlinuxarm.org/forum/viewtopic.php?f=65&t=17222

Dark-Sky, 2025-03-13:
> Which of the branches from https://github.com/jc-kynesim/rpi-ffmpeg/ are you building?
> 
https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/7.1/rpi_import_2

6by9, 2025-03-13:
@jc-kynesim which are the current branches on rpi-ffmpeg?
dev/7.1/rpi_import_2 does appear to be the default branch, but doesn't have the patches.

Dark-Sky, 2025-03-13:
What patches does it need?

popcornmix, 2025-03-13:
I think you need origin/test/7.1/main
Specifically "v4l2_req_hevc: Fix up for latest sand scheme"

6by9, 2025-03-13:
The top two on https://github.com/jc-kynesim/rpi-ffmpeg/commits/test/7.1/main/
[v4l2_req_hevc: Fix up for latest sand scheme](https://github.com/jc-kynesim/rpi-ffmpeg/commit/09d43080e235a998e2aa9a2098e4e0447e670543)
[Revert "frame: Avoid any cropping if drmprime and sand might be fmt"](https://github.com/jc-kynesim/rpi-ffmpeg/commit/af10b2fdc1719fce56bd95ab44cea2646d524e56)

Dark-Sky, 2025-03-13:
Switched to test/7.1/main. All good now. Thanks everyone!!

```
Using hardware decoding (drm).
● Video  --vid=1               (hevc [Main 10] 3840x2160 59.9401 fps)
● Audio  --aid=1  --alang=eng  (mp2 2ch 48000 Hz 128 kbps)
○ Audio  --aid=2  --alang=eng  (aac 2ch 48000 Hz 129 kbps)
○ Subs   --sid=1               (eia_608) [default]
VO: [gpu] 3840x2160 drm_prime[rpi4_10]
AV: 00:00:28 / 00:02:03 (23%) A-V:  0.000 Dropped: 1
```

closed 2025-03-13 by popcornmix