raspberrypi/linux #6719: Intel and Nvidia Graphics firmware installs by default?

geerlingguy, 2025-03-13:
### Describe the bug

I've noticed—as well as a number of people who have pinged me about it ([example](https://x.com/hbidamian/status/1900223584871571940)) that `firmware-intel-graphics` and `firmware-nvidia-graphics` are now installing by default on both Lite and Full Pi OS installs.

It doesn't seem like Intel nor Nvidia _kernel modules_ are being configured (at least not yet?) in the default Pi OS kernel, so I'm wondering if the inclusion of the graphics firmware in the default install is a mistake?

https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm64/configs/bcm2712_defconfig

If anything, the AMD graphics firmware would be more useful, since the community's been working on a set of patches that give almost complete functionality to most of the modern AMD graphics cards on the Pi...

### Steps to reproduce the behaviour

Run `sudo apt update && sudo apt upgrade` on a Pi OS Bookworm install.

```
$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libcamera0.3
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  firmware-intel-graphics firmware-intel-misc firmware-marvell-prestera firmware-mediatek firmware-nvidia-graphics
  libcamera0.4 linux-headers-6.6.74+rpt-common-rpi linux-headers-6.6.74+rpt-rpi-2712 linux-headers-6.6.74+rpt-rpi-v8
  linux-image-6.6.74+rpt-rpi-2712 linux-image-6.6.74+rpt-rpi-v8 linux-kbuild-6.6.74+rpt
...
```

### Device (s)

Raspberry Pi Zero 2 W, Raspberry Pi 5

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

$ vcgencmd version
2025/03/03 15:15:03 
Copyright (c) 2012 Broadcom
version 9116015f (release) (embedded)

$ uname -a
Linux pi5-wifi 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
```

geerlingguy, 2025-03-13:
_Separately..._, it would be nice to have `firmware-iwlwifi` if we're going to be adding in lots of extra firmware packages :D

XECDesign, 2025-03-13:
Some context - https://github.com/RPi-Distro/firmware-nonfree/issues/50#issuecomment-2721721163

The short version is that this was all previously already shipped in the 'misc' firmware package and now people will have the option to uninstall things they don't need. Future images will ship without the nvidia and intel graphics firmware packages.

I'll take a look at adding firmware-iwlwifi.

XECDesign, 2025-03-13:
> _Separately..._, it would be nice to have `firmware-iwlwifi` if we're going to be adding in lots of extra firmware packages :D

Hmm, not sure about that one...

```
Installed-Size: 100.0 MB
```

geerlingguy, 2025-03-13:
Heh, well it's worth a shot! ;)