raspberrypi/linux #6721: Hyperpixel 4 fails to boot on 6.12 kernel

LaaZa, 2025-03-16:
---
name: vc4-kms-dpi-hyperpixel4 breaks boot on kernel 6.12

---

**Describe the bug**
Boot fails with vc4-kms-dpi-hyperpixel4 enabled on 6.12 kernel, all signal is lost and boot does not proceed.

**To reproduce**
1. Run kernel 6.12
2. Enable Hyperpixel 4 overlay with vc4-kms-dpi-hyperpixel4 in config.txt
3. Boot fails and HDMI nor the Hyperpixel has signal

**Expected behaviour**
System boots normally and the Hyperpixel is working as it does on 6.6 kernel

**Actual behaviour**
Fail to boot and loss of signal

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? RPi 400
* Which OS and version (`cat /etc/rpi-issue`)? EndeavourOS(latest)/Alpine(3.21)
* Which firmware version (`vcgencmd version`)? Feb 17 2025 da7125
* Which kernel version (`uname -a`)? Any 6.12.y (official Raspberry Pi kernel)

**Logs**
I don't have logs because signal is lost and it fails during boot.

**Additional context**
Fixed on EndeavourOS by compiling 6.6 kernel. Didn't replace overlays, only kernel.img.

6by9, 2025-03-17:
Thanks for the report. Reproduced. Now investigating why it fails.

6by9, 2025-03-17:
Ah, downstream 6.6 had commit https://github.com/raspberrypi/linux/commit/7a5df2564ca1e83585326c183dedd9db662aa389 that added the driver for ili9806 as used on the HyperPixel4.

6.12 has gained a driver for the same display using the same filename via https://github.com/raspberrypi/linux/commit/baf272bac637d3275bb83c17ac849b44a4590655
The tree therefore doesn't have the downstream driver, so no configuration for the HyperPixel4.

Time to merge the two (hopefully they are close enough to compatible).

LaaZa, 2025-03-18:
I don't have one to test, but could this affect the square one too?

6by9, 2025-03-18:
> I don't have one to test, but could this affect the square one too?

No, it's only the ili9806e driver that changed. The square one uses the panel-tdo-y17p driver. (I tested the square one as part of my debug in order to eliminate it being a DPI driver issue).

6by9, 2025-03-18:
I should have posted yesterday that there is a hopeful fix in #6722.
`sudo rpi-update pulls/6722` should get that test kernel for you if you wanted to test.

One weirdness observed was that the display initially came up fine, but then changed to have a red tinge just as the command prompt arrived (I'm booting to console rather than desktop). The latest 6.6 also seemed to do the same thing. It'd be interesting to know if yours does the same.

LaaZa, 2025-03-18:
I'm not seeing any red or weirdness on the 6.6 kernel I built. I don't think I have rpi-update on EOS. I have to manually build I think. Just that the rotation does not apply to any graphical environment, I need to use their own rotation, but that isn't anything new. Everything looks normal while it's booting and no weird colors after either. But I haven't tried 6.12 with your patch yet.

LaaZa, 2025-03-18:
Might take a while.

[image]

LaaZa, 2025-03-18:
With initial testing, seems to work.

[image]

6by9, 2025-03-18:
Great.
I'm guess something's off with my Pi/display, so I'll ignore that for now.

6by9, 2025-03-18:
Sussed my weird images. Some numpty (aka me) had added `pinctrl set 25 op dh` to ~/.bashrc, so as it logged in we lost the MSB of blue.

The patch is now merged, so I'll close this down. Thanks for reporting and testing.

closed 2025-03-18 by 6by9