raspberrypi/linux #6725: RP1 Overclock regression in 6.12

schoolpost, 2025-03-18:
### Describe the bug

In order to support a higher frame rates at higher resolutions with CSI-2 camera modules, the RP1 clock must be increased from the default 200Mhz ( 400Mpx/s ) to a higher 300Mhz ( 600Mpx/s ) 

This was possible in kernel 6.6, in fact when overclock to 300Mhz in 6.6 it would actually set the clock to 333Mhz; but my latest attempts in 6.12 appear to have no effect anymore. ( stuck at 200Mhz ) 

It would be great to have a config.txt method of setting an increased RP1 clock for camera use cases. 

### Steps to reproduce the behaviour

Please refer here for more information: https://github.com/raspberrypi/libcamera/issues/243

### Device (s)

Raspberry Pi 5

### System

Linux raspberrypi 6.6.78-v8-16k+ #1 SMP PREEMPT Mon Mar  3 15:28:28 MST 2025 aarch64 GNU/Linux

( I've gone back to 6.6 in the mean time, so I can't provide a 6.12 log here )

pelwell, 2025-03-21:
I can't reproduce this - the clock still changes on 6.12, both with a self-built kernel and with the latest from rpi-update, although both require a modified DTB as you have found. Are you sure you copied in the hacked dtb file?

This kind of modification could be done with an overlay, but publishing that makes it look approved of. And I'd want to change the clock driver to make it legal. Currently I think it only works because the parent PLL is modified without looking at the maximum frequency, and since the sys clock is 1:1 with the parent it gets changed immediately.

What are your thoughts, @naushir?

schoolpost, 2025-03-22:
Perhaps it was an issue with a previous version of `6.12` I tried a few weeks ago, but seems to be fixed since. I am able to set the clock to a higher speed now:

```
       pll_sys                       2       2        0        333333333   0          0     50000      Y         deviceless                      no_connection_id         
          clk_mipi1_dpi              0       0        0        333333333   0          0     50000      Y            deviceless                      no_connection_id         
          clk_mipi0_dpi              0       0        0        333333333   0          0     50000      Y            deviceless                      no_connection_id         
          clk_dpi                    0       0        0        333333333   0          0     50000      Y            deviceless                      no_connection_id         
          clk_sdio_alt_src           0       0        0        166666666   0          0     50000      Y            deviceless                      no_connection_id         
             clk_gp2                 0       0        0        166666666   0          0     50000      Y               deviceless                      no_connection_id         
          clk_sys                    6       6        0        333333333   0          0     50000      Y            1f00080000.i2c                  no_connection_id  
```

Having this configurable through an overlay would still be a welcome add however.

closed 2025-03-22 by schoolpost

will127534, 2025-03-22:
For those who stumble across this issue, here is the dtoverlay to change RP1 clock without recompile the kernel:

```dts
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2712";

	fragment@0 {
		target = <&rp1_clocks>;
		__overlay__ {
			/*
			 * Re-specify the entire assigned-clock-rates array.
			 * Only the items for RP1_PLL_SYS (index #2) and
			 * RP1_CLK_SYS (index #7) have been changed to 300000000.
			 */
			assigned-clock-rates = <
				/* RP1_PLL_SYS_CORE  */ 1000000000
				/* RP1_PLL_AUDIO_CORE*/ 1536000000
				/* RP1_PLL_SYS       */ 300000000
				/* RP1_PLL_SYS_SEC   */ 125000000
				/* RP1_CLK_ETH       */ 125000000
				/* RP1_PLL_AUDIO     */ 61440000
				/* RP1_PLL_AUDIO_SEC */ 153600000
				/* RP1_CLK_SYS       */ 300000000
				/* RP1_PLL_SYS_PRI_PH*/ 100000000
				/* RP1_CLK_SLOW_SYS  */ 50000000
				/* RP1_CLK_SDIO_TIMER*/ 1000000
				/* RP1_CLK_SDIO_ALT_SRC*/ 200000000
				/* RP1_CLK_ETH_TSU   */ 50000000
			>;
		};
	};
};
```

Save it as `rp1-300mhz.dtso` and do `dtc -@ -I dts -O dtb -o rp1-300mhz.dtbo rp1-300mhz.dtso` to compile, copy the rp1-300mhz.dtbo to /boot/firmware/overlays/ and do dtoverlay=rp1-300mhz in config.txt

eva-00100, 2025-05-22:
May I ask in which version of the kernel the method of not modifying the kernel is effective?