raspberrypi/linux #6726: please enable EROFS

greyltc, 2025-03-19:
### Describe the bug

https://docs.kernel.org/filesystems/erofs.html

the EROFS file system does not seem to be enabled here

### Steps to reproduce the behaviour

n/a

### Device (s)

Raspberry Pi 5

### System

n/a

ragazenta, 2025-04-09:
I sent PR a long time ago: https://github.com/raspberrypi/linux/pull/6163

I can rebase to 6.12 if the maintainer here agree with enabling this.

greyltc, 2025-04-22:
@pelwell, would you consider allowing the EROFS module into your kernel?

pelwell, 2025-04-22:
I can't think of a good reason not too - the modules aren't enormous. See https://github.com/raspberrypi/linux/pull/6806.

greyltc, 2025-08-29:
@popcornmix Testing on kernel version 6.12.43, I can confirm this works! Thank you.

Would it be too much to ask for this to be built with `EROFS_FS_ZIP_ZSTD`?

pelwell, 2025-09-03:
I'm closing the issue because the main request has been addressed.