raspberrypi/linux #6733: sh1106-spi invalid buffer alignment on 6.12 kernel

gtrainavicius, 2025-03-20:
### Describe the bug

Trying to load the DT overlay and module for sh1106 display (possibly other SPI displays are affected too), fails with `invalid buffer alignment`.

It is not necessary to have the actual display to be hooked up to reproduce the problem.

Doesn't happen on 6.6 kernel versions.

### Steps to reproduce the behaviour

Execute:

```
sudo dtoverlay sh1106-spi,reset_pin=17,dc_pin=27
```

### Device (s)

Raspberry Pi 5

### System

```
patch@patchbox:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 64e3efc023558a4d4634a702c467f513d5b2bca1, stage-patchbox
patch@patchbox:~ $ vcgencmd version
2025/03/19 13:41:26
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)
patch@patchbox:~ $ uname -a
Linux patchbox 6.12.19-v8-16k+ #1865 SMP PREEMPT Wed Mar 19 13:48:20 GMT 2025 aarch64 GNU/Linux
```

### Logs

[6.12_error.log](https://github.com/user-attachments/files/19369888/6.12_error.log)

6.6 success log for comparison:

[6.6_success.log](https://github.com/user-attachments/files/19369893/6.6_success.log)

P33M, 2025-03-20:
The splat after that is from the AXI DMAC driver - presumably this error path hasn't been called before...

Also, I'm not entirely sure that the DMAC requires a word-aligned start address. It should only need naturally-aligned buffers of integer multiples of SRC_WIDTH.

6by9, 2025-03-20:
The solomon driver hasn't set a dma_mask, hence the 
```
[   71.879798] ------------[ cut here ]------------
[   71.879804] WARNING: CPU: 1 PID: 1136 at kernel/dma/mapping.c:209 __dma_map_sg_attrs+0x164/0x178
```

pelwell, 2025-03-20:
See https://github.com/raspberrypi/linux/pull/6734.

gtrainavicius, 2025-03-25:
Blacklisting the ssd130x-spi driver worked as a workaround for now, I will get around to testing a newer kernel version again hopefully by the end of the week.

gtrainavicius, 2025-03-27:
The overlay and module loads now without problems, however, the ssd130x-spi module does not work well with sh1106 OLED, I will stick to the older module for now.

closed 2025-03-27 by gtrainavicius

6by9, 2025-03-27:
Have you tried blacklisting fb_sh1106 instead of ssd130x-spi?

With the recent fixes that have been applied (use `sudo rpi-update` to grab the latest kernel), loading the overlay doesn't cause an issue for me (I don't have the hardware to confirm the output).

However you don't want two modules trying to implement the same functionality. If ssd130x-spi works, then dropping fb_sh1106 from the defconfigs would be a good move. All the FBTFT drivers are orphaned and stuck in staging.