raspberrypi/linux #6737: Packet RAM has to be on to store the packet when switching using a KVM

TrevorPH, 2025-03-22:
### Describe the bug

I have an rpi 5 with a newly installed and updated 64 bit raspios with kernel 6.12.19-v8-16k+. I have the rpi connected via an active HDMI to Displayport cable into an Aten CS1924 4 port Displayport KVM switch. It appears that whenever I use the KVM to switch to a different machine, I get an oops detailed below.

In case it makes a difference I am running a MATE desktop with software compositing switched off, using an 8GB RPi5 with an M.2 HAT and a 500GB Samsung 970 Evo SSD.

### Steps to reproduce the behaviour

Install raspios, use raspi-update to update to the latest 6.12 kernel (may also happen on the default 6.6 but have not tried). Attach rpi to a KVM switch and toggle between ports/machines. I imagine a similar condition could be induced by removing and replugging the video cable.

### Device (s)

Raspberry Pi 5

### System

root@rpi5:~# cat /etc/rpi-issue 
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

root@rpi5:~# vcgencmd version
2025/03/19 13:41:26 
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)

Linux rpi5.trevor.local 6.12.19-v8-16k+ #1865 SMP PREEMPT Wed Mar 19 13:48:20 GMT 2025 aarch64 GNU/Linux

Revision	: d04170
Serial		: d5c6c4f0a911ccbf
Model		: Raspberry Pi 5 Model B Rev 1.0

### Logs

2025-03-22T21:08:21.449815+00:00 rpi5 kernel: [ 4823.991461] ------------[ cut here ]------------
2025-03-22T21:08:21.449819+00:00 rpi5 kernel: [ 4823.991468] Packet RAM has to be on to store the packet.
2025-03-22T21:08:21.449820+00:00 rpi5 kernel: [ 4823.991495] WARNING: CPU: 2 PID: 1094 at drivers/gpu/drm/vc4/vc4_hdmi.c:726 vc4_hdmi_write_infoframe+0x3ec/0x468 [vc4]
2025-03-22T21:08:21.449820+00:00 rpi5 kernel: [ 4823.991523] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device joydev algif_hash algif_skcipher af_alg bnep ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables binfmt_misc nfnetlink brcmfmac_wcc vc4 brcmfmac snd_soc_hdmi_codec >
2025-03-22T21:08:21.449821+00:00 rpi5 kernel: [ 4823.991612]  ip_tables x_tables ipv6
2025-03-22T21:08:21.449821+00:00 rpi5 kernel: [ 4823.991619] CPU: 2 UID: 1000 PID: 1094 Comm: pipewire Not tainted 6.12.19-v8-16k+ #1865
2025-03-22T21:08:21.449822+00:00 rpi5 kernel: [ 4823.991623] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
2025-03-22T21:08:21.449822+00:00 rpi5 kernel: [ 4823.991626] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
2025-03-22T21:08:21.449823+00:00 rpi5 kernel: [ 4823.991629] pc : vc4_hdmi_write_infoframe+0x3ec/0x468 [vc4]
2025-03-22T21:08:21.449823+00:00 rpi5 kernel: [ 4823.991642] lr : vc4_hdmi_write_infoframe+0x3ec/0x468 [vc4]
2025-03-22T21:08:21.449824+00:00 rpi5 kernel: [ 4823.991654] sp : ffffc0008747b8a0
2025-03-22T21:08:21.449824+00:00 rpi5 kernel: [ 4823.991656] x29: ffffc0008747b8a0 x28: ffff800141454a00 x27: ffffc00082e6f800
2025-03-22T21:08:21.449824+00:00 rpi5 kernel: [ 4823.991661] x26: ffffc0008747b989 x25: 0000000000000000 x24: 0000000000000084
2025-03-22T21:08:21.449825+00:00 rpi5 kernel: [ 4823.991665] x23: 00000000fffffff4 x22: ffff8001402a2000 x21: ffff800002e06080
2025-03-22T21:08:21.449825+00:00 rpi5 kernel: [ 4823.991669] x20: 000000000000000e x19: ffff800002e065c0 x18: 0000000000000000
2025-03-22T21:08:21.449826+00:00 rpi5 kernel: [ 4823.991674] x17: 0000000000000000 x16: ffffd06fce49aa60 x15: 0000000000000000
2025-03-22T21:08:21.449826+00:00 rpi5 kernel: [ 4823.991678] x14: 0000000000000000 x13: 2e74656b63617020 x12: 6568742065726f74
2025-03-22T21:08:21.449827+00:00 rpi5 kernel: [ 4823.991682] x11: 73206f74206e6f20 x10: ffffd06fcfc5c478 x9 : ffffd06fce51b630
2025-03-22T21:08:21.449827+00:00 rpi5 kernel: [ 4823.991686] x8 : 00000000ffffefff x7 : ffffd06fcfc58380 x6 : 00000000000040f8
2025-03-22T21:08:21.449828+00:00 rpi5 kernel: [ 4823.991690] x5 : ffff80003f77b3c8 x4 : 0000000000000fff x3 : ffffaf906ff88000
2025-03-22T21:08:21.449828+00:00 rpi5 kernel: [ 4823.991694] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff800141454a00
2025-03-22T21:08:21.449828+00:00 rpi5 kernel: [ 4823.991698] Call trace:
2025-03-22T21:08:21.449829+00:00 rpi5 kernel: [ 4823.991700]  vc4_hdmi_write_infoframe+0x3ec/0x468 [vc4]
2025-03-22T21:08:21.449829+00:00 rpi5 kernel: [ 4823.991714]  write_device_infoframe+0x98/0x158 [drm_display_helper]
2025-03-22T21:08:21.449832+00:00 rpi5 kernel: [ 4823.991720]  drm_atomic_helper_connector_hdmi_update_audio_infoframe+0x8c/0xb8 [drm_display_helper]
2025-03-22T21:08:21.449832+00:00 rpi5 kernel: [ 4823.991724]  vc4_hdmi_audio_prepare+0x544/0xf70 [vc4]
2025-03-22T21:08:21.449833+00:00 rpi5 kernel: [ 4823.991737]  hdmi_codec_prepare+0xd8/0x138 [snd_soc_hdmi_codec]
2025-03-22T21:08:21.449833+00:00 rpi5 kernel: [ 4823.991741]  snd_soc_pcm_dai_prepare+0x80/0x120 [snd_soc_core]
2025-03-22T21:08:21.449834+00:00 rpi5 kernel: [ 4823.991770]  __soc_pcm_prepare+0x4c/0x120 [snd_soc_core]
2025-03-22T21:08:21.449834+00:00 rpi5 kernel: [ 4823.991792]  soc_pcm_prepare+0x38/0x68 [snd_soc_core]
2025-03-22T21:08:21.449835+00:00 rpi5 kernel: [ 4823.991814]  snd_pcm_do_prepare+0x38/0x60 [snd_pcm]
2025-03-22T21:08:21.449835+00:00 rpi5 kernel: [ 4823.991830]  snd_pcm_action_single+0x50/0xb8 [snd_pcm]
2025-03-22T21:08:21.450054+00:00 rpi5 kernel: [ 4823.991840]  snd_pcm_action_nonatomic+0xa8/0xb8 [snd_pcm]
2025-03-22T21:08:21.450190+00:00 rpi5 kernel: [ 4823.991850]  snd_pcm_prepare+0x94/0xf8 [snd_pcm]
2025-03-22T21:08:21.450192+00:00 rpi5 kernel: [ 4823.991861]  snd_pcm_common_ioctl+0xda8/0x1680 [snd_pcm]
2025-03-22T21:08:21.450193+00:00 rpi5 kernel: [ 4823.991871]  snd_pcm_ioctl+0x38/0x58 [snd_pcm]
2025-03-22T21:08:21.450193+00:00 rpi5 kernel: [ 4823.991880]  __arm64_sys_ioctl+0xb0/0x100
2025-03-22T21:08:21.450194+00:00 rpi5 kernel: [ 4823.991888]  invoke_syscall+0x50/0x120
2025-03-22T21:08:21.450199+00:00 rpi5 kernel: [ 4823.991894]  el0_svc_common.constprop.0+0xd0/0xf8
2025-03-22T21:08:21.450199+00:00 rpi5 kernel: [ 4823.991898]  do_el0_svc+0x28/0x40
2025-03-22T21:08:21.450200+00:00 rpi5 kernel: [ 4823.991902]  el0_svc+0x30/0xd0
2025-03-22T21:08:21.450201+00:00 rpi5 kernel: [ 4823.991906]  el0t_64_sync_handler+0x13c/0x158
2025-03-22T21:08:21.450201+00:00 rpi5 kernel: [ 4823.991908]  el0t_64_sync+0x190/0x198
2025-03-22T21:08:21.450202+00:00 rpi5 kernel: [ 4823.991911] ---[ end trace 0000000000000000 ]---

popcornmix, 2025-03-23:
> I imagine a similar condition could be induced by removing and replugging the video cable.

would be useful if you could confirm this.

popcornmix, 2025-03-23:
It would be useful if you could work out how to reproduce the issue.
From the kernel backtrace it looks like audio was playing when it occurred, which may be relevant.

TrevorPH, 2025-03-23:
Only thing I can think of is that the notifications that thunderbird is trying to raise will be trying to also send a notfication 'ding' at the same time. Audio would be routed over HDMI to the display.