raspberrypi/linux #6738: Raspberry Pi OS Lite Kernel Panic Occured

kirisaka, 2025-03-24:
MODEL : Raspberry Pi CM3 / CM3+

DESCRIPTION : I have Custom Board for CM3 / CM3+ / CM4S (SODIMM TYPE Compute Module) My Board was worked with Raspberry Pi OS with Kernel Version 5.15.84-v8+properly.

I installed the latest Raspberry Pi OS Lite to upgrade the Linux version for a system update, but a kernel panic occurred, and the system failed to boot and I also tested the Legacy version (Kernel 6.1.21, the next version after 5.18.84), just in case, but it resulted in the same kernel panic and failed to boot. It seems to be a recurring issue with Linux Kernel 6 and above.

LOG

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.51+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) raspberrypi/bookworm-feedback#1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 3 Plus Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001e400000..0x000000002e3fffff (262144 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85672 r8192 d29016 u122880
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=B8:27:EB:E0:E4:82 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=efd888da-02 rootfstype=ext4 fsck.repair=yes rootwait systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 650508K/970752K available (13440K kernel code, 2212K rwdata, 4268K rodata, 4864K init, 1083K bss, 58100K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43242 entries in 169 pages
[    0.000000] ftrace: allocated 169 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000397] Console: colour dummy device 80x25
[    0.000414] printk: console [tty1] enabled
[    0.001309] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001355] pid_max: default: 32768 minimum: 301
[    0.001468] LSM: initializing lsm=capability,integrity
[    0.001718] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001772] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002806] cgroup: Disabling memory control group subsystem
[    0.004846] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004997] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.005167] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.005508] rcu: Hierarchical SRCU implementation.
[    0.005534] rcu:     Max phase no-delay instances is 1000.
[... 452 lines cut ...]
[    9.042314] x2 : 0000000000000020 x1 : 0000000000000040 x0 : ffffff8001c01180
[    9.051216] Call trace:
[    9.055370]  __kmem_cache_alloc_node+0xd0/0x260
[    9.061641]  __kmalloc+0x60/0x1b0
[    9.066671]  ext4_htree_store_dirent+0x48/0x130
[    9.072878]  htree_dirblock_to_tree+0x1cc/0x348
[    9.079018]  ext4_htree_fill_tree+0xe0/0x368
[    9.084855]  ext4_readdir+0x4f0/0x998
[    9.090050]  iterate_dir+0x98/0x1a0
[    9.095020]  __arm64_sys_getdents64+0x74/0x160
[    9.100911]  invoke_syscall+0x50/0x128
[    9.106062]  el0_svc_common.constprop.0+0x48/0xf0
[    9.112150]  do_el0_svc+0x24/0x38
[    9.116803]  el0_svc+0x40/0xe8
[    9.121210]  el0t_64_sync_handler+0x100/0x130
[    9.126924]  el0t_64_sync+0x190/0x198
[    9.131906] Code: 54000a20 b9402a62 aa1603e1 aa1303e0 (f8626adb)
[    9.139355] ---[ end trace 0000000000000000 ]---
[    9.145360] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    9.154451] SMP: stopping secondary CPUs
[    9.159766] Kernel Offset: 0x2b00e00000 from 0xffffffc080000000
[    9.167113] PHYS_OFFSET: 0x0
[    9.171396] CPU features: 0x0,0000000d,00020000,0000421b
[    9.178157] Memory Limit: none
[    9.182621] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---

pelwell, 2025-03-24:
Your system seems to be suffering from kernel heap corruption:
```
[ 7.528848] pc : __kmem_cache_alloc_node+0xd0/0x260
[ 7.532780] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000060
```
The kernel heap management is so well tested that this can really only be down to two things:
1. You are using a rogue driver that writes outside its allocated memory.
2. There is a hardware issue in your system that causes a bus master to write where it shouldn't.
3. There is a hardware issue in your system that makes memory accesses unreliable.

Changes in kernel versions and compilers can often improve efficiency, or just change code enough to expose a deficiency somewhere in the system - often the power supply. I would start by unplugging any unneeded hardware, adding `arm_freq=600` to config.txt, and seeing if the system is more reliable.

P33M, 2025-03-24:
> 1. You are using a rogue driver that writes outside its allocated memory.

This is likely - there is a line in the oops with:
`[ 8.912178] CPU: 0 PID: 1 Comm: systemd Tainted: G D 6.6.51+rpt-rpi-v8 https://github.com/raspberrypi/bookworm-feedback/issues/1 Debian 1:6.6.51-1+rpt3`
G = proprietary module loaded. On a regular Raspberry Pi OS desktop install, this flag is not set (but C is due to the rpivid_hevc module).

P33M, 2025-03-24:
(Bogus comment deleted)