raspberrypi/linux #6751: After last upgrade to 6.12.20 from two identical CM4 one is not booting from NVME

ralf-9000, 2025-03-31:
### Describe the bug

I have 2 identical CM4s with the same Waveshare board with (only) NVME SSD. They run the same application and on Sunday I made on both the same official upgrade/update to 6.12.20. One is ok and one isn't anymore booting from NVME. The NVME drives are from different vendors, that is the only HW difference.

Because there are identical (except NVME drive vendor), I cloned the SSD of the booting one to the failed one. Again not booting.

So I made a SD card with the old 6.6 and the same disk UUID and boot the failed CM5 again. Immediately it is booting, recognized the SSD with the 6.6 kernel and mount the root filesystem with /dev/nvme0n1p1 (and not the SD one) and everything is fine and is working like normal.

Now I updated this SD card to 6.12.20 and - for sure it boots - but now NOT with SSD as root filesystem, it uses now the SD partition 2.

I logged in and checked the device entries. Only /dev/nvme0 was there, no ...0n1, no partition entries.

Now with booting SD 6.6. and filesystem on the NVME, I checked everything: both partitions of the two CM5 have the same content and structure as on the identical working CM4 with the different NVME vendor ....

Can it be that we have in the new kernel again a timing issue (as in old times) with recognizing the SSD?

Ralf

### Steps to reproduce the behaviour

See problem description ....

### Device (s)

Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

Pi OS 6.12.20

ralf-9000, 2025-03-31:
I have checked now between which versions it has happened: this error has came in between 6.12.18 and 6.12.20. 6.12.18 boots normal from this SSD. 6.12.19 not tested.

ralf-9000, 2025-03-31:
The NOT booting one is a SK hynix BC511 NVMe SSD (512GB):

`pi@cm4-io-base-box-1:~ $ sudo lspci -vvv
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20) (prog-if 00 [Normal decode])
        Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@0,0
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 27
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [48] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
        Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0
                        ExtTag- RBE+
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <1us, L1 <2us
                        ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s, Width x1
                        TrErr- Train- SlotClk+ DLActive- BWMgmt+ ABWMgmt+
                RootCap: CRSVisible+
                RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
                RootSta: PME ReqID 0000, PMEStatus- PMEPending-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
                         10BitTagComp- 10BitTagReq- OBFF Via WAKE#, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd-
                         AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled, ARIFwd-
                         AtomicOpsCtl: ReqEn- EgressBlck-
                LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS+
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported, DRS-
                         DownstreamComp: Link Down - Not Determined
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
                RootCmd: CERptEn+ NFERptEn+ FERptEn+
                RootSta: CERcvd- MultCERcvd- UERcvd- MultUERcvd-
                         FirstFatal- NonFatalMsg- FatalMsg- IntMsg 0
                ErrorSrc: ERR_COR: 0000 ERR_FATAL/NONFATAL: 0000
        Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
        Capabilities: [240 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=24us LTR1.2_Threshold=56320ns
                L1SubCtl2: T_PwrOn=26us
        Kernel driver in use: pcieport
[... 184 lines cut ...]
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
[... cut ...]
                LnkCtl3: LnkEquIntrruptEn- PerformEqu-
                LaneErrStat: 0
        Capabilities: [900 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1- ASPM_L1.2+ ASPM_L1.1- L1_PM_Substates+
                          PortCommonModeRestoreTime=32us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=0us LTR1.2_Threshold=48128ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: nvme
`

P33M, 2025-04-01:
What happens if you add `pcie_aspm=off` to /boot/firmware/cmdline.txt on the broken drive/kernel?

ralf-9000, 2025-04-01:
I have tried this: no difference, still not booting directly from the SSD.
In a 2nd try I have add this option to the 6.16.20 SD card, it boots completely from SD, but the SDD partitions are not there then, like before.
In a 3rd try with the 6.6 SD card and the 6.12.18 is no difference: the SDD partitions can be mounted. 

I don't think it has to do with the power management, I still assume it's the longer duration of this SSD to come up: 26us compared to 10us of the working one. Maybe now the kernal waits not any more long enough fr the SDD coming up ... only a thought.

P33M, 2025-04-01:
As you have a boot from the SD, please post the full output of `dmesg` in the non-working case.

ralf-9000, 2025-04-01:
Will do this in the  evening, I am now at work ....

ralf-9000, 2025-04-02:
Sorry, I came yesterday to late from work and then family has priority .... but now in the morning:

I have created again a fresh SD card with 64bit lite image, after that:
`pi@cm4-io-base-box-1:~ $ uname -a
Linux cm4-io-base-box-1 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
pi@cm4-io-base-box-1:~ $ ls -l /dev/nvme*
crw------- 1 root root 245, 0 Nov 19 14:40 /dev/nvme0
brw-rw---- 1 root disk 259, 0 Nov 19 14:40 /dev/nvme0n1
brw-rw---- 1 root disk 259, 1 Nov 19 14:40 /dev/nvme0n1p1
brw-rw---- 1 root disk 259, 2 Nov 19 14:40 /dev/nvme0n1p2
pi@cm4-io-base-box-1:~ $
`
Then I upgraded/updated everything and end up with 6.12.20 after reboot:
`pi@cm4-io-base-box-1:~ $ uname -a
Linux cm4-io-base-box-1 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
pi@cm4-io-base-box-1:~ $ ls -l /dev/nvme*
crw------- 1 root root 245, 0 Apr  2 09:21 /dev/nvme0
pi@cm4-io-base-box-1:~ $
`
Now I can post the dmesg output:
`pi@cm4-io-base-box-1:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.12.20+rpt-rpi-v8 (serge@raspberrypi.com) (aarch64-linux-gnu-gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000e800000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000e800000..0x000000002e7fffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef66760..0x000000003ef667a7 (0 KiB) nomap non-reusable nvram@0
[    0.000000] OF: reserved mem: 0x000000003ef66520..0x000000003ef6671f (0 KiB) nomap non-reusable nvram@1
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x0000000103ffffff] (4160MB)
[    0.000000] Faking node 1 at [mem 0x0000000104000000-0x00000001ffffffff] (4032MB)
[    0.000000] NUMA: Initialized distance table, cnt=2
[    0.000000] NODE_DATA(0) allocated [mem 0x103ffd300-0x103ffffff]
[    0.000000] NODE_DATA(1) allocated [mem 0x1feff5300-0x1feff7fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x0000000103ffffff]
[    0.000000]   node   1: [mem 0x0000000104000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000103ffffff]
[    0.000000] Initmem setup node 1 [mem 0x0000000104000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 33 pages/cpu s95064 r8192 d31912 u135168
[    0.000000] pcpu-alloc: s95064 r8192 d31912 u135168 alloc=33*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  numa=fake=2 system_heap.max_order=0 smsc95xx.macaddr=E4:5F:01:54:65:E4 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=eab00cab-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0-1'
[    0.000000] Fallback order for Node 0: 0 1
[    0.000000] Fallback order for Node 1: 1 0
[... 434 lines cut ...]
[    8.895876] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[    8.895884] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[    8.895890] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[    8.895895] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[    8.895925] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    8.898861] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[    8.900092] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    8.900843] Bluetooth: HCI UART protocol Broadcom registered
[... cut ...]
[   11.168260] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   11.168268] Bluetooth: hci0: Failed to set baudrate
[   11.350960] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   11.351931] bcmgenet fd580000.ethernet eth0: Link is Down
[   13.184202] Bluetooth: hci0: command 0xfc18 tx timeout
[   13.184233] Bluetooth: hci0: BCM: Reset failed (-110)
[   15.424686] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
pi@cm4-io-base-box-1:~ $
`

ralf-9000, 2025-04-02:
The only difference I see to my fully booting 2 in 1 is:

```
i@cm4-io-base-box-1:~ $ dmesg | grep nvme
[    0.709495] nvme nvme0: pci function 0000:01:00.0
[    0.709560] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[    0.712010] nvme nvme0: missing or invalid SUBNQN field.
[    0.714095] nvme nvme0: IO queues not created
pi@cm4-io-base-box-1:~ $
```

On the SSD booting 2 I see the correct output:

```
pi@cm4-io-base-box-2:~ $ dmesg | grep nvme
[    0.697302] nvme nvme0: pci function 0000:01:00.0
[    0.697343] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[    0.704022] nvme nvme0: failed to allocate host memory buffer.
[    0.704794] nvme nvme0: 1/0/0 default/read/poll queues
[    0.714206]  nvme0n1: p1 p2
[    1.673657] EXT4-fs (nvme0n1p2): mounted filesystem ce208fd3-38a8-424a-87a2-cd44114eb820 ro with ordered data mode. Quota mode: none.
[    2.319553] EXT4-fs (nvme0n1p2): re-mounted ce208fd3-38a8-424a-87a2-cd44114eb820 r/w. Quota mode: none.
pi@cm4-io-base-box-2:~ $
```

But if I go back to the 6.6.51 where 1 is booting fine I see also the "SUBNQN" line, so it can't be the case.

```
pi@cm4-io-base-box-1:~ $ uname -a
Linux cm4-io-base-box-1 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
pi@cm4-io-base-box-1:~ $ dmesg | grep nvme
[    1.311723] nvme nvme0: pci function 0000:01:00.0
[    1.311753] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[    1.321558] nvme nvme0: missing or invalid SUBNQN field.
[    1.325726] nvme nvme0: 4/0/0 default/read/poll queues
[    1.331070]  nvme0n1: p1 p2
[    4.210103] EXT4-fs (nvme0n1p2): mounted filesystem ce208fd3-38a8-424a-87a2-cd44114eb820 ro with ordered data mode. Quota mode: none.
[    5.295868] EXT4-fs (nvme0n1p2): re-mounted ce208fd3-38a8-424a-87a2-cd44114eb820 r/w. Quota mode: none.
pi@cm4-io-base-box-1:~ $

```

I think still that is a timing issue, is there a possibility to delay the SSD recognition a bit for a test?

Ralf

P33M, 2025-04-02:
Please post the full contents of dmesg in the failing case (to pastebin if too large).

P33M, 2025-04-02:
Ah. It's already an attachment. Something odd is happening - it can't allocate memory for the hardware descriptors. Do you have anything nonstandard in the config.txt?

The usual suspects are if a device (or host window) is 32-bit but tries to allocate from above 4GB. Or above 3GB in the case of older BCM2711 chips. It should all "just work" as the firmware fixes up any such mismatches.

ralf-9000, 2025-04-02:
There is only this SSD and the CPU fan (with 4-pin fan control) connected directly to this CM4. The CM4 has 8GB RAM and is without eMMC.

The other one is identical, the ONLY difference is the SSD model as describted before .... working in this setup for years with every update/upgrade without any problems up to 6.12.18 and then to 6.12.20 .... bum ...

ralf-9000, 2025-04-02:
> Do you have anything nonstandard in the config.txt of either board?

Here is my config.txt ....

[config.txt](https://github.com/user-attachments/files/19567075/config.txt)

pelwell, 2025-04-02:
Have you tried without `dtparam=pciex1_gen=3`?

ralf-9000, 2025-04-02:
Yes, no influence ....

P33M, 2025-04-02:
Aha. I have an SK Hynix SSD in my collection (random OEM) and I get a failure to probe on 6.12. Furthermore, about half the drives seem to come up with 1 queue enabled and not 4 (== number of CPUs).

P33M, 2025-04-02:
Same drive works on 6.12 with a Pi 5 8GB and HAT+.

ralf-9000, 2025-04-02:
The hat or CM4 carrier board does not matter in mv tests: 
- SK Hynix SSD 512GB up to 6.12.18 working on all tested constellations
- SK Hynix SSD 512GB 6.12.20 NOT anymore working in all tested constellations
Because I need a solution, I have order a brother of the working WD SN530 512GB now on Ebay for 33€ :-)

But I think that has in the actual change since 6.16.18 a root cause also affecting other vendors and SSDs .....

ralf-9000, 2025-04-02:
Minor side effect (maybe helpful to find the root cause with SDDs): 

On both CM4 I have a 4-pin fan (PWM controlled):

- Speed control of fan in every constellation 6.12.18 and below (except some old versions what is known)
- If the SK Hynix SSD is installed the fan runs always full speed with 6.12.20 on SD card. 
- With the WD SN530 the speed control is active (especially on 6.12.20)
- With SK Hynix only full speed with 6.12.20 
- On boards with removed SSD cards always normal speed control.

So it looks for me like some strange side effects, but maybe it gives a hint to the root cause ....

ralf-9000, 2025-04-02:
I can do this tomorrow on the CM4 with the drive problem. Have I to mount then the 1st partition with the SDD:

/dev/nvme0n1p1 -> /boot/firmware

Because at the moment /boot/firmware is mounted on the SD card, I will umount it and mount it on NVME ....

You said "drive enumeration failure", so you found something in the changed code?

ralf-9000, 2025-04-03:
So, I have tested 6761, see the file appended, what I have done ....

SUCCESS, it's now again booting from SSD only. And the minor fan problem (mentioned above) is also away ...

One question: Can I now let this CM4 participate in the normal update/upgrade cycles?

Many thanks, Ralf

[implementing-6761.txt](https://github.com/user-attachments/files/19581319/implementing-6761.txt)

popcornmix, 2025-04-03:
> One question: Can I now let this CM4 participate in the normal update/upgrade cycles?

Yes, the fix has been merged so will be present in any future kernel builds (apt or rpi-update).

closed 2025-04-03 by ralf-9000