raspberrypi/linux #6755: Kernel 6.12 makes external USB3 drive disappear.

braymullo, 2025-04-01:
### Describe the bug

Upgrading to Kernel 6.12 kills one of my external USB3 EXT4 drives.

I get a Kernel message: kernel: /dev/disk/by-uuid/e0875765-bbc0-458e-b40a-f4d91b5fcbb1: Can't lookup blockdev

The drive doesn't show up in OMV7 or with blkid

Downgrading to 6.6 fixes the issue completely. I downgraded using sudo `rpi-update 8bd0d75837d6a99138c7cbbc24888601b220405a`
Fine to live here for a while, but eventually one must keep up to date for security patches etc.

While I'm the furthest thing from an Advanced user, since this kernel is now master, I was hoping anyone here could help shed light on how I might solve this, and what additional logs I could share to illuminate the problem.

System is: RaspiOS Lite, Latest and fully updated. OMV7 running on top. Raspi 5 4gb. Working brilliantly since Raspi5 release. This is not a migrated system. Fresh install and set up. Drive in Question is a transcend StoreJet (ST2000LM003 HN-M201RAD), about a decade old. But in perfect working condition.

My other drives, two WD MyBook External USB3's work great no matter the kernel version. Just the transcend is saying no to whatever commits happened between 6.6 and 6.12.

Thanks!

```
=== START OF INFORMATION SECTION ===
Model Family:     Seagate Samsung SpinPoint M9T
Device Model:     ST2000LM003 HN-M201RAD
Serial Number:    S344J9AG700123
LU WWN Device Id: 5 0004cf 210092cfa
Firmware Version: 2BC10001
User Capacity:    2,000,398,934,016 bytes [2.00 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    5400 rpm
Form Factor:      2.5 inches
Device is:        In smartctl database 7.3/5671
ATA Version is:   ATA8-ACS T13/1699-D revision 6
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 3.0 Gb/s)
Local Time is:    Tue Apr  1 13:21:21 2025 CEST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
AAM level is:     128 (quiet), recommended: 254
APM feature is:   Disabled
Rd look-ahead is: Enabled
Write cache is:   Enabled
DSN feature is:   Unavailable
ATA Security is:  Disabled, NOT FROZEN [SEC1]
Wt Cache Reorder: Enabled

=== START OF READ SMART DATA SECTION ===
```

### Steps to reproduce the behaviour

Downgrade to 6.6 - Drive appears, works, read and writable and is fully performant as expected.
Upgrade to 6.12 - Drive disappears completely
Downgrade to 6.6 - Drive is back

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-07-04
2025/03/27 00:16:44 
Copyright (c) 2012 Broadcom
version 897f94ed (release) (embedded)

Working:
Linux raspberrypi 6.6.77-v8-16k+ #1851 SMP PREEMPT Thu Feb 13 13:02:57 GMT 2025 aarch64 GNU/Linux

and issue on:
Linux raspberrypi 6.12.21-v8-16k+ #1868 SMP PREEMPT Mon Mar 31 17:10:57 BST 2025 aarch64 GNU/Linux

[... 976 lines cut ...]
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
```

P33M, 2025-04-01:
Please post the full output of `dmesg` when booting 6.12 with the Transcend drive attached.

braymullo, 2025-04-01:
Sure! Here ya go :) 6.12 dmesg

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.12.21-v8-16k+ (dom@buildbot) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1868 SMP PREEMPT Mon Mar 31 17:10:57 BST 2025
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000003a000000..0x000000003dffffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd16640..0x000000003fd16677 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000001fffffff] (512MB)
[    0.000000] Faking node 1 at [mem 0x0000000020000000-0x000000003fffffff] (512MB)
[    0.000000] Faking node 2 at [mem 0x0000000040000000-0x000000005fffffff] (512MB)
[    0.000000] Faking node 3 at [mem 0x0000000060000000-0x000000007fffffff] (512MB)
[    0.000000] Faking node 4 at [mem 0x0000000080000000-0x000000009fffffff] (512MB)
[    0.000000] Faking node 5 at [mem 0x00000000a0000000-0x00000000bfffffff] (512MB)
[    0.000000] Faking node 6 at [mem 0x00000000c0000000-0x00000000dfffffff] (512MB)
[    0.000000] Faking node 7 at [mem 0x00000000e0000000-0x00000000ffffffff] (512MB)
[    0.000000] NUMA: Initialized distance table, cnt=8
[    0.000000] NODE_DATA(0) allocated [mem 0x1fffd200-0x1fffffff]
[    0.000000] NODE_DATA(1) allocated [mem 0x3fbfd200-0x3fbfffff]
[    0.000000] NODE_DATA(2) allocated [mem 0x5fffd200-0x5fffffff]
[    0.000000] NODE_DATA(3) allocated [mem 0x7fffd200-0x7fffffff]
[    0.000000] NODE_DATA(4) allocated [mem 0x9fffd200-0x9fffffff]
[    0.000000] NODE_DATA(5) allocated [mem 0xbfffd200-0xbfffffff]
[    0.000000] NODE_DATA(6) allocated [mem 0xdfffd200-0xdfffffff]
[    0.000000] NODE_DATA(7) allocated [mem 0xffd9c7c0-0xffd9f5bf]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000001fffffff]
[    0.000000]   node   1: [mem 0x0000000020000000-0x000000003fbfffff]
[    0.000000]   node   2: [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   node   3: [mem 0x0000000060000000-0x000000007fffffff]
[    0.000000]   node   4: [mem 0x0000000080000000-0x000000009fffffff]
[    0.000000]   node   5: [mem 0x00000000a0000000-0x00000000bfffffff]
[    0.000000]   node   6: [mem 0x00000000c0000000-0x00000000dfffffff]
[    0.000000]   node   7: [mem 0x00000000e0000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 1 [mem 0x0000000020000000-0x000000003fbfffff]
[    0.000000] Initmem setup node 2 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] Initmem setup node 3 [mem 0x0000000060000000-0x000000007fffffff]
[    0.000000] Initmem setup node 4 [mem 0x0000000080000000-0x000000009fffffff]
[    0.000000] Initmem setup node 5 [mem 0x00000000a0000000-0x00000000bfffffff]
[    0.000000] Initmem setup node 6 [mem 0x00000000c0000000-0x00000000dfffffff]
[    0.000000] Initmem setup node 7 [mem 0x00000000e0000000-0x00000000ffffffff]
[    0.000000] On node 2, zone DMA: 256 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 15 pages/cpu s193368 r8192 d44200 u245760
[    0.000000] pcpu-alloc: s193368 r8192 d44200 u245760 alloc=15*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[... 665 lines cut ...]
[   44.325682] eth0: renamed from veth7ee7e4e
[   44.393285] eth0: renamed from veth1624bf0
[   44.412837] eth0: renamed from vethe813fb2
[   44.413332] br-bf79d99a3dcf: port 1(vethe66ac25) entered blocking state
[   44.413340] br-bf79d99a3dcf: port 1(vethe66ac25) entered forwarding state
[   44.416333] eth0: renamed from veth97a3a3f
[   44.417901] br-aebcae31c0cf: port 1(vethb124f80) entered blocking state
[   44.417908] br-aebcae31c0cf: port 1(vethb124f80) entered forwarding state
[   44.418039] br-3464f67a78fd: port 1(veth16c5792) entered blocking state
[   44.418044] br-3464f67a78fd: port 1(veth16c5792) entered forwarding state
[   44.418539] br-64a4772d6cb3: port 1(vethe119190) entered blocking state
[... cut ...]
[   48.258643] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   93.761083] Bluetooth: Core ver 2.22
[   93.761123] NET: Registered PF_BLUETOOTH protocol family
[   93.761126] Bluetooth: HCI device and connection manager initialized
[   93.761134] Bluetooth: HCI socket layer initialized
[   93.761137] Bluetooth: L2CAP socket layer initialized
[   93.761142] Bluetooth: SCO socket layer initialized
[  186.745819] /dev/disk/by-uuid/e0875765-bbc0-458e-b40a-f4d91b5fcbb1: Can't lookup blockdev

```

braymullo, 2025-04-01:
Interesting bit by comparison, can't set config#1, error -110 below:

6.6
```
[    3.511087] usb 4-1.1: Manufacturer: StoreJet Transcend
[    3.516336] usb 4-1.1: SerialNumber: S344J9AG700123      
[    3.523323] usb-storage 4-1.1:1.0: USB Mass Storage device detected
[    3.529727] scsi host0: usb-storage 4-1.1:1.0

```

6.12

```
[   13.724770] usb 4-1.1: Product: StoreJet Transcend
[   13.724772] usb 4-1.1: Manufacturer: StoreJet Transcend
**[   18.844795] usb 4-1.1: can't set config #1, error -110**
[   18.916750] usb 4-1.3: new SuperSpeed USB device number 4 using xhci-hcd
[   18.936979] usb 4-1.3: New USB device found, idVendor=1058, idProduct=25ed, bcdDevice=10.31
[   18.936982] usb 4-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
```

braymullo, 2025-04-01:
On 6.12, if I unplug it from the USB3 hub, and replug it back in, I can see the device pop up in OMV. But it's completely unusable there. It populates the device list but is uneditable. So, soooomething is happening. But not fully. I can't quite remember where  in the interface this happens and I'm not home right now to test it. Can try later, or if you have another command I should run pre-post hotswap I'd be glad to paste it here :)

And thanks so much!

P33M, 2025-04-01:
```
[   13.724761] usb 4-1.1: New USB device found, idVendor=174c, idProduct=5106, bcdDevice=80.00
[   13.724767] usb 4-1.1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[   13.724770] usb 4-1.1: Product: StoreJet Transcend
[   13.724772] usb 4-1.1: Manufacturer: StoreJet Transcend
[   18.844795] usb 4-1.1: can't set config #1, error -110
```
That's bad. This is before the usb-storage has bound to the disk's interface, so it's in the very early device setup. This is going to be a fun regression to diagnose.

Does the device behave any differently if you hotplug it after booting?

P33M, 2025-04-01:
The fact that the device "appears" but is inaccessible suggests that it failed further along. Run `sudo dmesg -C` then hotplug the drive, and run `dmesg` afterwards and post the contents.

braymullo, 2025-04-01:
Well, what was reproducible 100% of the time, time after time, for the whole day, is now not. Maddening. 

The drive appears in both 6.6 and 6.12. I'll keep an eye on it. Next time it does it, if it does it, I'll open up this issue. Thanks for your time thus far

closed 2025-04-01 by braymullo