raspberrypi/linux #6756: Pi5 Fan broken after updating to 6.12

llllllllqq, 2025-04-01:
### Describe the bug

after updateing pi5 to 6.12 kernel, my pwm fan stopped to work with the fan device disappearing.
`cat: '/sys/devices/platform/cooling_fan/hwmon/*/fan1_input': No such file or directory`

If I poweroff the pi and boot it manually, the fan can work. But the fan will stop to work if I reboot the pi.

### Steps to reproduce the behaviour

poweroff the pi
unplug power line
plug in power line to boot the pi
(now the fan works)
sudo reboot
(now the fan not works)

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4
2025/03/19 13:41:26
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)
Linux rpi5 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
BOOTLOADER: up to date
   CURRENT: Wed Mar 19 01:41:26 PM UTC 2025 (1742391686)
    LATEST: Wed Mar 19 01:41:26 PM UTC 2025 (1742391686)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)

my config.txt: 
[all]
dtparam=rtc_bbat_vchg=3000000
dtparam=fan_temp0=60000
dtparam=fan_temp0_hyst=2000
dtparam=fan_temp0_speed=75
dtparam=fan_temp1=65000
dtparam=fan_temp1_hyst=3000
dtparam=fan_temp1_speed=125
dtparam=fan_temp2=70000
dtparam=fan_temp2_hyst=4000
dtparam=fan_temp2_speed=175
dtparam=fan_temp3=75000
dtparam=fan_temp3_hyst=5000
dtparam=fan_temp3_speed=250
#led
dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
dtparam=eth_led0=4
dtparam=eth_led1=4
#btwf
dtoverlay=disable-bt
dtoverlay=disable-wifi

### Logs

[1.txt](https://github.com/user-attachments/files/19550809/1.txt)

bsimmo, 2025-04-01:
If you use a standard/default config.txt does it work?

llllllllqq, 2025-04-01:
> If you use a standard/default config.txt does it work?

No, it still does not work

pelwell, 2025-04-01:
Is this a Noctua fan? We investigated reports of some of them failing to start after a reboot - see this thread: https://github.com/raspberrypi/rpi-eeprom/issues/67

This was eventually traced to the fact that the fans wait between 2 and 5 seconds after power is applied before they start to spin. This is fine on a cold boot, especially with the new boot menu facility, but a reboot is much quicker and may give up on finding the fan before it moves.

The recommended workaround for sluggish fans is to add `dtparam=cooling_fan=on` to config.txt.

pelwell, 2025-04-01:
Also, it's unlikely to be the kernel that made the difference here - what does `vcgencmd bootloader_version` report?

llllllllqq, 2025-04-01:
> Also, it's unlikely to be the kernel that made the difference here - what does `vcgencmd bootloader_version` report?

2025/03/19 13:41:26
version cec1d3ae40f4a1cb24fe3c42d60153968695385b (release)
timestamp 1742391686
update-time 1742855338
capabilities 0x0000007f

I don't know which brand my fan is, but dtparam=cooling_fan=on does work. Thanks!
(but even without this dtparam, fan on old 6.6 kernel works fine)

pelwell, 2025-04-03:
That's the current firmware, which some time back was changed to turn off the fan after the detection phase (people objected to fans running at full speed until the kernel booted). I suspect that, before your firmware was updated, the fan would run fast for a few seconds on a cold boot before slowing, but remaining at full speed on a reboot. Having looked into the problem in depth I'm certain that there's nothing we can do to make it magically work as one would like. Waiting longer for fan detection will penalise users with either no fan or a fan that spins sooner.

If you want to measure the time it does take your fan to start spinning (note that you'll lose USB connectivity for a while):
```
$ pinctrl USB_VBUS_EN op dl; sleep 2; pinctrl poll USB_VBUS_EN,FAN_TACH | head & sleep 1; pinctrl USB_VBUS_EN op dh
```
The first sleep is to allow the fan time to come to a halt - you may need to increase it. The second is to make sure the `pinctrl poll` has a chance to start.

This is what I see with one of our cooling fans:
```
29: hi // FAN_TACH/GPIO29
42: lo // USB_VBUS_EN/GPIO42
+1000742us
42: hi // USB_VBUS_EN/GPIO42
+36377us
29: lo // FAN_TACH/GPIO29
+120187us
29: hi // FAN_TACH/GPIO29
+19891us
29: lo // FAN_TACH/GPIO29
```
What we care about is the delay between VBUS going high and FAN_TACH going low for the first time, which in this case was 36ms.

pelwell, 2025-04-03:
N.B. After running that test you should reboot, to make sure everything is back to normal.

llllllllqq, 2025-04-03:
> That's the current firmware, which some time back was changed to turn off the fan after the detection phase (people objected to fans running at full speed until the kernel booted). I suspect that, [...]

Thanks for your working. However my pi relies on USB disk to work, so I cannot help to do this test.

closed 2025-04-04 by pelwell