raspberrypi/linux #6770: After Update Firmware To 6.12.20 (RPi 4 and RPi 5) Mempolicy Error

SL-1210MK7, 2025-04-04:
---
After Update Firmware To 6.12.20 (RPi 4 and RPi 5) Mempolicy Error
---

[   91.824592] mempolicy: Request to set policy to 5 ignored
[   91.824607] mempolicy: Request to set policy to 3 ignored
[  125.682622] mempolicy: Request to set policy to 5 ignored
[  125.682636] mempolicy: Request to set policy to 3 ignored

Upgrade from 6.6.74 to 6.12.20 start to see this problem. Working fine until this new firmware and after update this error show up constantly.

timg236, 2025-04-04:
Moved to kernel

popcornmix, 2025-04-05:
See [forum post](https://forums.raspberrypi.com/viewtopic.php?t=386126).
It is harmless and there will be an apt fix soon to address this.

closed 2025-04-05 by popcornmix