raspberrypi/linux #6772: `vc4_atomic_commit_tail` compiler warning log flood on CM4S

JinShil, 2025-04-08:
### Describe the bug

On a CM4S, using kernel v6.12.20, I'm experiencing a `vc4_atomic_commit_tail` compiler warning log flood.  The log is repeatedly filled with the following:

### Steps to reproduce the behaviour

1. Create a recent Bookworm OS with pi-gen
2. Boot on a CM4S
3. Examine the output of `dmesg`

### Device (s)

Other

### System

```
$ vcgencmd version
Mar 19 2025 18:24:21
Copyright (c) 2012 Broadcom
version ca6e8171a80ea46924ffaa629250bfb482f3a02c (clean) (release) (start)
```

```
$ uname -a
Linux comfilepi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

### Logs

dmesg:
```
[  219.965430] ------------[ cut here ]------------
[  219.965448] WARNING: CPU: 0 PID: 87 at drivers/gpu/drm/vc4/vc4_kms.c:507 vc4_atomic_commit_tail+0x900/0x920 [vc4]
[  219.965523] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfkill binfmt_misc rpi_hevc_dec bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc joydev videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videodev videobuf2_common mc snd_bcm2835(C) raspberrypi_gpiomem nvmem_rmem uio_pdrv_genirq uio i2c_dev ledtrig_pattern fuse dm_mod ip_tables x_tables ipv6 rtc_ds1307 regmap_i2c vc4 v3d gpu_sched snd_soc_hdmi_codec drm_shmem_helper drm_display_helper cec drm_dma_helper drm_kms_helper i2c_brcmstb drm drm_panel_orientation_quirks i2c_bcm2835 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd backlight
[  219.965789] CPU: 0 UID: 0 PID: 87 Comm: kworker/u20:2 Tainted: G        WC         6.12.20+rpt-rpi-v8 #1  Debian 1:6.12.20-1+rpt1~bpo12+1
[  219.965808] Tainted: [W]=WARN, [C]=CRAP
[  219.965814] Hardware name: Raspberry Pi Compute Module 4S Rev 1.0 (DT)
[  219.965823] Workqueue: events_unbound commit_work [drm_kms_helper]
[  219.965904] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  219.965917] pc : vc4_atomic_commit_tail+0x900/0x920 [vc4]
[  219.965975] lr : vc4_atomic_commit_tail+0x1f4/0x920 [vc4]
[  219.966030] sp : ffffffc08113bd00
[  219.966036] x29: ffffffc08113bd00 x28: 0000000000000000 x27: 0000000000000000
[  219.966057] x26: ffffff8041210080 x25: 0000000000000038 x24: ffffff8044fd5300
[  219.966078] x23: ffffff8041210080 x22: ffffff80410beb00 x21: ffffff804558a000
[  219.966098] x20: 0000000000000006 x19: 0000000001fb00f8 x18: 0000000000000000
[  219.966118] x17: 0000000000000000 x16: ffffffe614a2d010 x15: 00000055b2b10a50
[  219.966139] x14: 0000000000000000 x13: 000000000ee6b280 x12: 0000000480000008
[  219.966159] x11: ffffffc08002d000 x10: 0000000000001a40 x9 : ffffffe614a23c74
[  219.966179] x8 : 0000000000000040 x7 : 000000000ee6b280 x6 : 0ee6b28000000004
[  219.966200] x5 : 0000000000000001 x4 : 0000000000000000 x3 : 0000000000000000
[  219.966219] x2 : ffffff804592a100 x1 : ffffff804592a100 x0 : 00000000ffffffea
[  219.966239] Call trace:
[  219.966245]  vc4_atomic_commit_tail+0x900/0x920 [vc4]
[  219.966301]  commit_tail+0xac/0x1a0 [drm_kms_helper]
[  219.966377]  commit_work+0x1c/0x30 [drm_kms_helper]
[  219.966450]  process_one_work+0x158/0x3d0
[  219.966469]  worker_thread+0x2d8/0x3e8
[  219.966483]  kthread+0x11c/0x128
[  219.966494]  ret_from_fork+0x10/0x20
[  219.966508] ---[ end trace 0000000000000000 ]---
[  219.966645] ------------[ cut here ]------------
[  219.966655] WARNING: CPU: 0 PID: 87 at drivers/gpu/drm/vc4/vc4_kms.c:508 vc4_atomic_commit_tail+0x864/0x920 [vc4]
[  219.966722] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfkill binfmt_misc rpi_hevc_dec bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc joydev videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videodev videobuf2_common mc snd_bcm2835(C) raspberrypi_gpiomem nvmem_rmem uio_pdrv_genirq uio i2c_dev ledtrig_pattern fuse dm_mod ip_tables x_tables ipv6 rtc_ds1307 regmap_i2c vc4 v3d gpu_sched snd_soc_hdmi_codec drm_shmem_helper drm_display_helper cec drm_dma_helper drm_kms_helper i2c_brcmstb drm drm_panel_orientation_quirks i2c_bcm2835 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd backlight
[  219.966981] CPU: 0 UID: 0 PID: 87 Comm: kworker/u20:2 Tainted: G        WC         6.12.20+rpt-rpi-v8 #1  Debian 1:6.12.20-1+rpt1~bpo12+1
[  219.966998] Tainted: [W]=WARN, [C]=CRAP
[  219.967004] Hardware name: Raspberry Pi Compute Module 4S Rev 1.0 (DT)
[  219.967012] Workqueue: events_unbound commit_work [drm_kms_helper]
[  219.967094] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[... 97 lines cut ...]
[  220.965074] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[  220.965094] x14: 0000000000000000 x13: 000000000ee6b280 x12: 0000000480000008
[  220.965114] x11: ffffffc08002d000 x10: 0000000000001a40 x9 : ffffffe614a23c74
[  220.965135] x8 : 0000000000000040 x7 : 000000000ee6b280 x6 : 0ee6b28000000004
[  220.965155] x5 : 0000000000000001 x4 : 0000000000000000 x3 : 0000000000000000
[  220.965174] x2 : ffffff804592a100 x1 : ffffff804592a100 x0 : 00000000ffffffea
[  220.965194] Call trace:
[  220.965201]  vc4_atomic_commit_tail+0x900/0x920 [vc4]
[  220.965257]  commit_tail+0xac/0x1a0 [drm_kms_helper]
[  220.965333]  commit_work+0x1c/0x30 [drm_kms_helper]
[  220.965407]  process_one_work+0x158/0x3d0
[  220.965423]  worker_thread+0x2d8/0x3e8
[  220.965437]  kthread+0x11c/0x128
[  220.965449]  ret_from_fork+0x10/0x20
[  220.965463] ---[ end trace 0000000000000000 ]---

{ More of the same} 

```

### Additional context

The problems seems to only occur on a CM4S.  I don't see the issue on a CM3+ or CM4.

I don't notice any behavioral anomalies.  Everything seems to work fine.  The logs are just getting flooded repeatedly with the aforementioned compiler warning.

JinShil, 2025-04-11:
After enabling DRM debug messages with `echo 0x1f | sudo tee /sys/module/drm/parameters/debug` I can see more information in dmesg:

```
[  256.971742] ------------[ cut here ]------------
[  256.971749] WARNING: CPU: 2 PID: 40 at drivers/gpu/drm/vc4/vc4_kms.c:453 vc4_atomic_commit_tail+0x8f8/0x920 [vc4]
[  256.971813] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfkill binfmt_misc rpi_hevc_dec bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) joydev vc_sm_cma(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videodev videobuf2_common mc raspberrypi_gpiomem snd_bcm2835(C) nvmem_rmem uio_pdrv_genirq uio i2c_dev ledtrig_pattern fuse dm_mod ip_tables x_tables ipv6 rtc_ds1307 regmap_i2c vc4 snd_soc_hdmi_codec drm_display_helper v3d cec drm_dma_helper gpu_sched drm_shmem_helper drm_kms_helper drm drm_panel_orientation_quirks i2c_brcmstb i2c_bcm2835 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd backlight
[  256.972088] CPU: 2 UID: 0 PID: 40 Comm: kworker/u19:0 Tainted: G        WC         6.12.20+rpt-rpi-v8 #1  Debian 1:6.12.20-1+rpt1~bpo12+1
[  256.972107] Tainted: [W]=WARN, [C]=CRAP
[  256.972113] Hardware name: Raspberry Pi Compute Module 4S Rev 1.0 (DT)
[  256.972123] Workqueue: events_unbound commit_work [drm_kms_helper]
[  256.972209] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  256.972222] pc : vc4_atomic_commit_tail+0x8f8/0x920 [vc4]
[  256.972282] lr : vc4_atomic_commit_tail+0x394/0x920 [vc4]
[  256.972338] sp : ffffffc08017bd00
[  256.972345] x29: ffffffc08017bd00 x28: 0000000000000000 x27: 0000000000000000
[  256.972368] x26: ffffff80450110f0 x25: ffffff8045011080 x24: ffffff80434c3a80
[  256.972389] x23: ffffff8047070080 x22: ffffff8045011880 x21: ffffff8045915000
[  256.972410] x20: 000000001dcd6500 x19: ffffff8045011400 x18: ffffffffffffffff
[  256.972430] x17: 65726f6320656874 x16: ffffffd77642d010 x15: 205d5d3463765b20
[  256.972450] x14: 6c6961745f74696d x13: 7a48203030303030 x12: 3030303520746120
[  256.972470] x11: 6b636f6c63206572 x10: ffffffd77740a560 x9 : ffffffd776423c74
[  256.972490] x8 : 00000000fffff369 x7 : 0000000000000000 x6 : ffffff8040808000
[  256.972510] x5 : ffffff8040808000 x4 : 0000000000000000 x3 : 0000000000000000
[  256.972530] x2 : ffffff8040808000 x1 : ffffff8040808000 x0 : 00000000ffffffea
[  256.972550] Call trace:
[  256.972558]  vc4_atomic_commit_tail+0x8f8/0x920 [vc4]
[  256.972615]  commit_tail+0xac/0x1a0 [drm_kms_helper]
[  256.972692]  commit_work+0x1c/0x30 [drm_kms_helper]
[  256.972767]  process_one_work+0x158/0x3d0
[  256.972790]  worker_thread+0x2d8/0x3e8
[  256.972805]  kthread+0x11c/0x128
[  256.972818]  ret_from_fork+0x10/0x20
[  256.972833] ---[ end trace 0000000000000000 ]---
[  256.972892] ------------[ cut here ]------------
[  256.972898] WARNING: CPU: 2 PID: 40 at drivers/gpu/drm/vc4/vc4_kms.c:454 vc4_atomic_commit_tail+0x3a8/0x920 [vc4]
[  256.972963] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfkill binfmt_misc rpi_hevc_dec bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) joydev vc_sm_cma(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videodev videobuf2_common mc raspberrypi_gpiomem snd_bcm2835(C) nvmem_rmem uio_pdrv_genirq uio i2c_dev ledtrig_pattern fuse dm_mod ip_tables x_tables ipv6 rtc_ds1307 regmap_i2c vc4 snd_soc_hdmi_codec drm_display_helper v3d cec drm_dma_helper gpu_sched drm_shmem_helper drm_kms_helper drm drm_panel_orientation_quirks i2c_brcmstb i2c_bcm2835 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd backlight
[  256.973225] CPU: 2 UID: 0 PID: 40 Comm: kworker/u19:0 Tainted: G        WC         6.12.20+rpt-rpi-v8 #1  Debian 1:6.12.20-1+rpt1~bpo12+1
[  256.973242] Tainted: [W]=WARN, [C]=CRAP
[  256.973248] Hardware name: Raspberry Pi Compute Module 4S Rev 1.0 (DT)
[  256.973257] Workqueue: events_unbound commit_work [drm_kms_helper]
[  256.973342] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  256.973355] pc : vc4_atomic_commit_tail+0x3a8/0x920 [vc4]
[  256.973414] lr : vc4_atomic_commit_tail+0x3a4/0x920 [vc4]
[  256.973470] sp : ffffffc08017bd00
[  256.973476] x29: ffffffc08017bd00 x28: 0000000000000000 x27: 0000000000000000
[  256.973497] x26: ffffff80450110f0 x25: ffffff8045011080 x24: ffffff80434c3a80
[  256.973518] x23: ffffff8047070080 x22: ffffff8045011880 x21: ffffff8045915000
[  256.973538] x20: 000000001dcd6500 x19: ffffff8045011400 x18: ffffffffffffffff
[  256.973559] x17: 65726f6320656874 x16: ffffffd77642d010 x15: 205d5d3463765b20
[  256.973579] x14: 6c6961745f74696d x13: 7a48203030303030 x12: 3030303520746120
[  256.973599] x11: 6b636f6c63206572 x10: ffffffd77740a560 x9 : ffffffd776423c74
[  256.973619] x8 : 00000000fffff369 x7 : 0000000000000000 x6 : ffffff8040808000
[  256.973639] x5 : ffffff8040808000 x4 : 0000000000000000 x3 : 0000000000000000
[  256.973658] x2 : ffffff8040808000 x1 : ffffff8040808000 x0 : 00000000ffffffea
[  256.973678] Call trace:
[  256.973685]  vc4_atomic_commit_tail+0x3a8/0x920 [vc4]
[  256.973741]  commit_tail+0xac/0x1a0 [drm_kms_helper]
[  256.973820]  commit_work+0x1c/0x30 [drm_kms_helper]
[  256.973895]  process_one_work+0x158/0x3d0
[  256.973914]  worker_thread+0x2d8/0x3e8
[  256.973929]  kthread+0x11c/0x128
[  256.973941]  ret_from_fork+0x10/0x20
[  256.973955] ---[ end trace 0000000000000000 ]---
[... cut ...]
[  316.991436] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 33 (5)
[  316.991581] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 33 (4)
[  316.991727] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 718 (3)
[  316.991870] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 717 (2)
[  316.992012] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 720 (3)
[  316.992154] vc4-drm gpu: [drm:__drm_atomic_state_free [drm]] Freeing atomic state 0000000020d52b3f
[  317.005689] vc4-drm gpu: [drm:vblank_disable_fn [drm]] disabling vblank on crtc 3
```

JinShil, 2025-04-11:
Parsing the source code and additional DRM messages, I noticed that this seemed to be a clock frequency issue.  So, I decided to try commenting out `core_freq=250` in config.txt, and that seems resolve the issue.  The kernel warnings no longer appear in the output of dmesg.  However, I believe removing that from config.txt will cause communication problems on the mini-UART.

popcornmix, 2025-04-11:
I think `core_freq_fixed=1` is the preferred way of forcing clocks avoid issues with peripheral clocks varying. Does that work for you?

comfile-mike, 2025-04-11:
I think that would be fine if it does what I think it does.  Is it documented anywhere?  I can't seem to find it.

I'm assuming then the core clock would be set to 500MHz on the CM4S, right?  At that frequency, can I get standard baud rates with a low enough margin of error to work with standard equipment?

JinShil, 2025-04-11:
I think that would be fine if it does what I think it does. Is it documented anywhere? I can't seem to find it.

I'm assuming then the core clock would be set to 500MHz on the CM4S, right? At that frequency, can I get standard baud rates with a low enough margin of error to work with standard equipment?

popcornmix, 2025-04-11:
Yes, `core_freq_fixed=1` will pin the core clock at 500MHz on CM4S which should make baud rates stable.

JinShil, 2025-04-14:
I'm uncomfortable using an undocumented feature,  Is `core_freq_fixed` something I can rely upon being there in the future?  I can't find any reference to it anywhere.  Can someone please document it before I put it into production.

JinShil, 2025-04-14:
I've confirmed that setting `core_freq_fixed=1` on a CM3+ results in a core clock of 400MHz, not 250MHz.   The concern with that is whether that frequency can be divided to standard baud rates with a small enough error to be able to reliably communicate with standard equipment.  I don't know the algorithm that is used to compute the baud rate.

See https://www.raspberrypi.com/documentation/computers/configuration.html#mini-uart-and-cpu-core-frequency  The solution there is still `core_freq=250`.

How should I configured the CM3+ and CM4S so that the mini-UART can operate with both **standard** and **fixed** baud rates, and avoid the kernel warning log flood?

pelwell, 2025-04-14:
The documentation has not been updated to include `core_freq_fixed`. If you don't need the extra VPU speed and want 250MHz for the peripherals, use 250MHz, but I see no reason why 400MHz should be worse than 250MHz for baud rate accuracy. Starting from a larger number, the integer divisions closest to a chosen baud rate will end up being closer together so on average more accurate, but they are likely to move a little and could end up being slightly worse. The important point is that 250MHz is an arbitrary value which was not chosen to improve baud rate accuracy.

The baud rate formula for the mini UART is on page 11 of the BCM2835 Peripherals datasheet, in case you want to run the numbers yourself.

JamesH65, 2025-04-14:
I'll add that command to the docs today. Might take a while for it to percolate through to the website.

comfile-mike, 2025-04-14:
What about thermal throttling?  Does `core_freq_fixed` still allow the firmware to throttle down the core clock should temperatures get too hot?

popcornmix, 2025-04-14:
>What about thermal throttling? Does core_freq_fixed=1 still allow the firmware to throttle down the core clock should temperatures get too hot?

The 400MHz core frequency is a `required` frequency, so won't change with throttling.
The arm frequency will still be reduced with throttling.

6by9, 2025-04-14:
This has made me take a double-take at https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/gpu/drm/vc4/vc4_kms.c#L444-L445
Asking for a minimum of 500MHz regardless of scene complexity or output resolution sounds odd, particularly as the driver does attempt to calculate the minimum clock required. That appears to only do anything if you are needing the 550MHz for 4k60 or 600MHz for 4096x2160.

popcornmix, 2025-04-14:
While initial post is about CM4S (which will use 500MHz for a fixed core clock),
the conversation did switch to 
>  I've confirmed that setting core_freq_fixed=1 on a CM3+ results in a core clock of 400MHz, not 250MHz.

So, the 400MHz is for CM3.

The code you linked is for Pi4 and is required. There is some quirk in the hardware that says a "higher" clock is needed when switching, even is the before and after frequencies are lower. I don't know if 500MHz is the minimum, but it was "enough". Hard coding 500MHz does feel a bit ugly and I wonder if just the usual turbo core clock frequency would be fine (`max_core_rate`?)

JinShil, 2025-04-25:
`core_freq_fixed=1` appears to resolve this issue for my needs, but I'm assuming the log flood in the case of an explicitly set `core_freq` is still undesirable.  I'll leave this issue open, but feel free to close if it won't be addressed or if there are other plans to address it.