raspberrypi/linux #6774: ashmem dkms breaks after kernel upgrade

Botspot, 2025-04-09:
### Describe the bug

After the kernel update, all apt-get actions fail because compiling ashmem for the kernel always fails.
The `make` log near the bottom of this post seems to indicate an issue with the source code that was not caught before pushing to production. Hopefully this is a simple fix that gets priority due to how many users it likely effects.

### Steps to reproduce the behaviour

Upgrade the kernel

### Device (s)

Raspberry Pi 5

### System

```
$ uname -a
Linux raspberrypi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
$ vcgencmd version
2025/03/19 13:41:26 
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-11-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 146006916383be441d524b376f6e162dba854f4a, stage4
```

### Logs

```
pi@raspberrypi:~ $ sudo apt full-upgrade --autoremove
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
8 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] 
Setting up linux-image-6.12.20+rpt-rpi-2712 (1:6.12.20-1+rpt1~bpo12+1) ...
I: /initrd.img.old is now a symlink to boot/initrd.img-6.12.20+rpt-rpi-2712
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.12.20+rpt-rpi-2712.
Sign command: /lib/modules/6.12.20+rpt-rpi-2712/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.12.20+rpt-rpi-2712 all KERNEL_SRC=/lib/modules/6.12.20+rpt-rpi-2712/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-2712 (aarch64)
Consult /var/lib/dkms/anbox-ashmem/1/build/make.log for more information.
Sign command: /lib/modules/6.12.20+rpt-rpi-2712/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.12.20+rpt-rpi-2712 all KERNEL_SRC=/lib/modules/6.12.20+rpt-rpi-2712/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-2712 (aarch64)
Consult /var/lib/dkms/anbox-binder/1/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.12.20+rpt-rpi-2712 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.12.20+rpt-rpi-2712 (--configure):
 installed linux-image-6.12.20+rpt-rpi-2712 package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-image-rpi-2712:
 linux-image-rpi-2712 depends on linux-image-6.12.20+rpt-rpi-2712 (= 1:6.12.20-1+rpt1~bpo12+1); however:
  Package linux-image-6.12.20+rpt-rpi-2712 is not configured yet.
[... 81 lines cut ...]
  CC [M]  /var/lib/dkms/anbox-ashmem/1/build/ashmem.o
/var/lib/dkms/anbox-ashmem/1/build/deps.c:64:5: warning: no previous prototype for ‘shmem_zero_setup’ [-Wmissing-prototypes]
   64 | int shmem_zero_setup(struct vm_area_struct *vma)
      |     ^~~~~~~~~~~~~~~~
/var/lib/dkms/anbox-ashmem/1/build/ashmem.c: In function ‘ashmem_init’:
/var/lib/dkms/anbox-ashmem/1/build/ashmem.c:882:9: error: implicit declaration of function ‘register_shrinker’; did you mean ‘register_chrdev’? [-Werror=implicit-function-declaration]
  882 |         register_shrinker(&ashmem_shrinker, "android-ashmem");
      |         ^~~~~~~~~~~~~~~~~
      |         register_chrdev
/var/lib/dkms/anbox-ashmem/1/build/ashmem.c: In function ‘ashmem_exit’:
/var/lib/dkms/anbox-ashmem/1/build/ashmem.c:892:9: error: implicit declaration of function ‘unregister_shrinker’; did you mean ‘unregister_chrdev’? [-Werror=implicit-function-declaration]
  892 |         unregister_shrinker(&ashmem_shrinker);
      |         ^~~~~~~~~~~~~~~~~~~
      |         unregister_chrdev
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /var/lib/dkms/anbox-ashmem/1/build/ashmem.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/anbox-ashmem/1/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
make: *** [Makefile:12: all] Error 2
```

### Additional context

Reported by multiple users - https://forums.raspberrypi.com/viewtopic.php?p=2308733#p2308733

Botspot, 2025-04-09:
This may be a larger issue with dkms in general, based on this similar issue related to v4l2loopback-dkms reported here: https://forums.raspberrypi.com/viewtopic.php?p=2307084&hilit=dkms#p2307084

pelwell, 2025-04-09:
Please describe your system - which OS are you running, etc

Botspot, 2025-04-09:
I must have installed these dkms packages from this now-archived repo and forgot they were still enabled. https://github.com/anbox/anbox-modules
The code there most likely is now deprecated enough that it no longer compiles on the latest kernel. This would be a problem if I used Anbox, but I don't, so I will simply revert the installation steps, add a pi-apps error check to give a useful suggestion to anyone else who used anbox encountering this error, and close this issue here, as it's not your fault.

closed 2025-04-09 by Botspot