raspberrypi/linux #6776: Camera Module V2 "broken" in 6.12.20

tony-addyman, 2025-04-09:
### Describe the bug

After upgrading my RPi 3B using apt on April 2nd the kernel was upgraded from the previous standard issued kernel to 6.12.20. With the 6.12.20 kernel the camera is still able to produce images using rpicam-jpeg, for example, but at most resolutions the result is not what was requested, especially if the image requested is 16x9, rather than 4x3. The image appears to have been produced as 4x3 and simply stretched to be 16x9, or a 4x3 image may be a higher resolution. The problem is not limited to rpicam-apps. It affects the picam video software too, for example.

### Steps to reproduce the behaviour

1. Obtain a Raspberry Pi with camera module v2 attached. (I have seen the problem with both 3B and 3B+)
2. Flash Raspberry PI OS November 2024 image, either 32 bit or 64 bit. I used the Desktop version for my testing.
3. Boot the Pi, but don't apply any software updates.
4. Create an image using rpicam-jpeg at 1280x720 (say). Selecting 640x480 won't demonstrate the bug.
5. Apply the software updates.
6. Create another image using the same command, and observe the distortion.
7. To confirm it is the kernel that is the issue, you repeat step 2, then apply an apt hold to the linux-headers-rpi and linux-image-rpi packages.
8. Apply the software updates, and create an image again. There will be no distortion.

### Device (s)

Raspberry Pi 3 Mod. B, Raspberry Pi 3 Mod. B+

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b15527e15ba4d9e65126a454bfec2040b909dc1c, stage4

$ vcgencmd version
Mar 19 2025 18:25:26 
Copyright (c) 2012 Broadcom
version ca6e8171a80ea46924ffaa629250bfb482f3a02c (clean) (release) (start)

$ uname -a
Linux camera 6.12.20+rpt-rpi-v7 #1 SMP Raspbian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) armv7l GNU/Linux
tony@camera:~ $ 

The raspinfo output is in a pastebin. The link is: https://pastebin.com/LHU8Jd0P

I have provided the 3B+ information as it is a "clean" system that has just been used to try and identify the source of the distortion problem. The 3B is a live system that has a lot of additional packages installed, and it is not very accessible - fastened to a wall and 5 metres off the ground.

anujdeshpande, 2025-04-15:
@tony-addyman I ran into this issue as well and some engineers from RPi seem to have identified the issue. We are discussing that here - https://forums.raspberrypi.com/viewtopic.php?p=2309958#p2309937

6by9, 2025-04-16:
Fixed through #6793