raspberrypi/linux #6777: Device 'grafx' does not have a release() function

Noltari, 2025-04-10:
### Describe the bug

bcm2711 devices show the following trace:
```
[    0.216800] ------------[ cut here ]------------
[    0.216807] Device 'grafx' does not have a release() function, it is broken and must be fixed. See Documentation/core-api/kobject.rst.
[    0.216863] WARNING: CPU: 0 PID: 1 at drivers/base/core.c:2516 device_release+0x80/0x90
[    0.216887] Modules linked in:
[    0.216896] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.6.85 #0
[    0.216905] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[    0.216911] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    0.216919] pc : device_release+0x80/0x90
[    0.216928] lr : device_release+0x80/0x90
[    0.216936] sp : ffffffc080ccb6d0
[    0.216940] x29: ffffffc080ccb6d0 x28: 0000000000000000 x27: ffffff8040d49810
[    0.216953] x26: ffffff8040c904b8 x25: ffffff8040c904a0 x24: ffffffc0808a7f08
[    0.216964] x23: ffffffc080a1a6d8 x22: 0000000000000000 x21: ffffffc080a1afe0
[    0.216976] x20: 0000000000000000 x19: ffffff8040c900b8 x18: 000000000000009f
[    0.216987] x17: 206e656b6f726220 x16: 7369207469202c6e x15: ffffffc080b4ee88
[    0.216998] x14: 00000000000001dd x13: 000000000000009f x12: 00000000ffffffea
[    0.217009] x11: 00000000ffffefff x10: ffffffc080ba6e88 x9 : ffffffc080b4ee30
[    0.217020] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
[    0.217031] x5 : 0000000000000fff x4 : 0000000000000000 x3 : ffffffc080ccb4d0
[    0.217042] x2 : ffffffc080b4edc8 x1 : ffffffc080b4edc8 x0 : 000000000000007a
[    0.217053] Call trace:
[    0.217058]  device_release+0x80/0x90
[    0.217067]  kobject_put+0xac/0x100
[    0.217078]  put_device+0x14/0x24
[    0.217086]  genpd_remove+0x16c/0x244
[    0.217093]  pm_genpd_remove+0x2c/0x50
[    0.217099]  bcm2835_power_probe+0x1f8/0x2b8
[    0.217107]  platform_probe+0x68/0xc4
[    0.217118]  really_probe+0x148/0x2b0
[    0.217125]  __driver_probe_device+0x78/0x128
[    0.217131]  driver_probe_device+0x40/0xdc
[    0.217138]  __device_attach_driver+0xb8/0xf8
[    0.217145]  bus_for_each_drv+0x70/0xb8
[    0.217151]  __device_attach+0xa0/0x184
[    0.217158]  device_initial_probe+0x14/0x20
[    0.217165]  bus_probe_device+0xac/0xb0
[    0.217171]  device_add+0x5a8/0x760
[    0.217177]  platform_device_add+0x184/0x240
[    0.217184]  mfd_add_device+0x3e8/0x4c0
[    0.217195]  devm_mfd_add_devices+0xa0/0x118
[    0.217202]  bcm2835_pm_probe+0x144/0x1d8
[    0.217210]  platform_probe+0x68/0xc4
[    0.217217]  really_probe+0x148/0x2b0
[    0.217224]  __driver_probe_device+0x78/0x128
[    0.217230]  driver_probe_device+0x40/0xdc
[    0.217237]  __driver_attach+0x90/0x160
[    0.217243]  bus_for_each_dev+0x64/0xa4
[    0.217249]  driver_attach+0x24/0x30
[    0.217256]  bus_add_driver+0xe4/0x208
[    0.217262]  driver_register+0x5c/0x124
[    0.217269]  __platform_driver_register+0x28/0x34
[    0.217276]  bcm2835_pm_driver_init+0x1c/0x28
[    0.217286]  do_one_initcall+0x6c/0x1fc
[    0.217294]  kernel_init_freeable+0x208/0x2e4
[    0.217306]  kernel_init+0x28/0x1e4
[    0.217313]  ret_from_fork+0x10/0x20
[    0.217320] ---[ end trace 0000000000000000 ]---
```

### Steps to reproduce the behaviour

Install OpenWrt snapshot with kernel 6.6.85 from: https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/
Boot and see the trace.

### Device (s)

[... 343 lines cut ...]
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r29139+38-65b8a978de
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------

 OpenWrt recently switched to the "apk" package manager!

 OPKG Command           APK Equivalent      Description
 ------------------------------------------------------------------
 opkg install <pkg>     apk add <pkg>       Install a package
 opkg remove <pkg>      apk del <pkg>       Remove a package
 opkg upgrade           apk upgrade         Upgrade all packages
 opkg files <pkg>       apk info -L <pkg>   List package contents
 opkg list-installed    apk info            List installed packages
 opkg update            apk update          Update package lists
 opkg search <pkg>      apk search <pkg>    Search for packages
 ------------------------------------------------------------------

For more https://openwrt.org/docs/guide-user/additional-software/opkg-to-apk-cheatsheet

root@OpenWrt:~#

Noltari, 2025-04-10:
> 6.6 is now dropped in favour of 6.12 as the latest LTS. Any chance of retesting on 6.12?

OpenWrt 24.10 users kernel 6.6, so I could retest this on 6.12 but it would only be useful for the next OpenWrt release, which isn't going to happen probably until 2026...
Ideally we should find a solution for this on 6.6, but since it's harmless I think we can live with that...

6by9, 2025-04-10:
Your comment at https://github.com/openwrt/openwrt/pull/18443#issuecomment-2791711948 says that this has only just started happening. You didn't say as much here, or link to that issue until just now.

Do you use defconfigs from this repo?
Actually you must be applying your own set of patches to a mainline tree to get 6.6.86, when our latest is 6.6.78. Diffing those trees is going to be pretty unlikely for us to investigate.

I'm going to guess it will be a defconfig thing where one has a driver built-in vs as a module. Raspberry Pi OS has bcm2835-power initialise at boot+0.919. `bcm2835_power`, `clk_raspberrypi`, firmware `raspberrypi` are all built-in in our normal defconfigs.

Noltari, 2025-04-10:
> Your comment at [openwrt/openwrt#18443 (comment)](https://github.com/openwrt/openwrt/pull/18443#issuecomment-2791711948) says that this has only just started happening. You didn't say as much here, or link to that issue until just now.

Actually this was reported by @schuettecarsten and I just investigated whether it was due to the 6.6.86 update or not.
I'm using a RPi 5 as my main OpenWrt router, so I don't know when this started hapenning on RPi 4.

> Do you use defconfigs from this repo? Actually you must be applying your own set of patches to a mainline tree to get 6.6.86, when our latest is 6.6.78. Diffing those trees is going to be pretty unlikely for us to investigate.

No, we use our configs, which are updated from time to time with the configs from this repo:
https://github.com/openwrt/openwrt/blob/main/target/linux/bcm27xx/config-6.6
https://github.com/openwrt/openwrt/blob/main/target/linux/bcm27xx/bcm2711/config-6.6

> I'm going to guess it will be a defconfig thing where one has a driver built-in vs as a module. Raspberry Pi OS has bcm2835-power initialise at boot+0.919. `bcm2835_power`, `clk_raspberrypi`, firmware `raspberrypi` are all built-in in our normal defconfigs.

lategoodbye, 2025-04-12:
Not sure this is relevant, but this [commit](https://github.com/torvalds/linux/commit/f44fa354a0715577ca32b085f6f60bcf32c748dd) has recently backported.

schuettecarsten, 2025-04-13:
> Not sure this is relevant, but this [commit](https://github.com/torvalds/linux/commit/f44fa354a0715577ca32b085f6f60bcf32c748dd) has recently backported.

@Noltari

Noltari, 2025-04-22:
> Not sure this is relevant, but this [commit](https://github.com/torvalds/linux/commit/f44fa354a0715577ca32b085f6f60bcf32c748dd) has recently backported.

That commit doesn't apply to this issue because it's a fix for another commit introduced upstream in 6.8, which isn't present on downstream 6.6.

Noltari, 2025-04-22:
> I'm going to guess it will be a defconfig thing where one has a driver built-in vs as a module. Raspberry Pi OS has bcm2835-power initialise at boot+0.919. `bcm2835_power`, `clk_raspberrypi`, firmware `raspberrypi` are all built-in in our normal defconfigs.

Those 3 are built-in on OpenWrt too, so I don't think that's the issue.

After a bit of debugging I found out that v3d is being deferred at the first bcm2835-power try:
```
[    0.216943] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=grafx
[    0.216959] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    0.216966] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    0.216977] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=v3d
[    0.216988] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-517
[    0.216995] bcm2835-power bcm2835-power: bcm2835_power_probe: bcm2835_init_power_domain=-517
[    0.217002] bcm2835-power bcm2835-power: dom->name = grafx
[    0.217012]  grafx: dev->release = 0000000000000000
[    0.217018]  grafx: dev->type = 0000000000000000
[    0.217023]  grafx: dev->class = 0000000000000000
[    0.217029] ------------[ cut here ]------------
```

However, since grafx has already been added in bcm2835_init_power_domain (pm_genpd_init), it must be removed in bcm2835_power_probe (fail: pm_genpd_remove), and that's were the trace arises.

Interestingly, on the second try of bcm2835-power, the v3d has no clock and it isn't deferred:
```
[    2.239052] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=grafx
[    2.246300] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.253174] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.261709] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=v3d
[    2.268760] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.277287] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=image
[    2.284507] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.291381] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.299908] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=peri_image
[    2.307568] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=000000000f496c03
[    2.316096] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=isp
[    2.323142] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=000000001048902e
[    2.331669] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=h264
[    2.338805] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000021a60301
[    2.347333] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=usb
[    2.354376] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.361248] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.369772] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=dsi0
[    2.376905] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.383772] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.392296] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=dsi1
[    2.399430] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.406302] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.414825] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=cam0
[    2.421954] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.428826] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.437351] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=cam1
[    2.444481] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.451352] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.459876] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=ccp2tx
[    2.467186] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.474054] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.482578] bcm2835-power bcm2835-power: bcm2835_init_power_domain: name=hdmi
[    2.489711] bcm2835-power bcm2835-power: bcm2835_init_power_domain: ret=-2
[    2.496583] bcm2835-power bcm2835-power: bcm2835_init_power_domain: dom->clk=0000000000000000
[    2.505116] bcm2835-power bcm2835-power: bcm2835_power_probe: devm_reset_controller_register=0
[    2.513724] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
```

After taking a look at the .dts files, I can see why this is only triggered on bcm2711 and not on bcm2712:
https://github.com/raspberrypi/linux/blob/bba53a117a4a5c29da892962332ff1605990e17a/arch/arm/boot/dts/broadcom/bcm2711.dtsi#L117-L120
The clocks are commented on bcm2712...
https://github.com/raspberrypi/linux/blob/bba53a117a4a5c29da892962332ff1605990e17a/arch/arm64/boot/dts/broadcom/bcm2712.dtsi#L367-L370

6by9, 2025-04-22:
I think this is one we wait for upstream to clean up - they seem to be on it.

Noltari, 2025-04-22:
> I think this is one we wait for upstream to clean up - they seem to be on it.

Sure, but in the meantime we will fix it downstream on OpenWrt:
https://github.com/openwrt/openwrt/pull/18559

Thanks for the help @6by9

closed 2025-04-22 by Noltari