raspberrypi/linux #6778: dtoverlay sdio-pi5 broken on 6.12.20

giampiero7, 2025-04-10:
### Describe the bug

Loading the sdio-pi5 overlay fails both on RPi 5 as well as CM 5

### Steps to reproduce the behaviour

```
sudo dtoverlay sdio-pi5
```

### Device (s)

Raspberry Pi CM5 Lite, Raspberry Pi 5

### System

```
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2
```

```
2024/12/18 18:28:16 
Copyright (c) 2012 Broadcom
version cdaa755c (release) (embedded)
```

```
Linux SYOBKO 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

### Logs

```
[  722.736648] rp1-sdio-clk 1f000b0004.sdio_clk0: loaded sdio_clk0
[  722.744457] Unable to handle kernel execute from non-executable memory at virtual address ffffd06f5d4950d8
[  722.744472] Mem abort info:
[  722.744474]   ESR = 0x000000008600000f
[  722.744477]   EC = 0x21: IABT (current EL), IL = 32 bits
[  722.744482]   SET = 0, FnV = 0
[  722.744485]   EA = 0, S1PTW = 0
[  722.744488]   FSC = 0x0f: level 3 permission fault
[  722.744491] swapper pgtable: 16k pages, 47-bit VAs, pgdp=000000000148c000
[  722.744496] [ffffd06f5d4950d8] pgd=1000000001cc0003, p4d=1000000001cc0003, pud=1000000001cc0003, pmd=1000000060768003, pte=0060000020d74f83
[  722.744509] Internal error: Oops: 000000008600000f [#1] PREEMPT SMP
[  722.744515] Modules linked in: sdhci_of_dwcmshc sg stratopimax(O) algif_hash algif_skcipher af_alg bnep binfmt_misc spidev brcmfmac_wcc hci_uart btbcm bluetooth vc4 aes_ce_blk aes_ce_cipher ghash_ce gf128mul sha2_ce sha256_arm64 sha1_ce ecdh_generic sha1_generic snd_soc_hdmi_codec ecc libaes raspberrypi_hwmon drm_display_helper brcmfmac v3d cec brcmutil drm_dma_helper rpi_hevc_dec snd_soc_core i2c_brcmstb pisp_be v4l2_mem2mem spi_bcm2835 gpu_sched videobuf2_dma_contig cfg80211 snd_compress snd_pcm_dmaengine drm_shmem_helper snd_pcm videobuf2_memops videobuf2_v4l2 snd_timer gpio_keys videodev drm_kms_helper snd rfkill videobuf2_common dwc2 mc rp1_pio i2c_designware_platform rp1_mailbox rp1 raspberrypi_gpiomem rp1_adc i2c_designware_core nvmem_rmem drm dm_mod fuse i2c_dev drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 uio_pdrv_genirq uio
[  722.744616] CPU: 2 UID: 0 PID: 2462 Comm: kworker/u19:2 Tainted: G           O       6.12.20+rpt-rpi-2712 #1  Debian 1:6.12.20-1+rpt1~bpo12+1
[  722.744626] Tainted: [O]=OOT_MODULE
[  722.744629] Hardware name: Raspberry Pi Compute Module 5 Lite Rev 1.0 (DT)
[  722.744633] Workqueue: async async_run_entry_fn
[  722.744643] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  722.744649] pc : sdhci_dwcmshc_ops+0x0/0xfffffffffffdaf28 [sdhci_of_dwcmshc]
[  722.744660] lr : dwcmshc_probe+0x24c/0x6f8 [sdhci_of_dwcmshc]
[  722.744668] sp : ffffc00083ba3c20
[  722.744671] x29: ffffc00083ba3c20 x28: 0000000000000000 x27: 0000000000000000
[  722.744678] x26: ffffd06f5d494308 x25: ffff8000027aaa40 x24: ffff8000027aaa80
[  722.744685] x23: ffffd06f5d494cc8 x22: ffff800003c52c10 x21: ffff800003c52c00
[  722.744692] x20: ffff8000027aa540 x19: 0000000000000000 x18: ffffffffffffffff
[  722.744698] x17: 0000000000000000 x16: ffffd06fceeb3100 x15: ffff8000027aa590
[  722.744705] x14: ffffffffffffffff x13: 0000000000000018 x12: 0101010101010101
[  722.744712] x11: 7f7f7f7f7f7f7f7f x10: 0000506fae62ea60 x9 : ffffd06f5d46e0fc
[  722.744719] x8 : 0101010101010101 x7 : 0079636e65757165 x6 : 15070f011817415e
[  722.744726] x5 : 5e411718010f0715 x4 : 0000000000000000 x3 : ffffd06f5d4950d8
[  722.744732] x2 : ffff8000027aaa80 x1 : ffff8000027aa540 x0 : ffff800003c52c10
[  722.744739] Call trace:
[  722.744742]  sdhci_dwcmshc_ops+0x0/0xfffffffffffdaf28 [sdhci_of_dwcmshc]
[  722.744750]  platform_probe+0x70/0xd0
[  722.744759]  really_probe+0xc4/0x2a8
[  722.744765]  __driver_probe_device+0x80/0x140
[  722.744771]  driver_probe_device+0x44/0x170
[  722.744776]  __driver_attach_async_helper+0x54/0xc8
[  722.744782]  async_run_entry_fn+0x3c/0xf0
[  722.744786]  process_one_work+0x158/0x3d0
[  722.744793]  worker_thread+0x2d8/0x3e8
[  722.744799]  kthread+0x11c/0x128
[  722.744804]  ret_from_fork+0x10/0x20
[  722.744811] Code: 00000000 00000000 00000000 00000000 (00000000) 
[  722.744817] ---[ end trace 0000000000000000 ]---
```

JackieHIT, 2025-04-15:
I also found the same issue with my Raspberry Pi 5

giampiero7, 2025-04-15:
Thanks, I confirm it works.

closed 2025-04-15 by giampiero7