raspberrypi/linux #6780: Kernel 6.12.20+rpt-rpi-v8 appears to have broken WPA3 personal support

chrisruk, 2025-04-11:
### Describe the bug

Upgrading via:
```
sudo apt update
sudo apt upgrade
```

To kernel 6.12.20+rpt-rpi-v8, appears to have broken WPA3 personal support.  With the previous kernel
I had used `sudo nmtui` to successfully connect to WPA3 AP.

### Steps to reproduce the behaviour

Upgrade to latest kernel

### Device (s)

Raspberry Pi 4 Mod. B

### System

`6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux`

```Mar 19 2025 18:24:21 
Copyright (c) 2012 Broadcom
version ca6e8171a80ea46924ffaa629250bfb482f3a02c (clean) (release) (start)```

pelwell, 2025-04-16:
It's working for me:
```
Linux raspberrypi 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```
What do the following commands report?
```
$ dmesg | grep brcmfmac
$ cat /sys/kernel/debug/ieee80211/phy0/features
$ journalctl -b | grep -E "(Network|wpa)"
```

chrisruk, 2025-04-16:
Interesting, thanks for the reply.

From my router I notice the following relating to my Pi's wifi MAC address -
```
hostapd: SAE: <MAC> indicates support for SAE H2E, but did not use it
```

I'd used 'nmtui' to add the wifi ap previously which worked fine originally, and then after an upgrade my 
2 pis fail to connect to the same wifi ap.

dmesg | grep brcmfmac

```
[    4.365325] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.377196] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.377765] usbcore: registered new interface driver brcmfmac
[    4.663268] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    4.663668] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    5.777197] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

```
Features: 00640896
	MCHAN
	PNO
	P2P
	SCAN_RANDOM_MAC
	MFP
	DOT11H
	DUMP_OBSS
	SAE_EXT

Quirks:   00000000
```

In nmtui for the wifi connection I can see WPA3 personal, the ssid and the asterisks for the passphrase.  I try and activate it, but it says connecting for a while then pops up the password dialog and I try entering the passphrase again, and eventually get
```
Could not activate connection:  Activation failed: Secrets were required, but not provided 
```

```
Apr 16 18:38:50 pi1 wpa_supplicant[618]: wlan0: Trying to associate with SSID 'home'
Apr 16 18:38:50 pi1 NetworkManager[616]: <info>  [1744825130.6171] device (wlan0): supplicant interface state: scanning -> associating
Apr 16 18:38:50 pi1 NetworkManager[616]: <info>  [1744825130.6171] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Apr 16 18:38:50 pi1 wpa_supplicant[618]: wlan0: PMKSA-CACHE-ADDED <MAC> 0
Apr 16 18:38:50 pi1 wpa_supplicant[618]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Apr 16 18:38:50 pi1 wpa_supplicant[618]: wlan0: PMKSA-CACHE-REMOVED <MAC> 0
Apr 16 18:38:50 pi1 wpa_supplicant[618]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="home" auth_failures=1 duration=10 reason=CONN_FAILED
Apr 16 18:38:50 pi1 NetworkManager[616]: <info>  [1744825130.7080] device (wlan0): supplicant interface state: associating -> disconnected
Apr 16 18:38:50 pi1 NetworkManager[616]: <info>  [1744825130.7081] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Apr 16 18:39:08 pi1 NetworkManager[616]: <info>  [1744825148.1879] device (wlan0): supplicant interface state: disconnected -> scanning
Apr 16 18:39:08 pi1 NetworkManager[616]: <info>  [1744825148.1880] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Apr 16 18:39:08 pi1 NetworkManager[616]: <warn>  [1744825148.8481] device (wlan0): Activation: (wifi) association took too long
Apr 16 18:39:08 pi1 NetworkManager[616]: <info>  [1744825148.8482] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Apr 16 18:39:08 pi1 NetworkManager[616]: <warn>  [1744825148.8489] device (wlan0): Activation: (wifi) asking for new secrets
Apr 16 18:39:08 pi1 NetworkManager[616]: <warn>  [1744825148.8494] device (wlan0): no secrets: No agents were available for this request.
Apr 16 18:39:08 pi1 NetworkManager[616]: <info>  [1744825148.8495] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Apr 16 18:39:08 pi1 NetworkManager[616]: <warn>  [1744825148.8504] device (wlan0): Activation: failed for connection 'conn1'
Apr 16 18:39:08 pi1 NetworkManager[616]: <info>  [1744825148.8510] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Apr 16 18:39:08 pi1 wpa_supplicant[618]: wlan0: Reject scan trigger since one is already pending
Apr 16 18:39:08 pi1 NetworkManager[616]: <info>  [1744825148.8562] manager: startup complete
Apr 16 18:39:08 pi1 systemd[1]: Finished NetworkManager-wait-online.service - Network Manager Wait Online.
Apr 16 18:39:08 pi1 systemd[1]: Reached target network-online.target - Network is Online.
Apr 16 18:39:13 pi1 NetworkManager[616]: <info>  [1744825153.9578] device (wlan0): supplicant interface state: scanning -> disconnected
Apr 16 18:39:13 pi1 NetworkManager[616]: <info>  [1744825153.9578] device (p2p-dev-wlan0): supplicant management interface state: scanning -> disconnected
```

chrisruk, 2025-04-22:
I've been trying to narrow down which kernel/firmware combinations work for me, so far I have found -
 
* a6d04ba - Jan 27 - works

* dab3d92 - Feb 24 - doesn't work

* bbcf701 - Mar 18 - doesn't work