raspberrypi/linux #6782: 6.15.y NOTE: Does not boot on a Pi5

Dark-Sky, 2025-04-12:
### Describe the bug

Has this been fixed yet?

### Steps to reproduce the behaviour

Try to boot 6.15.y successfully on pi5.

### Device (s)

Raspberry Pi 5

### System

[ray@jellyfin ~]$ sudo vcgencmd version
2025/03/19 13:41:26 
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)

pelwell, 2025-04-12:
It boots for me, but without functional Ethernet.

pelwell, 2025-04-12:
You should - this branch is still at the playground stage.

Dark-Sky, 2025-04-14:
6.15-rc1 seems to be ok. Thanks Phil!

popcornmix, 2025-04-14:
I've rebased onto -rc2 which includes the fix upstream and that works for me.
I've removed the "NOTE: Does not boot on a Pi5" commit.

Dark-Sky, 2025-04-14:
Additional:

I have been getting some system wide freezing and one time I was able to click reboot but it never would shutdown waiting for some processes to shut down with both -rc1 and -rc2 16k kernels.

by, 2025-04-15:
Same here with the freezes, and it takes some effort to actually compile it (plenty of `gcc: fatal error: Killed signal terminated program cc1 compilation terminated.`) which did not happen up to and including 6.14.

Dark-Sky, 2025-04-15:
I was able to capture some logs. At first I thought it might be an issue with chromium and 16k as had that in the past. The last time I was only using firefox. It might be 20 minutes or 3 hours before it freezes everything except the mouse and am unable to reboot sometimes and other times it will not reboot or shutdown with it trying to close services.

Logs are big so will do one here and the other in next post.

journalctl:

```
Apr 15 09:57:17 jellyfin kernel: Unable to handle kernel paging request at virtual address ffff800083c23ff8
Apr 15 09:57:17 jellyfin kernel: Mem abort info:
Apr 15 09:57:17 jellyfin kernel:   ESR = 0x0000000096000007
Apr 15 09:57:17 jellyfin kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Apr 15 09:57:17 jellyfin kernel:   SET = 0, FnV = 0
Apr 15 09:57:17 jellyfin kernel:   EA = 0, S1PTW = 0
Apr 15 09:57:17 jellyfin kernel:   FSC = 0x07: level 3 translation fault
Apr 15 09:57:17 jellyfin kernel: Data abort info:
Apr 15 09:57:17 jellyfin kernel:   ISV = 0, ISS = 0x00000007, ISS2 = 0x00000000
Apr 15 09:57:17 jellyfin kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Apr 15 09:57:17 jellyfin kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Apr 15 09:57:17 jellyfin kernel: swapper pgtable: 16k pages, 48-bit VAs, pgdp=0000000001dec000
Apr 15 09:57:17 jellyfin kernel: [ffff800083c23ff8] pgd=10000000023f0003, p4d=10000000023f0003, pud=10000001c001c403, pmd=1000000140448403, pte=0000000000000000
Apr 15 09:57:17 jellyfin kernel: Internal error: Oops: 0000000096000007 [#1]  SMP
Apr 15 09:57:17 jellyfin kernel: Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device vhost_vsock vmw_vsock_virtio_transport_common vhost vhost_iotlb vsock zram algif_hash algif_skcipher af_alg bnep binfmt_misc joydev spidev hci_uart btbcm bluetooth ecdh_generic ecc brcmfmac_wcc aes_ce_blk aes_ce_cipher ghash_ce gf128mul raspberrypi_hwmon rpi_hevc_dec sha2_ce pisp_be v4l2_mem2mem videobuf2_dma_contig sha256_arm64 vc4 sha1_ce sha1_generic snd_soc_hdmi_codec drm_exec drm_display_helper videobuf2_memops brcmfmac brcmutil cec videobuf2_v4l2 i2c_brcmstb drm_client_lib drm_dma_helper v3d gpu_sched spi_bcm2835 snd_soc_core drm_shmem_helper snd_compress cfg80211 snd_pcm_dmaengine videodev rfkill snd_pcm gpio_keys snd_timer videobuf2_common pwm_fan snd drm_kms_helper rp1_pio mc raspberrypi_gpiomem rp1 rp1_mailbox rp1_adc squashfs nvmem_rmem uio_pdrv_genirq uio sch_fq_codel usbip_host usbip_core i2c_dev crypto_user drm fuse drm_panel_orientation_quirks backlight dm_mod nfnetlink ip_tables x_tables ipv6
Apr 15 09:57:18 jellyfin kernel: CPU: 1 UID: 0 PID: 66 Comm: kswapd0 Not tainted 6.15.0-rc2-1-MANJARO-RPI5 #1 PREEMPT 
Apr 15 09:57:18 jellyfin kernel: Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
Apr 15 09:57:18 jellyfin kernel: pstate: a0400009 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Apr 15 09:57:18 jellyfin kernel: pc : __pi_memcpy_generic+0x1dc/0x230
Apr 15 09:57:18 jellyfin kernel: lr : zs_obj_write+0x1a0/0x1d0
Apr 15 09:57:18 jellyfin kernel: sp : ffff80008284b150
Apr 15 09:57:18 jellyfin kernel: x29: ffff80008284b150 x28: ffff000000000000 x27: ffffff7fc1e67b40
Apr 15 09:57:18 jellyfin kernel: x26: 0000008040000000 x25: ffff000005530dc0 x24: ffff0001819d4000
Apr 15 09:57:18 jellyfin kernel: x23: ffff800083c24000 x22: ffffffffffffffef x21: ffff000005530d90
Apr 15 09:57:18 jellyfin kernel: x20: 0000000000000338 x19: ffffff7fc02fa280 x18: ffffa0df4f575c98
Apr 15 09:57:18 jellyfin kernel: x17: ffffffffffffffff x16: ffffa0df4e7f7fb8 x15: 1a2eab048aa7b5a6
Apr 15 09:57:18 jellyfin kernel: x14: 000000000000000f x13: ffbd909106540102 x12: 8da2c611d1ad8c8e
Apr 15 09:57:18 jellyfin kernel: x11: 10ac96fed482a890 x10: c078c512030815a9 x9 : 79097f56fb5b8792
Apr 15 09:57:18 jellyfin kernel: x8 : b87a6b11b7bfa131 x7 : 37fd7a2c2c636356 x6 : af0048761f760018
Apr 15 09:57:18 jellyfin kernel: x5 : ffff00002fa27d20 x4 : ffff800083c24018 x3 : ffff0001e67b7f80
Apr 15 09:57:18 jellyfin kernel: x2 : fffffffffffffca0 x1 : ffff800083c24338 x0 : ffff00002fa28000
Apr 15 09:57:18 jellyfin kernel: Call trace:
Apr 15 09:57:18 jellyfin kernel:  __pi_memcpy_generic+0x1dc/0x230 (P)
Apr 15 09:57:18 jellyfin kernel:  zram_write_page+0x15c/0x518 [zram]
Apr 15 09:57:18 jellyfin kernel:  zram_submit_bio+0x53c/0x6a0 [zram]
Apr 15 09:57:18 jellyfin kernel:  __submit_bio+0x178/0x238
Apr 15 09:57:18 jellyfin kernel:  submit_bio_noacct_nocheck+0x16c/0x388
Apr 15 09:57:18 jellyfin kernel:  submit_bio_noacct+0x260/0x558
Apr 15 09:57:18 jellyfin kernel:  submit_bio+0xac/0x238
Apr 15 09:57:18 jellyfin kernel:  submit_bio_wait+0x5c/0xf0
Apr 15 09:57:18 jellyfin kernel:  swap_writepage_bdev_sync+0xe8/0x138
Apr 15 09:57:18 jellyfin kernel:  __swap_writepage+0x100/0x2f8
Apr 15 09:57:18 jellyfin kernel:  swap_writepage+0x290/0x440
Apr 15 09:57:18 jellyfin kernel:  pageout+0x140/0x3c0
Apr 15 09:57:18 jellyfin kernel:  shrink_folio_list+0x704/0xe00
Apr 15 09:57:18 jellyfin kernel:  evict_folios+0x254/0x798
Apr 15 09:57:18 jellyfin kernel:  try_to_shrink_lruvec+0x218/0x268
Apr 15 09:57:18 jellyfin kernel:  shrink_one+0x78/0x220
Apr 15 09:57:18 jellyfin kernel:  shrink_node+0xb04/0xdc0
Apr 15 09:57:18 jellyfin kernel:  balance_pgdat+0x484/0x9b0
Apr 15 09:57:18 jellyfin kernel:  kswapd+0x218/0x4b8
Apr 15 09:57:18 jellyfin kernel:  kthread+0x14c/0x210
Apr 15 09:57:18 jellyfin kernel:  ret_from_fork+0x10/0x20
Apr 15 09:57:18 jellyfin kernel: Code: 54000169 a93f1ca6 a97f1c86 a93e24a8 (a97e2488) 
[... cut ...]
                                                 #12 0x0000fffee66b5300 n/a (libc.so.6 + 0x85300)
                                                 #13 0x0000fffee671aa0c n/a (libc.so.6 + 0xeaa0c)
                                                 ELF object binary architecture: AARCH64
Apr 15 09:57:25 jellyfin systemd[1]: systemd-coredump@0-8233-0.service: Deactivated successfully.
Apr 15 09:57:25 jellyfin systemd[1]: systemd-coredump@0-8233-0.service: Consumed 4.570s CPU time, 1G memory peak.
```

Dark-Sky, 2025-04-15:
dmesg:

```
[ 2433.863886] Unable to handle kernel paging request at virtual address ffff800083c23ff8
[ 2433.863915] Mem abort info:
[ 2433.863931]   ESR = 0x0000000096000007
[ 2433.863934]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 2433.863937]   SET = 0, FnV = 0
[ 2433.863939]   EA = 0, S1PTW = 0
[ 2433.863941]   FSC = 0x07: level 3 translation fault
[ 2433.863944] Data abort info:
[ 2433.863946]   ISV = 0, ISS = 0x00000007, ISS2 = 0x00000000
[ 2433.863948]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[ 2433.863950]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[ 2433.863953] swapper pgtable: 16k pages, 48-bit VAs, pgdp=0000000001dec000
[ 2433.863957] [ffff800083c23ff8] pgd=10000000023f0003, p4d=10000000023f0003, pud=10000001c001c403, pmd=1000000140448403, pte=0000000000000000
[ 2433.863969] Internal error: Oops: 0000000096000007 [#1]  SMP
[ 2433.945388] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device vhost_vsock vmw_vsock_virtio_transport_common vhost vhost_iotlb vsock zram algif_hash algif_skcipher af_alg bnep binfmt_misc joydev spidev hci_uart btbcm bluetooth ecdh_generic ecc brcmfmac_wcc aes_ce_blk aes_ce_cipher ghash_ce gf128mul raspberrypi_hwmon rpi_hevc_dec sha2_ce pisp_be v4l2_mem2mem videobuf2_dma_contig sha256_arm64 vc4 sha1_ce sha1_generic snd_soc_hdmi_codec drm_exec drm_display_helper videobuf2_memops brcmfmac brcmutil cec videobuf2_v4l2 i2c_brcmstb drm_client_lib drm_dma_helper v3d gpu_sched spi_bcm2835 snd_soc_core drm_shmem_helper snd_compress cfg80211 snd_pcm_dmaengine videodev rfkill snd_pcm gpio_keys snd_timer videobuf2_common pwm_fan snd drm_kms_helper rp1_pio mc raspberrypi_gpiomem rp1 rp1_mailbox rp1_adc squashfs nvmem_rmem uio_pdrv_genirq uio sch_fq_codel usbip_host usbip_core i2c_dev crypto_user drm fuse drm_panel_orientation_quirks backlight dm_mod nfnetlink ip_tables x_tables ipv6
[ 2434.033909] CPU: 1 UID: 0 PID: 66 Comm: kswapd0 Not tainted 6.15.0-rc2-1-MANJARO-RPI5 #1 PREEMPT 
[ 2434.042817] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[ 2434.048669] pstate: a0400009 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 2434.055656] pc : __pi_memcpy_generic+0x1dc/0x230
[ 2434.060291] lr : zs_obj_write+0x1a0/0x1d0
[ 2434.064313] sp : ffff80008284b150
[ 2434.067632] x29: ffff80008284b150 x28: ffff000000000000 x27: ffffff7fc1e67b40
[ 2434.074795] x26: 0000008040000000 x25: ffff000005530dc0 x24: ffff0001819d4000
[ 2434.081957] x23: ffff800083c24000 x22: ffffffffffffffef x21: ffff000005530d90
[ 2434.089120] x20: 0000000000000338 x19: ffffff7fc02fa280 x18: ffffa0df4f575c98
[ 2434.096282] x17: ffffffffffffffff x16: ffffa0df4e7f7fb8 x15: 1a2eab048aa7b5a6
[ 2434.103444] x14: 000000000000000f x13: ffbd909106540102 x12: 8da2c611d1ad8c8e
[ 2434.110606] x11: 10ac96fed482a890 x10: c078c512030815a9 x9 : 79097f56fb5b8792
[ 2434.117769] x8 : b87a6b11b7bfa131 x7 : 37fd7a2c2c636356 x6 : af0048761f760018
[ 2434.124931] x5 : ffff00002fa27d20 x4 : ffff800083c24018 x3 : ffff0001e67b7f80
[ 2434.132093] x2 : fffffffffffffca0 x1 : ffff800083c24338 x0 : ffff00002fa28000
[ 2434.139255] Call trace:
[ 2434.141702]  __pi_memcpy_generic+0x1dc/0x230 (P)
[ 2434.146333]  zram_write_page+0x15c/0x518 [zram]
[ 2434.150883]  zram_submit_bio+0x53c/0x6a0 [zram]
[ 2434.155427]  __submit_bio+0x178/0x238
[ 2434.159100]  submit_bio_noacct_nocheck+0x16c/0x388
[ 2434.163906]  submit_bio_noacct+0x260/0x558
[ 2434.168013]  submit_bio+0xac/0x238
[ 2434.171421]  submit_bio_wait+0x5c/0xf0
[ 2434.175181]  swap_writepage_bdev_sync+0xe8/0x138
[ 2434.179813]  __swap_writepage+0x100/0x2f8
[ 2434.183833]  swap_writepage+0x290/0x440
[ 2434.187678]  pageout+0x140/0x3c0
[ 2434.190915]  shrink_folio_list+0x704/0xe00
[ 2434.195022]  evict_folios+0x254/0x798
[ 2434.198692]  try_to_shrink_lruvec+0x218/0x268
[ 2434.203060]  shrink_one+0x78/0x220
[ 2434.206467]  shrink_node+0xb04/0xdc0
[ 2434.210049]  balance_pgdat+0x484/0x9b0
[ 2434.213806]  kswapd+0x218/0x4b8
[ 2434.216952]  kthread+0x14c/0x210
[ 2434.220186]  ret_from_fork+0x10/0x20
[ 2434.223773] Code: 54000169 a93f1ca6 a97f1c86 a93e24a8 (a97e2488) 
[ 2434.229887] ---[ end trace 0000000000000000 ]---
[ 2434.234647] ------------[ cut here ]------------
[ 2434.234654] WARNING: CPU: 1 PID: 66 at kernel/exit.c:900 do_exit+0x81c/0x988
[ 2434.246368] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device vhost_vsock vmw_vsock_virtio_transport_common vhost vhost_iotlb vsock zram algif_hash algif_skcipher af_alg bnep binfmt_misc joydev spidev hci_uart btbcm bluetooth ecdh_generic ecc brcmfmac_wcc aes_ce_blk aes_ce_cipher ghash_ce gf128mul raspberrypi_hwmon rpi_hevc_dec sha2_ce pisp_be v4l2_mem2mem videobuf2_dma_contig sha256_arm64 vc4 sha1_ce sha1_generic snd_soc_hdmi_codec drm_exec drm_display_helper videobuf2_memops brcmfmac brcmutil cec videobuf2_v4l2 i2c_brcmstb drm_client_lib drm_dma_helper v3d gpu_sched spi_bcm2835 snd_soc_core drm_shmem_helper snd_compress cfg80211 snd_pcm_dmaengine videodev rfkill snd_pcm gpio_keys snd_timer videobuf2_common pwm_fan snd drm_kms_helper rp1_pio mc raspberrypi_gpiomem rp1 rp1_mailbox rp1_adc squashfs nvmem_rmem uio_pdrv_genirq uio sch_fq_codel usbip_host usbip_core i2c_dev crypto_user drm fuse drm_panel_orientation_quirks backlight dm_mod nfnetlink ip_tables x_tables ipv6
[ 2434.334902] CPU: 1 UID: 0 PID: 66 Comm: kswapd0 Tainted: G      D             6.15.0-rc2-1-MANJARO-RPI5 #1 PREEMPT 
[ 2434.345382] Tainted: [D]=DIE
[ 2434.348266] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[... cut ...]
[ 2434.524780]  swap_writepage+0x290/0x440
[ 2434.528625]  pageout+0x140/0x3c0
[ 2434.531861]  shrink_folio_list+0x704/0xe00
[ 2434.535970]  evict_folios+0x254/0x798
[ 2434.539639]  try_to_shrink_lruvec+0x218/0x268
[ 2434.544007]  shrink_one+0x78/0x220
[ 2434.547415]  shrink_node+0xb04/0xdc0
[ 2434.550997]  balance_pgdat+0x484/0x9b0
[ 2434.554754]  kswapd+0x218/0x4b8
[ 2434.557900]  kthread+0x14c/0x210
[ 2434.561134]  ret_from_fork+0x10/0x20
[ 2434.564720] ---[ end trace 0000000000000000 ]---

```

pelwell, 2025-04-15:
That looks seriously borked - like RAM corruption.

Dark-Sky, 2025-04-15:
LOL. Way over my head. Glad it is not me trying to figure it out....

Dark-Sky, 2025-04-15:
I do not think I was getting any compile issues mentioned above but here is my log from doing the latest commit today:

https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-rpi5-rc/-/jobs/15682

popcornmix, 2025-04-15:
@Dark-Sky can you disable zram (I assume you have that in cmdline.txt?) and see if crashing stops?

Dark-Sky, 2025-04-15:
Will do

by, 2025-04-15:
It appears after there might be an issue with too many parallel threads: when compiling the kernel natively with -j6 (according to https://www.raspberrypi.com/documentation/computers/linux_kernel.html#native-build), the errors arise (and memory usage blows up), without -j, it seems to work.

popcornmix, 2025-04-15:
> It appears after there might be an issue with too many parallel threads: when compiling the kernel natively with -j6 (according to https://www.raspberrypi.com/documentation/computers/linux_kernel.html#native-build), the errors arise (and memory usage blows up), without -j, it seems to work.

of course. Running 6 compile threads in parallel needs 6 times the RAM. If you are unlucky and they are all compiling large files (or even worse linking) then out-of-memory is possible.

by, 2025-04-15:
Makes sense; as it worked out well according to the instructions up until 6.14, it might make even more sense to put a hint into the kernel build doc.?

Dark-Sky, 2025-04-15:
> [@Dark-Sky](https://github.com/Dark-Sky) can you disable zram (I assume you have that in cmdline.txt?) and see if crashing stops?

I disabled zram and I can not seem to make it freeze. I had chromium open and firefox and did some browsing while compiling the latest kernel with distcc and -j19 with vim3pro and rockpro64 as slaves.

If this turns out to be the issue I am guessing this is something you all won't fix. zram works well with the older kernels.

pelwell, 2025-04-16:
A lot of zram changes went into 6.15-rc1:
```
2ad951865aa78 zram: add might_sleep to zcomp API
a6d2193b3ef53 zram: do not leak page on writeback_store error path
5b683d4e987d2 zram: do not leak page on recompress_store error path
f66140eb71053 zram: permit reclaim in zstd custom allocator
82f91900c7222 zram: switch to new zsmalloc object mapping API
7e1b0212d4d59 zram: move post-processing target allocation
b0624f0b22342 zram: rework recompression loop
9724bef96df47 zram: filter out recomp targets based on priority
d7fdc5a620aee zram: limit max recompress prio to num_active_comps
f3b0c6c8996a8 zram: remove writestall zram_stats member
9c7ccc8d99ad4 zram: add GFP_NOWARN to incompressible zsmalloc handle allocation
80af56cb29332 zram: remove second stage of handle allocation
4127e13c9302f zram: remove max_comp_streams device attr
be656187b8a9c zram: remove unused crypto include
2efa9e9eb4db2 zram: permit preemption with active compression stream
bd175a1d84e3f zram: sleepable entry locking
```
If nothing changes upstream we could revert these (or a subset) to see if that helps.

Dark-Sky, 2025-04-22:
I am still getting some sort of weird memory issue after a period of about 1 1/2 days of continuous running before I have to reboot. I added a swap file and also tried setting `swapinness = 10`. When monitoring the memory and the swap file neith one will get close to maxing out but having firefox minimized and trying to restore it all I see is a black window and eventually it crashes. If I try to restart it I see a ton of `drm: unable to allocate memory` errors. Then chromium will fail to start also.

None of this happens with the older kernels.

pelwell, 2025-05-08:
Has anything changed with the recent releases? We're up to -rc5 now.

Dark-Sky, 2025-05-08:
> Has anything changed with the recent releases? We're up to -rc5 now.

I still can not use zram. I have no issue while using a swapfile.