raspberrypi/linux #6783: Pi 5 ACT LED: Incorrect mmc0 idle state & inverted brightness control (Kernel 6.12.20+rpt-rpi-2712)

tome9111991, 2025-04-13:
### Describe the bug

The green ACT LED (controlled via AON_GPIO_09) on the Raspberry Pi 5 exhibits non-standard and unexpected behavior with common triggers and direct sysfs brightness control with Kernel 6.12.20+rpt-rpi-2712 (and potentially other recent versions).

**Issue 1: `mmc0` Trigger Idle State**
- When `dtparam=act_led_trigger=mmc0` is set, the trigger correctly becomes active (`[mmc0]`).
- The LED correctly blinks during SD card activity.
- **Problem:** The LED remains **solid green** when the SD card is idle, contrary to the `default-state = "off"` set in the device tree.

**Issue 2: `cpu` Trigger State**
- When `dtparam=act_led_trigger=cpu` is set, the trigger correctly becomes active (`[cpu]`).
- **Problem:** The LED remains **permanently off**, even under high CPU load (e.g., `sha256sum /dev/zero`).

**Issue 3: Manual Brightness Control Inverted**
- When the trigger is set to `none` (`[none]` confirmed active), direct control via sysfs shows inverted logic:
    - `echo 1 > /sys/class/leds/ACT/brightness` turns the LED **OFF**.
    * `echo 0 > /sys/class/leds/ACT/brightness` turns the LED **ON**.

**Conclusion/Hypothesis:**
The inverted manual brightness control strongly suggests a driver bug related to handling the `GPIO_ACTIVE_LOW` property for `AON_GPIO_09`. This likely causes the incorrect idle state for `mmc0` (brightness 0 interpreted as ON) and the lack of output for `cpu` (brightness 1 interpreted as OFF).

### Steps to reproduce the behaviour

1. Use a Raspberry Pi 5 with a recent Raspberry Pi OS (Debian Bookworm base) and Kernel (e.g., 6.12.20+rpt-rpi-2712).
2. Edit `/boot/firmware/config.txt` and ensure any existing `dtparam=act_led_...` lines are commented out.
3. **Test `mmc0`:** Add `dtparam=act_led_trigger=mmc0`. Save and reboot. Observe LED is solid green when idle. Check with `cat /sys/class/leds/ACT/trigger` (shows `[mmc0]`).
4. **Test `cpu`:** Change the line to `dtparam=act_led_trigger=cpu`. Save and reboot. Observe LED is permanently off, even under load (e.g., `sha256sum /dev/zero`). Check with `cat /sys/class/leds/ACT/trigger` (shows `[cpu]`).
5. **Test Manual Control:** Change the line to `dtparam=act_led_trigger=none`. Save and reboot. Confirm trigger is `[none]` with `cat /sys/class/leds/ACT/trigger`. Then execute:
    - `echo 1 | sudo tee /sys/class/leds/ACT/brightness` -> Observe LED turns OFF.
    - `echo 0 | sudo tee /sys/class/leds/ACT/brightness` -> Observe LED turns ON.

### Device (s)

Raspberry Pi 5

### System

cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

vcgencmd version
2024/09/23 14:02:56
Copyright (c) 2012 Broadcom
version 26826259 (release) (embedded)

uname -a
Linux raspberrypi 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

pelwell, 2025-04-13:
Are you saying that this behaviour has changed since 6.6?

tome9111991, 2025-04-13:
not sure i have only test 6.12

pelwell, 2025-04-13:
OK, thanks - no need to try 6.6.

tome9111991, 2025-04-13:
dtparam=act_led_activelow=off 

fix this issue

tome9111991, 2025-04-14:
Thank you for the detailed explanation! I understand the reasoning behind the intended Pi Zero-like dual-duty behavior for the ACT LED (solid green for power/idle, blinking off for activity) and the resulting inverted manual brightness control as a side effect.

Regarding the `mmc0` trigger and the manual control inversion: I tested setting **both** `dtparam=act_led_trigger=mmc0` AND `dtparam=act_led_activelow=off` in config.txt.
With this specific combination, the ACT LED now behaves in the 'traditional' way: it is correctly **OFF** when the SD card is idle and blinks (turns briefly ON) during I/O activity. The manual brightness control (with trigger `none`) also works non-inverted (1=ON, 0=OFF) when `act_led_activelow=off` is set.

So, it seems that adding `dtparam=act_led_activelow=off` not only corrects the manual brightness control logic but also restores the 'off-at-idle' behavior for the `mmc0` trigger on my RPi 5 / Kernel 6.12.x system. This behavior is preferable for my use case as it provides a clearer distinction between idle and activity states. Thanks again for the insights!

closed 2025-04-14 by tome9111991