raspberrypi/linux #6784: drm/v3d: incorrect drm information in fdinfo.

xiaohuirong, 2025-04-14:
### Describe the bug

In a window manager, commands run in a terminal that do not use GPU resources will still show `drm-engine-render` usage information under the `fdinfo` of the command's pid. This information appears to be inherited from the WM.

[image]

### Steps to reproduce the behaviour

1. launch a window manager; here I am using wayfire.  
2. open a terminal in the window manager and run the command `sleep 300`.  
3. check the `fdinfo` of the command’s pid, you can find that `drm-engine-render` is being used.

### Device (s)

Raspberry Pi 4 Mod. B

### System

os: `ArchlinuxArm aarch64`
kernel: 
```
Linux netpi 6.12.22-1-rpi #1 SMP PREEMPT Fri Apr 11 06:09:22 MDT 2025 aarch64 GNU/Linux
```
firmware:
```
Mar 31 2025 17:12:03 
Copyright (c) 2012 Broadcom
version 897f94edff080e2b325930669fa4e9caf469e090 (clean) (release) (start)
```

popcornmix, 2025-04-14:
@mairacanal I wonder if you could confirm what the expected behaviour here should be?

mairacanal, 2025-04-14:
I'm unable to reproduce this issue on RPi OS with labwc or Wayfire. Therefore, I'd need more information. On RPi 4, running libwc, I get the expected behaviour:

```
mairacanal@raspberrypi:~ $ uname -r
6.12.20+rpt-rpi-v8
mairacanal@raspberrypi:~ $ ps -x
    PID TTY      STAT   TIME COMMAND
    826 ?        Ss     0:00 /lib/systemd/systemd --user
    827 ?        S      0:00 (sd-pam)
    842 ?        S<sl   0:00 /usr/bin/pipewire
    843 ?        Ssl    0:00 /usr/bin/pipewire -c filter-chain.conf
    844 ?        S<sl   0:00 /usr/bin/wireplumber
    845 ?        S<sl   0:00 /usr/bin/pipewire-pulse
    846 ?        Ssl    0:01 /usr/bin/labwc -m
    857 ?        Ss     0:00 /usr/bin/dbus-daemon --session --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
    915 ?        Ss     0:00 /usr/bin/ssh-agent /usr/bin/labwc-pi
   1022 ?        S      0:00 /bin/sh /usr/bin/lwrespawn /usr/bin/pcmanfm --desktop --profile LXDE-pi
   1023 ?        S      0:00 /bin/sh /usr/bin/lwrespawn /usr/bin/wf-panel-pi
   1024 ?        S      0:00 /usr/bin/kanshi
   1029 ?        S      0:00 /usr/bin/python3 /usr/share/system-config-printer/applet.py
   1033 ?        S      0:00 systemd-inhibit --what=handle-power-key gtk-nop
   1054 ?        Sl     0:00 /usr/libexec/polkit-mate-authentication-agent-1
   1060 ?        Sl     0:00 gtk-nop
   1066 ?        Sl     0:01 /usr/bin/pcmanfm --desktop --profile LXDE-pi
   1067 ?        Sl     0:01 /usr/bin/wf-panel-pi
   1092 ?        Ssl    0:00 /usr/libexec/gvfsd
   1113 ?        Ssl    0:00 /usr/libexec/gvfs-udisks2-volume-monitor
   1124 ?        Ssl    0:00 /usr/libexec/gvfs-gphoto2-volume-monitor
   1134 ?        Ssl    0:00 /usr/libexec/gvfs-goa-volume-monitor
   1135 tty1     S+     0:00 -bash
   1144 ?        Ssl    0:00 /usr/libexec/gvfs-afc-volume-monitor
   1146 ?        Sl     0:00 /usr/lib/menu-cache/menu-cached /run/user/1000/menu-cached-:0
   1163 ?        Ssl    0:00 /usr/libexec/gvfs-mtp-volume-monitor
   1168 ?        Sl     0:00 /usr/libexec/gvfsd-trash --spawner :1.17 /org/gtk/gvfs/exec_spaw/0
   1754 ?        S      0:00 sshd: mairacanal@pts/0
   1755 pts/0    Ss     0:00 -bash
   1785 ?        Sl     0:00 lxterminal
   1796 pts/1    Ss     0:00 bash
   1820 pts/1    S+     0:00 sleep 300
   1821 pts/0    R+     0:00 ps -x
mairacanal@raspberrypi:~ $ cd /proc/1820
mairacanal@raspberrypi:/proc/1820 $ cat fdinfo/*
pos:	0
flags:	02000002
mnt_id:	23
ino:	4
pos:	0
flags:	02000002
mnt_id:	23
ino:	4
pos:	0
flags:	02000002
mnt_id:	23
ino:	4
mairacanal@raspberrypi:/proc/1820 $ cat cmdline
sleep300
```

In order to have more information for debugging, could you share the following information with me?

1. Which terminal emulator are you using?
2. Which shell are you using?
3. Are you using GNU coreutils, BusyBox, or other? 
4. While `sleep 300` is running, could you run `sudo lsof | grep sleep` and share the output with me?
5. While `sleep 300` is running,  could you run the following command line and share the output with me?

```
for p in `sudo lsof /dev | awk 'NR > 1 {print $2}'`; do if sudo grep -qr drm-client-id /proc/$p/fdinfo; then cat /proc/$p/cmdline; echo " ($p):"; sudo grep -rh 'drm-client-id:' /proc/$p/fdinfo | cut -d':' -f2 | sort | uniq ; fi; done
```

xiaohuirong, 2025-04-14:
@mairacanal  I encountered this issue when upgrading from 6.12.20 to 6.12.22. Could you try the 6.12.22 kernel version?

mairacanal, 2025-04-14:
@xiaohuirong I couldn't reproduce this issue on RPi OS when using the 6.12.22 kernel. I got the same output as 6.12.20.

Is it possible that another package was updated simultaneously? I would still appreciate it if you could share more information about your setup.

xiaohuirong, 2025-04-15:
> I'm unable to reproduce this issue on RPi OS with labwc or Wayfire. Therefore, I'd need more information. On RPi 4, running libwc, I get the expected behaviour:
>
> ```
> mairacanal@raspberrypi:~ $ uname [...]

Sorry for the delayed reply. Here is the response to the above questions.

1. The terminal is `xfce4-terminal`
2. The shell is `fish`
3.  I am using `coreutils`
4.  Here is lsof output
```
[I] alarm@netpi:~
➤ sudo lsof | grep sleep
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse.portal file system /run/user/1000/doc
      Output information may be incomplete.
sleep     20761                            alarm cwd       DIR               0,25     1394       1421 /home/alarm
sleep     20761                            alarm rtd       DIR               0,25      140        256 /
sleep     20761                            alarm txt       REG               0,25    67832    2797976 /usr/bin/sleep
sleep     20761                            alarm mem       REG               0,23             2797976 /usr/bin/sleep (path dev=0,25)
sleep     20761                            alarm mem       REG               0,23             2798093 /usr/share/locale/zh_CN/LC_MESSAGES/coreutils.mo (path dev=0,25)
sleep     20761                            alarm mem       REG               0,23             2445773 /usr/lib/locale/locale-archive (path dev=0,25)
sleep     20761                            alarm mem       REG               0,23             2445056 /usr/lib/libc.so.6 (path dev=0,25)
sleep     20761                            alarm mem       REG               0,23             2445047 /usr/lib/ld-linux-aarch64.so.1 (path dev=0,25)
sleep     20761                            alarm mem       REG               0,23             2445776 /usr/lib/gconv/gconv-modules.cache (path dev=0,25)
sleep     20761                            alarm   0r     FIFO               0,14      0t0      14601 pipe
sleep     20761                            alarm   1w      REG               0,25    57847    2025511 /home/alarm/.cache/wayfire.log
sleep     20761                            alarm   2w      REG               0,25    57847    2025511 /home/alarm/.cache/wayfire.log
sleep     20761                            alarm  15u      CHR            226,128      0t0        286 /dev/dri/renderD128
sleep     20761                            alarm  20u      CHR            226,128      0t0        286 /dev/dri/renderD128
```
5. This is the output file from running it in zsh.

[output.txt](https://github.com/user-attachments/files/19752309/output.txt)

mairacanal, 2025-04-15:
First, from the information you provided, this is not a kernel bug. `sleep` indeed has two opened V3D fd and you can see it in `lsof` and `output.txt`. Therefore, the kernel is just reporting the GPU usage of the DRM clients that this process has, which is the expected behaviour.

You are seeing a misleading result because there is probably a file descriptor leak in user-space.

Note from `output.txt` that Wayfire, Alacritty and Fish have the fds 279 and 282, which are the DRM fds that `sleep` has. As far as I understand, this indicates a file descriptor leak in the user-space. Searching a bit in Alacritty's repo, I could see some old issues related to it [1], showing how common such issue is. I'm not saying that it is Alacritty, I don't know which software is the culprit here.

I'd recommend you to investigate which one of the programs is the causing this issue. It can be Wayfire, wlroots, Alacritty, Fish... You can try to eliminate suspects by using different applications. For example, to eliminate Fish, you can use zsh and see if you can still reproduce the issue and so on. Another important information is the other packages that were upgraded with the kernel. You can also use `ktrace` to investigate it.

@popcornmix, as this isn't a kernel issue, I believe you can close the issue if the user is okay with it.

[1] https://github.com/alacritty/alacritty/issues?q=is%3Aissue%20state%3Aclosed%20leak

xiaohuirong, 2025-04-15:
Well, it seems this really isn't a kernel issue. Thanks for your help. I'll close this issue for now and investigate further on my own to figure out where the leak is coming from.

closed 2025-04-15 by xiaohuirong