raspberrypi/linux #6786: Compiling error with drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c

hoorna, 2025-04-14:
### Describe the bug

With cross compiling the RPi kernel I always get the following warming:

drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.c:67:3: warning: 'snprintf' will always be truncated; specified size is 9, but format string expands to at least 10 [-Wformat-truncation]
   67 |                 snprintf(text, 9, " %4d:%1u  ", hfnum.b.frnum/8, hfnum.b.frnum & 7);
        |                 ^

I cross compile the RPi kernel on Ubuntu (Windows Subsystem for Linux 2) with CC=clang and LLVM=1.  On this moment I am using branch version 6.14.2 but the warning came also up using earlier versions.

What can I do to solve this issue?

### Steps to reproduce the behaviour

- Download the RPi 6.14.2 branch from Github to a Raspberry Pi 4 running Raspberry OS
- Run on the Raspberry Pi 4:
    "make KERNEL=kernel8 bcm2711_defconfig"
- Download the same RPi 6.14.2 branch as above from Github to Ubuntu (WSL2)
- Copy .config from the Raspberry Pi 4 to Ubuntu
- Run on Ubuntu:
    "make -j10 LLVM=1 CC=clang ARCH=arm64 KERNEL=kernel8 CROSS_COMPILE=aarch64-linux-gnu- Image
     modules dtbs"

### Device (s)

Other

### System

Cross compiling on Ubuntu with WSL; 5.15.167.4-microsoft-standard-WSL2

### Logs

See above

### Additional context

-

pelwell, 2025-04-14:
I should also add that deleting the second space does not change the result - it is overwritten by the input string - just avoids the warning.

P33M, 2025-04-14:
It is a typo but not by me. The buffer print code is from @ghollingworth's first implementation of the FIQ in late 2012.

pelwell, 2025-04-14:
It's hard to believe, but I'll take your word for it (and submit a fixing PR).

pelwell, 2025-04-14:
Closure was automatic, triggered by the Fixes: tag in the commit, but I hope you won't need to reopen this issue.

closed 2025-04-14 by pelwell

hoorna, 2025-04-15:
My apologies for posting again. I see a failure (red cross) with this fixup. What is wrong?

Furthermore I am sorry that I have another stupid question. When will this fixup in the 6.12.y branch be included in the 6.14.y branch?

pelwell, 2025-04-15:
The build errors were random download failures experienced by the GitHub workflows - I've kicked them off again, but I know they will pass (eventually).

And I'm just about to port the recent 6.12 changes forward to rpi-6.14.y and rpi-6.15.y.

hoorna, 2025-04-15:
Thank you very much for your response and your patient answers!

Last question from this relative newbie: how can I see when the port to rpi-6.14.y (and rpi-6.15.y) has taken place?

pelwell, 2025-04-15:
In general, fetch the tree until it's there (or hit refresh on the relevant GitHub list of commits). You also get merge notifications in the issues ("pelwell added a commit that references this issue...") if we've remembered to mention the issue in the commit message.

In this case, they've just been done.