raspberrypi/linux #6787: Please enable CONFIG_POWERZ for USB-C power meter

DanaGoyette, 2025-04-15:
### Describe the bug

Please enable the PowerZ sensors driver, to allow using the USB-C power monitor.

https://www.phoronix.com/news/Linux-POWER-Z-USB-Power-Driver

```
$ grep POWERZ /boot/config-6.12.20+rpt-rpi-2712
# CONFIG_SENSORS_POWERZ is not set
```

I'd build the module myself out-of-tree, but doing `apt-get source linux-image-$(uname -r)` gets me the source for Debian's 6.1 kernel, not the raspberry pi kernel source.

### Steps to reproduce the behaviour

Obtain one of the Power-Z power meters; connect it inline with the power input; connect a USB A-to-C data cable to the side USB-C port; notice that the powerz driver does not load, so you don't get the voltage, current, or power stats in `sensors`.

### Device (s)

Raspberry Pi 5

### System

OS:
```
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4
```

Firmware:
```
2025/03/19 13:41:26
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)
```

Kernel:
```
Linux pi5 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

pelwell, 2025-05-09:
Sorry - this one slipped down the gaps. There's now a pull request for the change: https://github.com/raspberrypi/linux/pull/6845
In about 40 minutes you will be able to install a trial kernel including the PowerZ module using `sudo rpi-update pulls/6845`. Backup any important data first.

DanaGoyette, 2025-05-21:
I installed that kernel using rpi-update, but it fails to boot, thanks to not it building an initramfs.  Since my root is btrfs, it just dumps me to the initramfs shell.  I had to use the initramfs shell to copy the packaged kernel back to the boot partition.

There's a bug report for that lack of initramfs, but it got closed without having been fixed: https://github.com/Hexxeh/rpi-update/issues/252

I'll have to just wait until the next released version has the change.

DanaGoyette, 2025-05-21:
I copied my btrfs install to a different device, converting to ext4 at the same time, and I can confirm that the driver works.  I think IBUS is negative because I have the device plugged in at the device end instead of at the charger end.

```
VBUS:          5.12 V  (avg =  +5.12 V)
VCC1:          1.60 V
VCC2:          5.00 mV
VDP:          16.00 mV
VDM:          16.00 mV
VDD:           3.20 V
TEMP:         +30.3°C
IBUS:        -665.00 mA (avg =  -0.67 A)
```

pelwell, 2025-05-22:
Thanks for the feedback

closed 2025-05-22 by pelwell