raspberrypi/linux #6792: Unable to handle kernel NULL pointer dereference

Tony763, 2025-04-15:
### Describe the bug

Hi, just deployed a fresh sdcard, with latest rpios arm64 image and applied all available updates from apt.
RPI freezes over some small amount of time with following in dmesg.

### Steps to reproduce the behaviour

Just connected over ssh, RPI in idle.

### Device (s)

Raspberry Pi 5

### System

will add

### Logs

Apr 15 13:58:03.028579 raspberrypi kernel: Unable to handle kernel paging request at virtual address ffffc04000004000
Apr 15 13:58:03.028788 raspberrypi kernel: Mem abort info:
Apr 15 13:58:03.097078 raspberrypi kernel:   ESR = 0x0000000096000005
Apr 15 13:58:03.118977 raspberrypi kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Apr 15 13:58:03.139927 raspberrypi kernel:   SET = 0, FnV = 0
Apr 15 13:58:03.162898 raspberrypi kernel:   EA = 0, S1PTW = 0
Apr 15 13:58:03.186231 raspberrypi kernel:   FSC = 0x05: level 1 translation fault
Apr 15 13:58:03.210809 raspberrypi kernel: Data abort info:
Apr 15 13:58:03.237807 raspberrypi kernel:   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
Apr 15 13:58:03.260124 raspberrypi kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Apr 15 13:58:03.282217 raspberrypi kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Apr 15 13:58:03.304589 raspberrypi kernel: Unable to handle kernel paging request at virtual address ffffc04000004000
Apr 15 13:58:03.324790 raspberrypi kernel: swapper pgtable: 16k pages, 47-bit VAs, pgdp=000000000148c000
Apr 15 13:58:03.346407 raspberrypi kernel: Mem abort info:
Apr 15 13:58:03.405576 raspberrypi kernel: [ffffc04000004000] pgd=0000000000000000
Apr 15 13:58:03.446019 raspberrypi kernel:   ESR = 0x0000000096000005
Apr 15 13:58:03.467434 raspberrypi kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Apr 15 13:58:03.517873 raspberrypi kernel: , p4d=0000000000000000
Apr 15 13:58:03.535937 raspberrypi kernel:   SET = 0, FnV = 0
Apr 15 13:58:03.535977 raspberrypi kernel: , pud=0000000000000000
Apr 15 13:58:03.562195 raspberrypi kernel:   EA = 0, S1PTW = 0
Apr 15 13:58:03.562235 raspberrypi kernel: 
Apr 15 13:58:03.586392 raspberrypi kernel: Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
Apr 15 13:58:03.586429 raspberrypi kernel:   FSC = 0x05: level 1 translation fault
Apr 15 13:58:03.612968 raspberrypi kernel: Modules linked in: veth nf_conntrack_netlink
Apr 15 13:58:04.697086 raspberrypi kernel: Data abort info:
Apr 15 13:58:04.697139 raspberrypi kernel:  xt_nat xt_tcpudp xt_conntrack xt_MASQUERADE bridge
Apr 15 13:58:07.147706 raspberrypi kernel:   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
Apr 15 13:58:07.147769 raspberrypi kernel:  stp
Apr 15 13:58:08.365615 raspberrypi kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Apr 15 13:58:08.365682 raspberrypi kernel:  llc
Apr 15 13:58:08.432451 raspberrypi kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Apr 15 13:58:08.432504 raspberrypi kernel:  xt_set
Apr 15 13:58:08.477128 raspberrypi kernel: swapper pgtable: 16k pages, 47-bit VAs, pgdp=000000000148c000
Apr 15 13:58:08.477176 raspberrypi kernel:  ip_set
Apr 15 13:58:08.501925 raspberrypi kernel: [ffffc04000004000] pgd=0000000000000000
Apr 15 13:58:08.501970 raspberrypi kernel:  nft_chain_nat
Apr 15 13:58:08.545927 raspberrypi kernel: , p4d=0000000000000000
Apr 15 13:58:08.545949 raspberrypi kernel:  nf_nat
Apr 15 13:58:08.545968 raspberrypi kernel: , pud=0000000000000000
Apr 15 13:58:08.545986 raspberrypi kernel:  nf_conntrack
Apr 15 13:58:08.546006 raspberrypi kernel: 
Apr 15 13:58:08.546024 raspberrypi kernel:  nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype nft_compat nf_tables nfnetlink xfrm_user xfrm_algo overlay algif>
Apr 15 13:58:08.548830 raspberrypi kernel: CPU: 2 UID: 0 PID: 1842 Comm: dockerd Not tainted 6.12.20+rpt-rpi-2712 #1  Debian 1:6.12.20-1+rpt1~bpo12+1
Apr 15 13:58:08.548858 raspberrypi kernel: Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
Apr 15 13:58:08.548878 raspberrypi kernel: pstate: 80400009 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Apr 15 13:58:08.548898 raspberrypi kernel: pc : __arch_copy_to_user+0xe8/0x240
Apr 15 13:58:08.548918 raspberrypi kernel: lr : _copy_to_iter+0xf4/0x6a8
Apr 15 13:58:08.548945 raspberrypi kernel: sp : ffffc00084693b60
Apr 15 13:58:08.548966 raspberrypi kernel: x29: ffffc00084693b60 x28: ffff8000037bd140 x27: 0000000000000068
Apr 15 13:58:08.548985 raspberrypi kernel: x26: ffffc04000004000 x25: 0000000000000001 x24: 0000000000000000
[... 158 lines cut ...]
Apr 15 14:00:20.035890 raspberrypi kernel:  ksys_read+0xf0/0x118
Apr 15 14:00:20.035905 raspberrypi kernel:  __arm64_sys_read+0x24/0x38
Apr 15 14:00:20.035924 raspberrypi kernel:  invoke_syscall+0x50/0x120
Apr 15 14:00:20.035942 raspberrypi kernel:  el0_svc_common.constprop.0+0x48/0xf0
Apr 15 14:00:20.035959 raspberrypi kernel:  do_el0_svc+0x24/0x38
Apr 15 14:00:20.035976 raspberrypi kernel:  el0_svc+0x30/0xd0
Apr 15 14:00:20.035994 raspberrypi kernel:  el0t_64_sync_handler+0x100/0x130
Apr 15 14:00:20.036012 raspberrypi kernel:  el0t_64_sync+0x190/0x198
Apr 15 14:00:20.036030 raspberrypi kernel: Code: b8404423 b80008c3 910010c6 36080082 (78402423) 
Apr 15 14:00:20.036073 raspberrypi kernel: ---[ end trace 0000000000000000 ]---
Apr 15 14:02:44.170159 raspberrypi kernel: vethb020483: renamed from eth0
Apr 15 14:02:44.170281 raspberrypi kernel: br-80875c151c00: port 1(veth7a29126) entered disabled state
Apr 15 14:02:44.227003 raspberrypi kernel: br-80875c151c00: port 1(veth7a29126) entered disabled state
Apr 15 14:02:44.227154 raspberrypi kernel: veth7a29126 (unregistering): left allmulticast mode
Apr 15 14:02:44.227182 raspberrypi kernel: veth7a29126 (unregistering): left promiscuous mode
Apr 15 14:02:44.240497 raspberrypi kernel: br-80875c151c00: port 1(veth7a29126) entered disabled state
Apr 15 14:02:44.618011 raspberrypi kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000231
Apr 15 14:02:44.618379 raspberrypi kernel: Mem abort info:
Apr 15 14:02:44.685091 raspberrypi kernel:   ESR = 0x0000000096000005
Apr 15 14:02:44.734234 raspberrypi kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Apr 15 14:02:44.763133 raspberrypi kernel:   SET = 0, FnV = 0
Apr 15 14:02:44.793663 raspberrypi kernel:   EA = 0, S1PTW = 0
Apr 15 14:02:44.829040 raspberrypi kernel:   FSC = 0x05: level 1 translation fault
Apr 15 14:02:44.869361 raspberrypi kernel: Data abort info:
Apr 15 14:02:44.892306 raspberrypi kernel:   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000

pelwell, 2025-04-15:
1. What hardware do you have attached to this Pi 5?
2. What are you running in docker?

Tony763, 2025-04-15:
1. USB Keyboard, ETH, HDMI
2. For now nothing, it fails even without container running

alessandromrc, 2025-04-30:
Hi @Tony763 have you tried to run any other SD card with this Pi, does it still happen on the same kernel?

It could simply be because of a defective Pi / SD card... I really doubt that the kernel is the actual issue.

Tony763, 2025-06-23:
Yes, but result was same. I moved to diferrent SBC so I cannot test it any future right now.